marcone/teslausb #2: Stuck at Creating backing disk files.

SamGabbay, 2019-03-29:
Doing a headless install pulling from the main-dev branch. Gets stuck at Creating backing disk files.

Here are the logs from teslausb-headless-setup.log for reference.

Tue Oct 30 18:41:06 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Tue Oct 30 18:41:06 GMT 2018 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Tue Oct 30 18:41:06 GMT 2018 : Rebooting...
Tue Oct 30 18:41:31 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Tue Oct 30 18:41:31 GMT 2018 : Grabbing main setup file.
Tue Oct 30 18:41:32 GMT 2018 : Starting setup.
Tue Oct 30 18:41:32 GMT 2018 : Updating package index files...
Fri Mar 29 17:23:07 GMT 2019 : Verifying that the requested configuration is valid...
Fri Mar 29 17:23:08 GMT 2019 : Downloaded /tmp/verify-configuration.sh ...
Fri Mar 29 17:23:08 GMT 2019 : Verifying that there is sufficient space available on the MicroSD card...
Fri Mar 29 17:23:08 GMT 2019 : There is sufficient space available.
Fri Mar 29 17:23:18 GMT 2019 : Downloading additional setup scripts.
Fri Mar 29 17:23:19 GMT 2019 : Downloaded /tmp/create-backingfiles-partition.sh ...
Fri Mar 29 17:23:19 GMT 2019 : Downloaded /tmp/create-backingfiles.sh ...
Fri Mar 29 17:23:20 GMT 2019 : Downloaded /tmp/make-root-fs-readonly.sh ...
Fri Mar 29 17:23:20 GMT 2019 : Downloaded /root/configure.sh ...
Fri Mar 29 17:23:20 GMT 2019 : Fixing the modules-load parameter in /boot/cmdline.txt...
Fri Mar 29 17:23:20 GMT 2019 : Fixed cmdline.txt.
Fri Mar 29 17:23:29 GMT 2019 : Starting to create backing files partition...
Fri Mar 29 17:23:29 GMT 2019 : Checking existing partitions...
Fri Mar 29 17:23:30 GMT 2019 : Modifying partition table for backing files partition...
Fri Mar 29 17:23:30 GMT 2019 : Modifying partition table for mutable (writable) partition for script usage...
Fri Mar 29 17:23:30 GMT 2019 : Writing updated partitions to fstab and /boot/cmdline.txt
Fri Mar 29 17:23:31 GMT 2019 : Formatting new partitions...
Fri Mar 29 17:23:41 GMT 2019 : Mounting the partition for the backing files...
Fri Mar 29 17:23:41 GMT 2019 : Mounted the partition for the backing files.
Fri Mar 29 17:23:41 GMT 2019 : Creating backing disk files.

therealmarcone, 2019-04-01:
What was your starting point? Did you start with rtgoodwin's prebuilt boot image, or did you make one yourself?
I just put a flashable boot image at https://github.com/therealmarcone/teslausb/releases that's preconfigured to work with my main-dev branchs, so if CIFS backup is what you want, give that a try.

SamGabbay, 2019-04-01:
@therealmarcone just tried yours. Having the same issue of getting stuck at Creating backing disk files.

therealmarcone, 2019-04-01:
That message is the last thing printed [setup-teslausb](https://github.com/therealmarcone/teslausb/blob/main-dev/setup/pi/setup-teslausb#L171) before it calls the [create-backingfiles.sh script](https://github.com/therealmarcone/teslausb/blob/main-dev/setup/pi/create-backingfiles.sh). Looks like something goes wrong in that script, and so it just exits.
You could try running it manually with the same arguments as setup-teslausb does, and see what goes wrong.

SamGabbay, 2019-04-01:
I set the cam size to 80% and then the Music Folder to 2G. I have a 32GB Micro SD Card. Now this is the last line.

Mon  1 Apr 19:02:00 BST 2019 : Creating backing disk files.
Mon  1 Apr 19:02:15 BST 2019 : Main setup completed. Remounting file systems read only

When I reboot, I get the pulsating light but it doesn't show up as a drive on my Mac.

therealmarcone, 2019-04-02:
What did you have the sizes set to before you set them to 80%/2G ?

SamGabbay, 2019-04-02:
I was setting cam size to 90% and leaving the music empty for the remainder. Now what I am doing is the following:

1. Loading Stretch Lite on the SD Card
2. Make the SD Card SSH-able (dw=2, etc, etc..)
3. curl -O https://raw.githubusercontent.com/therealmarcone/teslausb/main-dev/setup/pi/setup-teslausb
4. chmod +x setup-teslausb
5. ./setup-teslausb
6. Watch the logs run through a bunch of commands until I reach "All done"...
7. Reboot.
8. No mounting is done.

therealmarcone, 2019-04-02:
Does the /backingfiles folder exist at that point? Are there any files in it? Is archiveloop running?

To be honest I don't know if starting with plain Stretch Lite is supposed to work. It's probably easier to start with my prebuilt image since it has all the required modifications in terms of additional packages, kernel command line, boot config, etc. I did just upload a fix for the script bailing out if you didn't explicitly specify a music size.

SamGabbay, 2019-04-02:
If I pre-load your image, the headless install starts right away. It gets stuck at the Main setup completed stage.

Sam

SamGabbay, 2019-04-02:
Also, I checked the Rc.local, there’s no edits made in there.

therealmarcone, 2019-04-02:
Sorry, no idea what's going wrong for you. Here's what I do (I do this on a Windows machine):
1. Flash the [prebuilt image](https://github.com/therealmarcone/teslausb/releases) using Etcher
1. After flashing, remove the card from the PC and reinsert it back into the PC to see the "boot" drive
1. Rename the `teslausb_setup_variables.conf.sample` file in the boot drive to `teslausb_setup_variables.conf` and fill in the values I need for wifi, cifs, etc.
1. Insert the card in to the Pi and let it boot up.

SamGabbay, 2019-04-03:
@marcone I did the following:

1. Formatted the SD Card, flashed it with the prebuilt image zip file from [here](https://github.com/marcone/teslausb/releases/download/v1.0a/teslausb-20190402.zip) .
2. After flashing, renamed the config file to teslausb_setup_variables.conf and filled in all the values.
3. Inserted the card into the Pi, let it boot up and SSHed into it using the IP Address.
4. Ran "cd /boot && tail -f teslausb-headless-setup.log"
5. Last line is Main setup completed. Remounting file systems read only. Get's stuck here.

Here is all the lines after running #4.
pi@teslausb:~ $ cd /boot && tail -f teslausb-headless-setup.log
Wed  3 Apr 05:32:50 BST 2019 : Starting to create backing files partition...
Wed  3 Apr 05:32:51 BST 2019 : Checking existing partitions...
Wed  3 Apr 05:32:51 BST 2019 : Modifying partition table for backing files partition...
Wed  3 Apr 05:32:51 BST 2019 : Modifying partition table for mutable (writable) partition for script usage...
Wed  3 Apr 05:32:52 BST 2019 : Writing updated partitions to fstab and /boot/cmdline.txt
Wed  3 Apr 05:32:52 BST 2019 : Formatting new partitions...
Wed  3 Apr 05:33:02 BST 2019 : Mounting the partition for the backing files...
Wed  3 Apr 05:33:02 BST 2019 : Mounted the partition for the backing files.
Wed  3 Apr 05:33:02 BST 2019 : Creating backing disk files.
Wed  3 Apr 05:33:16 BST 2019 : Main setup completed. Remounting file systems read only.

marcone, 2019-04-03:
Hmm, on my Pi, I get a few more lines of log after that:

Wed  3 Apr 03:57:21 BST 2019 : Main setup completed. Remounting file systems read only.
Wed  3 Apr 03:59:27 BST 2019 : Upgrading installed packages...
Wed  3 Apr 03:59:33 BST 2019 : All done.

so it looks like your setup somehow does not make it past the /root/configure.sh script.
Does that file exist on your Pi?
I assume you didn't set "CONFIGURE_ARCHIVING" to false?
When it gets "stuck", is archiveloop running? (`ps aux|grep archiveloop`)

marcone, 2019-04-03:
BTW note there's about 2 minutes between "Main setup completed" and "All done" in my logs. Did you wait long enough?

SamGabbay, 2019-04-03:
To answer the second question, I wait around 10-15 minutes.

To answer the first.. I am re-flashing and trying again. Will get more info for you in just a moment. Hopped out of bed for this one!

SamGabbay, 2019-04-03:
@marcone Okay.

1.  /root/configure.sh is a file on my Pi after the process starts.
2. ps aux|grep archiveloop returns the following:
root      1320  0.0  0.4   4360  1972 pts/2    S+   06:24   0:00 grep archiveloop

marcone, 2019-04-03:
Looks like I'm gonna have to add some additional logging so we can figure out where it's going wrong.

marcone, 2019-04-04:
Can you try again, but point the teslausb_setup_variables.conf to the 'diagnostics' branch?
That prints some more things to the log, hopefully enough to see where it goes wrong.

SamGabbay, 2019-04-05:
Ok ok ok. Here is the logs AFTER the Main setup completed. Remounting file systems read only.

Fri  5 Apr 01:37:41 BST 2019 : Main setup completed. Remounting file systems read only.
Fri  5 Apr 01:37:41 BST 2019 : calling configure.sh
Fri  5 Apr 01:37:41 BST 2019 : /root/configure.sh starting with REPO=marcone, BRANCH=diagnostics, ARCHIVE_SYSTEM=cifs, INSTALL_DIR=/root/bin
Fri  5 Apr 01:37:41 BST 2019 : Getting files from marcone:diagnostics
Fri  5 Apr 01:37:41 BST 2019 : Pushover not configured.
Fri  5 Apr 01:37:41 BST 2019 : Starting download for /root/bin/send-pushover
Fri  5 Apr 01:37:42 BST 2019 : Done
Fri  5 Apr 01:37:42 BST 2019 : Checking archive configs: 
Fri  5 Apr 01:37:42 BST 2019 : done
Fri  5 Apr 01:37:42 BST 2019 : Using archive module: run/cifs_archive
Fri  5 Apr 01:37:42 BST 2019 : Installing base archive scripts into /root/bin
Fri  5 Apr 01:37:42 BST 2019 : Starting download for /root/bin/archiveloop
Fri  5 Apr 01:37:42 BST 2019 : Done
Fri  5 Apr 01:37:42 BST 2019 : Starting download for /root/bin/remountfs_rw
Fri  5 Apr 01:37:43 BST 2019 : Done
Fri  5 Apr 01:37:43 BST 2019 : Installing archive module scripts
Fri  5 Apr 01:37:43 BST 2019 : Starting download for /root/bin/verify-archive-configuration.sh
Fri  5 Apr 01:37:43 BST 2019 : Done
Fri  5 Apr 01:37:43 BST 2019 : Starting download for /root/bin/configure-archive.sh
Fri  5 Apr 01:37:44 BST 2019 : Done
Fri  5 Apr 01:37:44 BST 2019 : Starting download for /root/bin/archive-clips.sh
Fri  5 Apr 01:37:44 BST 2019 : Done
Fri  5 Apr 01:37:44 BST 2019 : Starting download for /root/bin/connect-archive.sh
Fri  5 Apr 01:37:45 BST 2019 : Done
Fri  5 Apr 01:37:45 BST 2019 : Starting download for /root/bin/disconnect-archive.sh
Fri  5 Apr 01:37:45 BST 2019 : Done
Fri  5 Apr 01:37:45 BST 2019 : Starting download for /root/bin/write-archive-configs-to.sh
Fri  5 Apr 01:37:46 BST 2019 : Done
Fri  5 Apr 01:37:46 BST 2019 : Starting download for /root/bin/archive-is-reachable.sh
Fri  5 Apr 01:37:46 BST 2019 : Done
Fri  5 Apr 01:37:46 BST 2019 : Starting download for /root/bin/copy-music.sh
Fri  5 Apr 01:37:47 BST 2019 : Done

After the last "Done", it gets stuck.

Sam

marcone, 2019-04-05:
OK, the next step after that is running the `verify-archive-configuration.sh` and `configure-archive.sh` scripts that it just downloaded.

I didn't add any additional logging to those, yet, but you can run them manually after sourcing the conf file, i.e:
```
. /boot/teslausb_setup_variables.conf
/root/bin/verify-archive-configuration.sh
/root/bin/configure-archive.sh
```

SamGabbay, 2019-04-05:
@marcone I ran both of the commands and here is what I got. No devices coming up as mounting though.

root@teslausb:~# ./bin/verify-archive-configuration.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
hping3 is already the newest version (3.a2.ds2-7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Verifying that the archive server 192.168.7.241 is reachable...
The archive server is reachable.
Verifying that the archive share is mountable...
Mount command-line: 
mount -t cifs //192.168.7.241/TeslaCam /tmp/archivetestmount -o vers=3.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
STOP: The archive couldn't be mounted with CIFS version 3.0. Try specifying a lower number for the CIFS version like this:
  export cifs_version=2.1
Other versions you can try are 2.0 and 1.0
root@teslausb:~# ./bin/configure-archive.sh
Configuring the archive...
Configured the archive.

marcone, 2019-04-05:
It's failing to mount your CIFS share:
```
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
STOP: The archive couldn't be mounted with CIFS version 3.0. Try specifying a lower number for the CIFS version like this:
```
Maybe a typo in the  authentication credentials for your CIFS server? Check that the credentials file has the correct info. Otherwise, if the CIFS version is the problem, try copy/pasting that mount command line, but substituting 2.0 or 1.0 instead of 3.0

SamGabbay, 2019-04-05:
When typing in my IP Address should I be putting smb:// before it or just the IP?

marcone, 2019-04-05:
Just the DNS name or IP

SamGabbay, 2019-04-05:
Ok. I did the manual install. I got the "All done" at the end and the archive server is running. When a file is created on the CAM drive, nothing is synced to the samba server. Is there a way to force/check the uploads?

marcone, 2019-04-05:
The files are only backed up after re-connecting to wi-fi, not continuously. E.g. you go for a drive, which disconnects from wi-fi, then come back home, which reconnects.
This can be simulated by touching /tmp/archive_is_unreachable

lapean111, 2019-04-08:
I have this same issue, can't get past creating backing disk files.

marcone, 2019-04-08:
@lapean111 please attach/paste the setup log (/boot/teslausb-headless-setup.log)

lapean111, 2019-04-08:
@marcone this is with the manual setup.  I have never been able to get the headless setup to finish.

marcone, 2019-04-08:
Sorry, you're on your own then :)

marcone, 2019-04-12:
Closing this for now. If you're still seeing setup get stuck, please open a new bug, with logs.

closed 2019-04-12 by marcone