marcone/teslausb #3: How to update scripts?

leofaoro, 2019-04-03:
What is a good method to update the scripts as new commits are added? I don’t want to have to start over each time.
Thanks

marcone, 2019-04-03:
There isn't an easy way to do this currently, but that's a good idea.

marcone, 2019-04-20:
I've tweaked various scripts so that setup-teslausb can be run again without ill effect. When it runs, it will re-download the other scripts it uses. It's not a complete self-updating system, but that's not my goal for this project.

closed 2019-04-20 by marcone

THX723, 2020-12-11:
@marcone Just so I fully understand, simply running `setup-teslausb` will pull all the latest commits up to that point in time?

marcone, 2020-12-11:
@THX723 https://github.com/marcone/teslausb/wiki/FAQ#q-how-do-i-update-to-the-latest-version-of-teslausb

THX723, 2020-12-11:
@marcone  I did see that but wasn't sure if that meant upgrade to the point of latest release or commits. It's clear now. Thanks as always.