marcone/teslausb #4: Syncing music CIFS share

thijsdejong, 2019-04-03:
Hi,

In the Readme it says it supports syncing music from CIFS share. But later in the readme the only way to sync the music that is discussed is unplugging the pi from the car and into the computer. Can you add a guide to setup syncing from a CIFS share?

marcone, 2019-04-03:
It does say that that "the instructions below may be outdated" :)

There's a brief description in the `teslausb_setup_variables.conf.sample` file. Basically, you define the "musicsharename" variable to point at the cifs share and folder. The share currently must exist on the same server as the one where recordings will be backed up, and use the same credentials.
For example, I have all my music on a share called 'Music', and on that share I created a folder called 'CarMusic' where I copied all the songs that I want to have available in the car. I use `export musicsharename=Music/CarMusic` in the setup file.

I'll update the readme to include this.

closed 2019-04-03 by thijsdejong

lapean111, 2019-04-08:
How does the syncing job work?  Does it just sync the entire structure when it connects?  For instance, if I delete an album from the Pi, will it grab it again from the share next time it connects?

marcone, 2019-04-08:
Every file that's in the share but not on the Pi will be copied from the share to the Pi. If you want to delete something from the Pi permanently, you'll have to delete it from both the Pi and the share.

lapean111, 2019-04-08:
@marcone Thanks!

I must have a corrupt file or something, my sync is hanging after a few artists.  Are there logs for this sync job somewhere?

marcone, 2019-04-08:
It doesn't log each file, just a summary at the end.
(check /mutable/archiveloop.log)

lapean111, 2019-04-08:
I can see all of my artists, albums, and songs in the musicarchive directory under mnt, but not all the music appears on my car, and it doesn't seem like the pi has grabbed all the files from my server.  Is there a way to reset the sync?

marcone, 2019-04-08:
Files are copied every time the Pi reconnects to wifi.
You can simulate that by touching /tmp/archive_is_unreachable.
Then, check /mutable/archiveloop.log, it should print the number of files copied and the number of files skipped because they already existed on the Pi.

XmemphistoX, 2019-05-17:
This seems to only sync smaller music files for me.  My larger flac dont come across.