marcone/teslausb #6: Getting "The MicroSD card is too small" in teslausb-20190402

richardyc, 2019-04-03:
I got the error 

> STOP: The MicroSD card is too small.

 error during the one-step setup for both _teslausb-20190402_ and _teslausb-20190331_. The one-step setup halts after 4th stage(by number of light flashes). 
When I tried, to manually install using `sudo -i` and then run `/etc/rc.local`, it produced the following log:

> pi@teslausb:~ $ sudo -i
> 
> SSH is enabled and the default password for the 'pi' user has not been changed.
> This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.
> 
> root@teslausb:~# /etc/rc.local
> My IP address is 192.168.92.178
> dos2unix: converting file /root/teslausb_setup_variables.conf to Unix format...
> Detecting whether to update wpa_supplicant.conf
> dos2unix: converting file /root/teslausb_setup_variables.conf to Unix format...
> Starting setup.
> Updating package index files...
> Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
> Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
> Reading package lists... Done
> Verifying that the requested configuration is valid...
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100   690  100   690    0     0   2463      0 --:--:-- --:--:-- --:--:--  2473
> Downloaded /tmp/verify-configuration.sh ...
> Verifying that there is sufficient space available on the MicroSD card...
> STOP: The MicroSD card is too small.

I have also attached my _teslausb-healess-setup.log_. It seems like its the same error. I used Etcher to flash the image, set to 65% camera percentage, and have a 64GB microSD card.
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/3039333/teslausb-headless-setup.log)

Greatly appreciate your effort! Thanks!

marcone, 2019-04-04:
That usually means the sd card isn't partitioned right (i.e. all the space is already taken up by existing partitions).
If you try it with the 'diagnostics' branch, it should print some more info.

marcone, 2019-04-08:
Closing for now. If you still see this issue with the latest release, please attach logs (main-dev includes more verbose setup-logs now).

closed 2019-04-08 by marcone