marcone/teslausb #8: USB drive not showing up on PC or car

mattster98, 2019-04-06:
Just built this from scratch and while the log says it's connected to USB host and dmesg shows that it is connected (at least electrically) the drive does not show up.  The backing file is fine, can mount it and see directories.  Any ideas?

marcone, 2019-04-06:
What does `fdisk -l /backingfiles/cam_disk.bin` say?
Does the drive show up if you `modprobe -r g_mass_storage` followed by `modprobe g_mass_storage` ?

mattster98, 2019-04-06:
OK - it's a new day and this morning the drive will show up on my Chromebook.  Maybe just my Windows machine doesn't like it?  Maybe it fully baked overnight? ;)  Who knows.

Confirmed everything looked good and took it out to the car (Model X - 2019.8.5) - several observations.

After initial attempts failed as before, I plugged my go-to USB thumb drive in which is what I'm using in the meantime and it didn't work either.  Rebooted the MCU and it started working again - tried the USB stick in both USB ports to make sure they acted similar.  They did.

Plugged in the teslausb raspi (both pi ports into both tesla ports) and waited.. got the two-flash pattern and waited, and still no gray or red dot camera icon.  Switched back to the USB stick and it wasn't recognized either.  Seems like something about the raspi is freaking it out.  Rebooted it again.  Also tried powering the raspi from a 12v accessory adapter and only having the usb connection to the car's usb port.  No difference.

Logs with requested commands below.

[192.168.10.119_20190406102006.txt](https://github.com/marcone/teslausb/files/3050635/192.168.10.119_20190406102006.txt)

lapean111, 2019-04-06:
I have this issue too.  Mine will show up and record once, but the minute it loses power when the Tesla goes to sleep, it won't come back after it boots.  If I run fsck, it will work again once, but fail again after it reboots.

marcone, 2019-04-06:
With the Pi attached to the host of your choice (whether PC, Tesla or otherwise), can you ssh into the Pi, then `touch /tmp/archive_is_unreachable` and look at /mutable/archiveloop.log ? That should show the drive being disconnected from mass storage, fsck being run on the image, then the drive being published as mass storage again.

For what it's worth, I use a battery (a.k.a portable phone charger) connected to the power port of the Pi to provide continuous power. I also made a "data-only" USB cable by putting a bit of tape over the 5V lead of the USB cable, because I noticed the car pulling power from the battery through the Pi when the car was off (the battery showed a load of 400mA with the car off, whereas normally the Pi only used about 100-150mA)

marcone, 2019-04-07:
I noticed that I sometimes, but not always, get kernel stack dumps in dmesg when connecting the Pi to a host. This is apparently a kernel issue, and running `rpi-update` to update the kernel to 4.19 appears to have fixed that. Might be a worth a try for you, @mattster98, @lapean111

mattster98, 2019-04-07:
Isn't that part of the setup script already?  Pretty sure I watched it
update..

marcone, 2019-04-07:
The setup scripts run `apt-get upgrade`, but that only updates packages, not the kernel.
You can check which kernel you're running with `uname -r`. For me, it was running 4.14.98+ before, and now it's running 4.19.32+

mattster98, 2019-04-07:
You're absolutely right.  It was 4.14, not 4.19 after rpi-update.  I guess
some part of the setup script looked similar.

I'll do some more testing with a single usb cable later.

mattster98, 2019-04-07:
Car isn't seeing the usb drive after rpi-update.  Will need to diagnose.. :(

injured10, 2019-04-07:
ssh pi@teslausb.local
and running 
sudo rpi-update 
made the drive visible for me

mattster98, 2019-04-07:
I got mine to show for a moment but then went away.. nothing I can see in
logs to explain.  Did want to ask if this is what you see..  specifically
the first "LUN" line with "no medium".  From dmesg.

[  767.495316] Mass Storage Function, version: 2009/09/11
[  767.495343] LUN: removable file: (no medium)
[  767.495571] LUN: removable file: /backingfiles/cam_disk.bin
[  767.495585] Number of LUNs=1
[  767.503748] g_mass_storage gadget: Mass Storage Gadget, version:
2009/09/11
[  767.503774] g_mass_storage gadget: g_mass_storage ready
[  767.503794] dwc2 20980000.usb: bound driver g_mass_storage
[  797.201145] dwc2 20980000.usb: new device is high-speed
[  797.274254] dwc2 20980000.usb: new device is high-speed
[  797.332336] dwc2 20980000.usb: new address 6
[  797.344770] g_mass_storage gadget: high-speed config #1: Linux
File-Backed Storage
root@teslausb:~#

marcone, 2019-04-08:
I see the same "no medium" line, both in the working and non-working case.

Non-working case (turns out I had a bad, perhaps charge-only, cable):
```
[   33.935768] Mass Storage Function, version: 2009/09/11
[   33.935788] LUN: removable file: (no medium)
[   33.935961] LUN: removable file: /backingfiles/music_disk.bin
[   33.936071] LUN: removable file: /backingfiles/cam_disk.bin
[   33.936079] Number of LUNs=2
[   33.939169] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   33.939187] g_mass_storage gadget: g_mass_storage ready
[   33.939202] dwc2 20980000.usb: bound driver g_mass_storage
```

working:
```
[   32.452060] Mass Storage Function, version: 2009/09/11
[   32.452078] LUN: removable file: (no medium)
[   32.452236] LUN: removable file: /backingfiles/music_disk.bin
[   32.452342] LUN: removable file: /backingfiles/cam_disk.bin
[   32.452351] Number of LUNs=2
[   32.462678] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   32.462698] g_mass_storage gadget: g_mass_storage ready
[   32.462713] dwc2 20980000.usb: bound driver g_mass_storage
[   32.568266] dwc2 20980000.usb: new device is high-speed
[   33.224901] dwc2 20980000.usb: new device is high-speed
[   33.326938] dwc2 20980000.usb: new device is high-speed
[   33.524499] dwc2 20980000.usb: new address 20
[   33.538451] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
```

lapean111, 2019-04-08:
No matter what I do, I can't get the drive to reliably show up for the Tesla.  I have the pi permanently powered with a battery, so it's not hard rebooting ever.  I can manually get the camera drive to show up by doing a modprob -r followed by a modprobe on the g_mass_storage.

I can't think of a way to schedule this without losing some recording time.

Anyone have any ideas??

marcone, 2019-04-09:
If your Pi is already continuously-powered, then the only other things that come to mind are (1) check your cable is good and (2) run rpi-update to update to kernel 4.19.
Does it work if you plug into a PC (using the same cable) ?

lapean111, 2019-04-09:
I was doing some more tests yesterday, and have determined that keeping the pi always powered may be causing some issues.  I have switched it now to be powered from the cig lighter, and have a data only cable going to the USB port on the Tesla.  When I pull into the garage, and the pi connects to the WIFI, the drive disconnects from the Tesla.  When I start my next drive, it takes a minute or two for the drive to show up while the pi is booting, but seems to always work.

Not sure what the perfect way to do this is...

Rykemapo, 2019-04-12:
@marcone How are you able to run rpi-update with the read-only filesystem?

marcone, 2019-04-12:
Run remountfs_rw first, then rpi-update

Rykemapo, 2019-04-12:
I don't see that script anywhere

marcone, 2019-04-12:
The headless setup script downloads it to /root/bin/
If you use manual setup, you can either download it yourself, or just do "mount -o remount,rw /; mount -o remount,rw /boot"

marcone, 2019-04-12:
@lapean111 re: your comment above, I've been running mine continuously powered for 3 days now without issue.

Rykemapo, 2019-04-12:
@marcone Found it. Thanks.

rikbrown, 2019-04-13:
I'm also experiencing this problem. 

Running `modprobe -r g_mass_storage; modprobe g_mass_storage` makes it reappear.

I checked `/etc/modules` and `g_mass_storage` wasn't there, so I added it.

Now on startup I see this in `dmesg`:

```
[    3.497541] Mass Storage Function, version: 2009/09/11
[    3.497570] LUN: removable file: (no medium)
[    3.523594] lun0: unable to open backing file: /backingfiles/cam_disk.bin
[    3.523779] g_mass_storage 20980000.usb: failed to start g_mass_storage: -2
```

Continuing to debug,

mattster98, 2019-04-13:
Did you run the setup script?  Sounds like you're missing a lot.

rikbrown, 2019-04-13:
I followed [the one step](https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md) guide. It went through everything, sequences of flashes etc, and was flashing the green light as described.

Oddly `/root/bin/remountfs_rw` was missing, which makes me wonder if something went wrong.

Here's the `teslausb-headless-setup.log` output. Let me know if I can provide anything else:

```
Sat  6 Apr 23:16:33 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat  6 Apr 23:16:33 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat  6 Apr 23:16:33 BST 2019 : Rebooting...
Sat  6 Apr 23:16:58 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat  6 Apr 23:16:58 BST 2019 : Grabbing main setup file.
Sat  6 Apr 23:17:00 BST 2019 : Starting setup.
Sat  6 Apr 23:17:00 BST 2019 : Updating package index files...
Sat 13 Apr 22:39:25 BST 2019 : Verifying that the requested configuration is valid...
Sat 13 Apr 22:39:26 BST 2019 : Downloaded /tmp/verify-configuration.sh ...
Sat 13 Apr 22:39:26 BST 2019 : Verifying that there is sufficient space available on the MicroSD card...
Sat 13 Apr 22:39:26 BST 2019 : There is sufficient space available.
Sat 13 Apr 22:39:35 BST 2019 : Downloading additional setup scripts.
Sat 13 Apr 22:39:35 BST 2019 : Downloaded /tmp/create-backingfiles-partition.sh ...
Sat 13 Apr 22:39:36 BST 2019 : Downloaded /tmp/create-backingfiles.sh ...
Sat 13 Apr 22:39:36 BST 2019 : Downloaded /tmp/make-root-fs-readonly.sh ...
Sat 13 Apr 22:39:37 BST 2019 : Downloaded /root/configure.sh ...
Sat 13 Apr 22:39:37 BST 2019 : Fixing the modules-load parameter in /boot/cmdline.txt...
Sat 13 Apr 22:39:37 BST 2019 : Fixed cmdline.txt.
Sat 13 Apr 22:39:46 BST 2019 : Starting to create backing files partition...
Sat 13 Apr 22:39:46 BST 2019 : Checking existing partitions...
Sat 13 Apr 22:39:46 BST 2019 : Modifying partition table for backing files partition...
Sat 13 Apr 22:39:47 BST 2019 : Modifying partition table for mutable (writable) partition for script usage...
Sat 13 Apr 22:39:47 BST 2019 : Writing updated partitions to fstab and /boot/cmdline.txt
Sat 13 Apr 22:39:47 BST 2019 : Formatting new partitions...
Sat 13 Apr 22:40:24 BST 2019 : Mounting the partition for the backing files...
Sat 13 Apr 22:40:24 BST 2019 : Mounted the partition for the backing files.
Sat 13 Apr 22:40:24 BST 2019 : Creating backing disk files.
Sat 13 Apr 22:40:24 BST 2019 : create-backingfiles: starting
Sat 13 Apr 22:40:24 BST 2019 : create-backingfiles: Allocating 114471676K for /backingfiles/cam_disk.bin...
Sat 13 Apr 22:40:28 BST 2019 : create-backingfiles: Creating filesystem with label 'CAM'
Sat 13 Apr 22:40:30 BST 2019 : create-backingfiles: updated /etc/fstab for /mnt/cam
Sat 13 Apr 22:40:30 BST 2019 : create-backingfiles: created camera backing file
Sat 13 Apr 22:40:34 BST 2019 : create-backingfiles: done
Sat 13 Apr 22:40:43 BST 2019 : Main setup completed.
Sat 13 Apr 22:40:43 BST 2019 : calling configure.sh
Sat 13 Apr 22:40:43 BST 2019 : configure: /root/configure.sh starting with REPO=marcone, BRANCH=main-dev, ARCHIVE_SYSTEM=none, INSTALL_DIR=/root/bin
Sat 13 Apr 22:40:43 BST 2019 : configure: Skipping archive configuration.
Sat 13 Apr 22:40:44 BST 2019 : make-root-fs-readonly: start
Sat 13 Apr 22:40:44 BST 2019 : make-root-fs-readonly: Removing unwanted packages...
Sat 13 Apr 22:41:44 BST 2019 : make-root-fs-readonly: Installing ntp and busybox-syslogd...
Sat 13 Apr 22:42:39 BST 2019 : make-root-fs-readonly: Configuring system...
Sat 13 Apr 22:42:39 BST 2019 : make-root-fs-readonly: Mounting the mutable partition...
Sat 13 Apr 22:42:39 BST 2019 : make-root-fs-readonly: Mounted.
Sat 13 Apr 22:42:39 BST 2019 : make-root-fs-readonly: Moving fake-hwclock data
Sat 13 Apr 22:42:39 BST 2019 : make-root-fs-readonly: done
Sat 13 Apr 22:42:39 BST 2019 : Upgrading installed packages...
Sat 13 Apr 22:46:49 BST 2019 : All done.
```

rikbrown, 2019-04-13:
I'm going to flash it and run the one-step setup again to see if I can reproduce this.

rikbrown, 2019-04-13:
I just noticed `configure: Skipping archive configuration.` in the previous log. Thought I made a typo but think I worked it out.

In your example instructions you [give some copy+pasta](https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md#for-headless-automatic-setup) for setting up the `/boot/teslausb_setup_variables.conf`. That doesn't include the evidently required `export ARCHIVE_SYSTEM=cifs`. On my second attempt I copied the sample file and edited it, which did include this.

I think this is the problem? Happy to submit a PR to modify your docs, but maybe you meant to default it if not specified?

marcone, 2019-04-14:
Yes, the headless setup scripts have required ARCHIVE_SYSTEM to be set for a while (since November 2018, in the upstream branch), but it looks like the documentation wasn't updated to reflect that. I just updated it.

travelton, 2019-04-25:
I know this may sound retarded, but I was having the same issue until I realized there's two USB ports on the Pi. Need to use the data port, not the power only port. DOH!

mattster98, 2019-04-25:
Worth pointing out, but yes I've confirmed I'm in the USB port. :)

clean-ev, 2019-04-26:
I have been trying to get the car to see this and am completely unsuccessful. I posted in another thread here - https://github.com/marcone/teslausb/issues/9#issuecomment-487080919

clean-ev, 2019-04-26:
@marcone - setting up multiple (4) RPi’s I came across yet another issue wherein I had to add cifs_archive=2.0 as by default it uses 3.0. I tried 2.1 and my CIFS share did not mount, 2.0 was the setting that helped me get past the drive mapping.

marcone, 2019-04-26:
You're the 2nd or 3rd person that mentions having to tweak the CIFS version. Might be worthwhile to have it try all the possible versions automatically. I'm certainly open to accepting a PR for that.

bbartizek, 2019-04-26:
> I have been trying to get the car to see this and am completely unsuccessful. I posted in another thread here - [#9 (comment)](https://github.com/marcone/teslausb/issues/9#issuecomment-487080919)

I'm having this same problem.  Plugged into my PC, RPi boots normally, CAM partition shows as a USB storage drive, etc.  Plugged into a raw power-only USB, RPi boots normally and enters the archive loop (consistent LED flashes).  Plugged into the front ports on my dash, and it powers on, lots of LED activity, but never settles into the archive loop, and never enables the dashcam on the car.

marcone, 2019-04-26:
@bbartizek while it's in that state (the "never settles into the archive loop" state), ssh into the pi and grab /mutable/archiveloop.log and the output of dmesg.

bbartizek, 2019-04-26:
I fully read through the other thread, and tried the dual power source suggestion.  I have the same USB-A addon board, which used to work fine in the past, but I plugged a battery into the power micro-USB on the RPi, and got it to actually boot and show the dashcam.  Once it was booted, removing the secondary power had no effect and it stayed working.  I then tried it with the power port plugged into the splitter I have for my Nomad charger, and it successfully booted the RPi and entered the archive loop, but never started the dashcam in the car, so presumably something went wrong with the filesystem and it didn't correct itself.

So the root cause of the RPi never booting properly does seem to be that the power is, for some reason, insufficient/inconsistent/etc. when only connecting the data port to the car.

I'm not sure why the dashcam stopped working though.

marcone, 2019-04-26:
The rpi unpublishes the cam drive when it transitions from "not connected to wi-fi" to "connected to wi-fi". It then backs up whatever needs backing up, and published the cam drive again, while still connected to wi-fi. The snippet of archiveloop.log you included shows it publishing the cam drive and then waiting to become disconnected from wi-fi again, so that looks OK.

Your dmesg on the other hand doesn't look right at all. It should print a few messages when connected, not the same thing over and over like yours does.

On my car it works with a single USB cable (or at least it did last time I tried a few weeks ago, I normally have it on permanent power pulled from the OBD2 port), so it seems like this is car-specific, with some cars not providing enough power to the USB ports for the rpi to run reliably.

bbartizek, 2019-04-26:
I was hoping that when I left the house, it would lose WiFi, remount cam, and things would work properly, but no such luck.  It did revert back to the repeating single blink loop, but the dashcam in the car never activated.  Not exactly sure how to trouble sheet from here since I can't connect to the RPi if it's out of WiFi range, lol.

wnichols85, 2019-04-28:
I commented out the following lines in the /root/bin/archiveloop script so I could keep the drives mounted while connected to WIFI so I could troubleshoot as well as record while in my drive way.   Just comment out the umount commands and I confirmed on reboot that the mount points (run "df -ah" to see) were still there

function archive_teslacam_clips () {
  log "Starting..."
  ensure_cam_file_is_mounted
  fix_errors_in_cam_file
  /root/bin/archive-clips.sh
  #unmount_cam_file
}
function copy_music_files () {
  log "Starting..."
  ensure_music_file_is_mounted
  fix_errors_in_music_file
  /root/bin/copy-music.sh
  #unmount_music_file
}

Now to see if I can just get this working as I just got the car a few days ago and no idea how it's supposed to work.  If i'm missing the point of the script and I just broke it please let me know.

marcone, 2019-04-28:
If you leave the drives mounted, then both the Pi and the Tesla will be accessing the filesystem at the same time, which is likely going to confuse one or both, and/or lead to filesystem corruption.

marcone, 2019-04-28:
What you probably want to do instead is, once the pi is plugged into the car, booted up, and you're ssh'd into the pi, run `modprobe -r g_mass_storage` (this is equivalent to unplugging the drive from the car), followed by `modprobe g_mass_storage` (equivalent to plugging it in to the car), then check the output of `dmesg`.

wnichols85, 2019-04-28:
I'm getting the following in dmesg

[   34.682879] CIFS VFS: Error -104 sending data on socket to server
[   34.682945] CIFS VFS: cifs_mount failed w/return code = -104
[   35.613853] CIFS VFS: cifs_mount failed w/return code = -112
[   35.844321] random: crng init done
[   35.844346] random: 7 urandom warning(s) missed due to ratelimiting
[   42.158478] Mass Storage Function, version: 2009/09/11
[   42.158500] LUN: removable file: (no medium)
[   42.158662] LUN: removable file: /backingfiles/music_disk.bin
[   42.158774] LUN: removable file: /backingfiles/cam_disk.bin
[   42.158783] Number of LUNs=2
[   42.167177] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   42.167195] g_mass_storage gadget: g_mass_storage ready
[   42.167210] dwc2 20980000.usb: bound driver g_mass_storage
[   42.868862] CIFS VFS: error -9 on ioctl to get interface list
[   43.207047] CIFS VFS: error -9 on ioctl to get interface list
[   51.830123] Mass Storage Function, version: 2009/09/11
[   51.830142] LUN: removable file: (no medium)
[   51.830302] LUN: removable file: /backingfiles/music_disk.bin
[   51.830399] LUN: removable file: /backingfiles/cam_disk.bin
[   51.830409] Number of LUNs=2
[   51.830744] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   51.830757] g_mass_storage gadget: g_mass_storage ready
[   51.830772] dwc2 20980000.usb: bound driver g_mass_storage

If I have sentry mode on will I see a camera icon on the screen?  Will having my phone key near by interfere with testing recordings by walking around the car?

marcone, 2019-04-28:
Here's what I see after `modprobe -r g_mass_storage` followed by `modprobe g_mass_storage`:

```
[519360.546295] Mass Storage Function, version: 2009/09/11
[519360.546322] LUN: removable file: (no medium)
[519360.546547] LUN: removable file: /backingfiles/music_disk.bin
[519360.546686] LUN: removable file: /backingfiles/cam_disk.bin
[519360.546701] Number of LUNs=2
[519360.547163] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[519360.547181] g_mass_storage gadget: g_mass_storage ready
[519360.547201] dwc2 20980000.usb: bound driver g_mass_storage
[519360.556099] dwc2 20980000.usb: new device is full-speed
```

this is with the car having been sitting in the garage without me using it for a while. The car's USB ports are disabled/unpowered at this point, but the pi is on because it has permanent power.
When I remotely turn on sentry mode, the car's usb ports become enabled, which results in these additional lines in dmesg:

```
[519587.481622] dwc2 20980000.usb: new device is high-speed
[519587.637275] dwc2 20980000.usb: new device is high-speed
[519587.734891] dwc2 20980000.usb: new address 9
[519587.750439] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
```
If you don't see those lines in dmesgs, then the pi and the car are not seeing each other. Check that you're using the correct port in the pi, and that you don't have a power-only USB cable.

wnichols85, 2019-04-28:
Turns out I had a charge only USB cable (used one that I found).  Switch the cable to one that I know works and we are in business.  Thanks!

mattster98, 2019-04-29:
I just got updated to 2019.12.1.1 (MX) and everything is back to normal.
 Single cable on the USB port and boots and starts recording.

Will update later after it's seen power cycles and saved recordings.

mattster98, 2019-04-30:
Quick drive this morning and it booted up fine and connected before I
backed out.  Saved a recording along the way and it was archiving when I
got out of the car back at home.  Looks like I'm back in business.

bbartizek, 2019-05-01:
So I managed to fix the issue I was having previously where the RPi itself would seem to be working, but the car couldn't see it.  In the other thread I read, the ports on the RPi board were described as"

> the first one ("USB") goes to the USB port in the car, the second port on the board ("PWR IN")

This seemed pretty straightforward, so I didn't spend more than half a second thinking about it.  Turns out, that was a mistake.  I have a USB-A connector addon board for my RPi, so the "first" port is clearly the one next to the USB-A connector, right?  Not the case.  The inboard port, which I had assumed was the "second" port, is actually the data port, and the outboard port is power only.  So what was happening with my RPi, is that it was connected on the data port to both the front USB of the car, and the USB port on my splitter, which seems to confuse the hell out of everything, and make it not visible to the car.  Once I switched my secondary power to the outboard port, the car saw the drive almost immediately and the dashcam is working again.  User error ftw.

clean-ev, 2019-05-02:
Got 12.1.1 on Model 3. Didn’t change a thing from the original build using RPiZeroW. Simply plugged in and viola, got the camera icon on the screen. Seems it was something with the software release of the past that made us all go crazy with this cable swapping rebuilding.
Finally it is working and hope that Tesla doesn’t break it again

kewvention, 2019-05-09:
so i did manage to install it successfully, but whenever i plug the rpi zerow into the car, i see the quick 1 second blinks, but the dashcam icon never comes. i plugged the usb cable into the port labeled USB on the pi and the left port in the car.  the cable itself, if i connect it to the pc, i see the boot partition as well. so im kind of lost.

marcone, 2019-05-09:
When you said "the cable itself, if i connect it to the pc, i see the boot partition as well", did you mean "the **card** itself ..." ?
Because you definitely should not be seeing the boot partition when the card is plugged in to the pi, and the pi is plugged in to the PC.

kewvention, 2019-05-09:
> 
> 
> When you said "the cable itself, if i connect it to the pc, i see the boot partition as well", did you mean "the **card** itself ..." ?
> Because you definitely should not be seeing the boot partition when the card is plugged in to the pi, and the pi is plugged in to the PC.

sorry yes. i meant that i see the boot partition when using a microsd card adapter .however, if i plug the pi into the pc and sdcard into the pi, i dont see any partition.this should be correct right.

marcone, 2019-05-09:
If you plug the pi into the PC and let it boot up, the PC should see the cam and music drives, just like the car would. Should take under a minute for those to show up after plugging in the pi.
If you're not seeing those drives appear on the PC, ssh into the pi and look at the various logs: /boot/teslausb-headless-setup.log, /mutable/archiveloop.log, the output of `dmesg`

kewvention, 2019-05-09:
> If you plug the pi into the PC and let it boot up, the PC should see the cam and music drives, just like the car would. Should take under a minute for those to show up after plugging in the pi.
> If [...]

ah ok. so i tried another cable and saw the CAM folder come up since i dedicated 100% to it. I will try the car shortly and report back on it.

acraigl, 2019-05-11:
I think something's not quite right?
The TESLAUSB_SETUP_FINISHED and WIFI_ENABLED files are present, and I see the TeslaCam drive when I connect it to my PC and Car, so I think I'm good, but the device is flashing twice, repeatedly.  It also seems to take a while to appear, definitely more than a minute.

I'm trying to sudo to the pi and it's asking me for the password? I don't recall setting that in the .conf file. I can access SSH only by IP address, not hostname.

marcone, 2019-05-11:
Not sure either, since I don't know what you did. All I can say is that the prebuilt 20190505 image using CIFS works for me with a Model S with 2019.12.1.2
Does the rpi still flash its LED? Can you still ssh into it? If so what do the logs say?

acraigl, 2019-05-11:
I can SSH into it. LED is still flashing, and connecting to my PC seems to be fine. A bit of a noob here, sorry. What logs should I be looking for? Tried switching cables too, FWIW. Works consistently on my PC but not the car after that first load.

marcone, 2019-05-11:
If the LED is flashing then the archiveloop script is running, or at least run up until the point where it put the LED in flashing mode. So look at /mutable/archiveloop.log, and the output of `dmesg`

acraigl, 2019-05-11:
Here's a snippet from archiveloop.log:
`
-bash: /dev/loop0:: No such file or directory
pi@teslausb:/mutable $ Sat 11 May 20:21:32 BST 2019: Finished fsck on /mnt/cam.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Moving clips to archive...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Creating output directory ''
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: /mnt/cam/TeslaCam/SavedClips does not exist, skipping
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Moved 0 file(s), failed to copy 0, deleted 0.
-bash: syntax error near unexpected token `('
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Finished moving clips to archive.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Unmounting /mnt/cam...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Unmounted /mnt/cam.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Finished archiving.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: No music archive configured-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Connecting usb to host...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Connected usb to host.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Waiting for archive to be unreachable...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:23:06 BST 2019: Sleeping before retry...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:23:07 BST 2019: Retrying...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:24:41 BST 2019: Sleeping before retry...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:24:42 BST 2019: Retrying...
-bash: Sat: command not found
`
Seems like a lot of retry going on, Is that normal?
And dmesg:
`pi@teslausb:/mutable $ Sat 11 May 20:21:32 BST 2019: Finished fsck on /mnt/cam.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Moving clips to archive...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Creating output directory ''
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: /mnt/cam/TeslaCam/SavedClips does not exist, skipping
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Moved 0 file(s), failed to copy 0, deleted 0.
-bash: syntax error near unexpected token `('
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Finished moving clips to archive.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:33 BST 2019: Unmounting /mnt/cam...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Unmounted /mnt/cam.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Finished archiving.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: No music archive configured-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Connecting usb to host...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Connected usb to host.
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:21:34 BST 2019: Waiting for archive to be unreachable...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:23:06 BST 2019: Sleeping before retry...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:23:07 BST 2019: Retrying...
-bash: Sat: command not found
pi@teslausb:/mutable $ Sat 11 May 20:24:41 BST 2019: Sleeping before retry...
-bash: Sat: command not found
[... 281 lines cut ...]
[   35.349041] LUN: removable file: (no medium)
[   35.349201] LUN: removable file: /backingfiles/cam_disk.bin
[   35.349211] Number of LUNs=1
[   35.356735] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   35.356754] g_mass_storage gadget: g_mass_storage ready
[   35.356770] dwc2 20980000.usb: bound driver g_mass_storage
[   56.377467] FAT-fs (loop0): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   62.796031] loop: Write error at byte offset 1048576, length 512.
[   62.796055] print_req_error: I/O error, dev loop0, sector 0
[   62.796070] Buffer I/O error on dev loop0, logical block 0, lost async page write
[   62.807858] loop: Write error at byte offset 1049088, length 512.
[   62.807881] print_req_error: I/O error, dev loop0, sector 1
[   62.807895] Buffer I/O error on dev loop0, logical block 1, lost async page write
[   62.819727] loop: Write error at byte offset 48398336, length 512.
[   62.819751] print_req_error: I/O error, dev loop0, sector 92480
[   62.819765] Buffer I/O error on dev loop0, logical block 92480, lost async page write
[   63.059460] Mass Storage Function, version: 2009/09/11
[   63.059479] LUN: removable file: (no medium)
[   63.059635] LUN: removable file: /backingfiles/cam_disk.bin
[   63.059645] Number of LUNs=1
[   63.066578] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   63.066597] g_mass_storage gadget: g_mass_storage ready
[   63.066612] dwc2 20980000.usb: bound driver g_mass_storage
[  182.251365] CIFS VFS: Server 10.0.0.62 has not responded in 120 seconds. Reconnecting...
`

marcone, 2019-05-11:
Not sure what you did there, but it looks like you tried to execute the log file, or pasted its content into a shell, hence all the errors like: `-bash: Sat: command not found` (it's interpreting the "Sat 11 May" as a command to run)
The error messages in dmesg are similar to the ones reported in https://github.com/marcone/teslausb/issues/26, which I think should be fixed now, so try redoing setup again.

acraigl, 2019-05-12:
@marcone thanks... so the 20190505 is newer now? Just download that one again? I noticed it has the same size as before so just wanted to make sure.

marcone, 2019-05-12:
20190505 hasn't changed, you don't have to download it again. But if you *flash* it again and go through setup again, the setup process will download the latest version of the configuration scripts.

[13 more replies]

closed 2019-06-04 by marcone