marcone/teslausb #9: Tracking single-cord USB power issues

mattster98, 2019-04-06:
I wanted to create a separate issue on the off chance there's something we can do on the RasPi side to get a single usb cable to work again.

I used my USB power meter to see what happens when the RasPi reboots, and based on what the meter is showing, the power to the port itself is actually going away - no apparent overload that would seem to trigger it - max current is maybe 230mA - well below even very basic USB power limits.

I thought maybe Tesla added some logic to actively reject "usb attacks" given the visibility on the hacking stuff recently, so I copied the USB device ID, etc from a working USB thumb drive and it didn't seem to have any effect.  (I plugged it into the g_mass_storage.conf file)

Video of what I observed: http://imgur.com/gallery/qecn7Ke

All cables were plugged in the whole time - the power meter just saw a complete loss of power (or low enough to reboot).

Anybody else tried things to get a single cable to work again?  I finally got two cables working again so I'm at least back to "working".  Just hoping to improve further.

marcone, 2019-04-06:
Your link doesn't work

mattster98, 2019-04-06:
Weird - works for me.
[Another try, direct.](https://i.imgur.com/HYbance.gifv)

marcone, 2019-04-08:
I normally use the Pi with a battery so it can backup recordings without having to leave the car on, but tried with a single USB cable today, and it works without problems for me. Tried with both kernel 4.14 and 4.19.
2018 Model S, firmware 2019.8.5

mattster98, 2019-04-08:
Oh man.  If I'm dealing with USB cable issues I'm gonna lose my mind.

Rfvgyhn, 2019-04-25:
I'm also dealing with what I believe are power issues. If I use a single cable (I'm actually using the [usb a addon board](https://www.amazon.com/gp/product/B07BK2BR6C/)), it seems to bootloop before mass storage can connect. If I plug in a second cable the boot process finishes and the car sees the drive. I can then unplug the second cable and everything continues to run fine. Looking at dmesg, before it reboots, it makes it to
```
...
[   15.291893] brcmfmac: power management disabled
[   24.046855] FS-Cache: Netfs 'cifs' registered for caching
[   24.047796] Key type cifs.spnego registered
[   24.047846] Key type cifs.idmap registered
[   25.967676] random: crng init done
[   25.967702] random: 7 urandom warning(s) missed due to ratelimiting
[   27.245354] FAT-fs (loop0): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```
which is one line from the mass storage messages when it boots properly.

I've attempted to disable HDMI and blindly blacklist modules but nothing has changed.

`/usr/bin/tvservice -o` in `/etc/rc.local`

```
blacklist btbcm
blacklist hci_uart
blacklist bcm2835_v4l2
blacklist bcm2835_codec
blacklist bcm2835_mmal_vchiq
blacklist bcm2835_alsa
blacklist bcm2835_vc_sm_cma_probe
blacklist snd_bcm2835
blacklist vc_sm_cma
blacklist ipv6
```

Kernel 4.19
2018 Model 3, 2019.8.5

mattster98, 2019-04-25:
I'm still not having any luck even with two cords..  totally happy of
plugged into my PC but not the car

marcone, 2019-04-25:
@mattster98 which model car do you have? Could this be a 3 vs S thing?

marcone, 2019-04-25:
Oh I see in another issue that you have an X. No clue then.

mattster98, 2019-04-25:
Yeah.. I'm not alone but some people have it working.  Going to try another
raspi and SD to rule out hardware.

marcone, 2019-04-25:
I'd suspect cables more than the Pi or sd card. You used the same cables when connecting to the PC and that worked?

mattster98, 2019-04-25:
Yep, and tried half a dozen other cables in the car.

clean-ev, 2019-04-25:
I have similar setup with 4 RPi Zero W with USB add-on board @Rfvgyhn mentioned. I have 6 MicroSD cards from 16gb to 128gb.

2018 Model 3 LR-RWD

So far I have tried both 0406 build and then updating it with rpi-update to get to 4.19.
Also tried 0421 with 4.19 kernel baked in.

In any combination of SD card or one of the RPI ZeroW each time I can find the "CAM" mounted on Mac with "TeslaCam" folder.  However no matter what I do there will be no recording in the car. If I pull the RPI0W and plug into Mac I can see "CAM" again every single time.

Why is it not recording on either of the prebuilt images?
What do you all mean when you say second cable? As I am using add-on board I do not use any cables and plug USB-A directly to one of the USB ports

Last few lines of dmesg output
[   28.244097] random: 7 urandom warning(s) missed due to ratelimiting
[   29.055885] FAT-fs (loop0): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   33.608111] Mass Storage Function, version: 2009/09/11
[   33.608130] LUN: removable file: (no medium)
[   33.608292] LUN: removable file: /backingfiles/cam_disk.bin
[   33.608302] Number of LUNs=1
[   33.612651] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   33.612668] g_mass_storage gadget: g_mass_storage ready
[   33.612683] dwc2 20980000.usb: bound driver g_mass_storage
[   33.870113] dwc2 20980000.usb: new device is high-speed
[   33.925383] dwc2 20980000.usb: new device is high-speed
[   33.980843] dwc2 20980000.usb: new device is high-speed
[   34.036632] dwc2 20980000.usb: new device is high-speed
[   34.092086] dwc2 20980000.usb: new device is high-speed
[   34.147500] dwc2 20980000.usb: new device is high-speed
[   34.202646] dwc2 20980000.usb: new device is high-speed
[   34.258355] dwc2 20980000.usb: new device is high-speed
[   34.314149] dwc2 20980000.usb: new device is high-speed
[   34.406110] dwc2 20980000.usb: new address 9
[   35.136190] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage

fdisk shows - **this is with 64gb card - 100% cam usage**
pi@teslausb:~ $ fdisk -l /backingfiles/cam_disk.bin
Disk /backingfiles/cam_disk.bin: 53.7 GiB, 57679810560 bytes, 112655880 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6c64afc5

Device                      Boot Start       End   Sectors  Size Id Type
/backingfiles/cam_disk.bin1       2048 112655879 112653832 53.7G  c W95 FAT32 (LBA)

mattster98, 2019-04-26:
Using the bare Raspi Zero W there are two USB ports.  the first one ("USB")
goes to the USB port in the car, the second port on the board  ("PWR IN")
goes to either a second port in the car, a 12V adapter, or some are using
rechargeable battery banks to provide reliable power.

clean-ev, 2019-04-26:
@mattster98 - I tried using all 4 RPi0W bare with 2 separate MicroUSB cables (deemed working) has 5 at hand all of them working.

I found that plugging in cable to car and other end to USB on Zero fired up the RPi. I still plugged in power cable using a portable battery. Kept on changing cables and zero’s but at no time did the car see the zero as a USB to start recording. 

Using same hardware and cables I plugged them in to my Mac and “CAM” shows up each time. Also put them together with USB-A adapters I have and “CAM” shows up.

Wonder where the problem is - this setup worked from cimryan repo for the longest time, however since 9.x came I have tried to switch to this without any success.

Any other items I should check? @marcone

marcone, 2019-04-26:
The only change that I'm aware of in 9.x is that 9.x requires a partitioned drive. The current scripts do that, and the fdisk output you posted above shows yours is partioned correctly.

Rfvgyhn, 2019-05-01:
Just wanted to echo @mattster98 from #8. My M3 was updated to 2019.12.1.1 and I no longer need the extra usb cable to power my pi.

jasoncross, 2019-05-04:
My X is on 2019.12.1.1 4b1dd29 and does not work with one cable.

mattster98, 2019-05-04:
@jasoncross can you confirm which port on the pi you're using?

I just got bumped to 2019.12.1.2 and its still happy.

jasoncross, 2019-05-04:
> @jasoncross can you confirm which port on the pi you're using? I just got bumped to 2019.12.1.2 and its still happy.
> […](#)

The USB port (no the PWR port).

The setup all worked last year and then stopped with 9.X.  Thought I'd try again now that people were saying it worked again.  Re-built my SD card using the latest from marcone.  Connected to the computer the CAM drive shows up immediately.  No such luck in the car.

mattster98, 2019-05-04:
Try with two USB cables just to confirm the car is happy with it
otherwise.  I wish we had some level of debug or log viewer on the car..
this would be so much easier.

jasoncross, 2019-05-04:
Those who it works with again - are all running campercent at 100 or are they dividing the card?  Just looking for other possible data points as to why it works for some and not others.

mattster98, 2019-05-04:
Success with campercent 100

marcone, 2019-05-04:
@jasoncross I'm using
```
export camsize=32G
export musicsize=4G
```
myself. The specific size or whether you specify it as a percentage or an absolute size shouldn't matter though, especially if as you're saying it works when connected to a PC. Did you use the same USB cable for both car and PC, connected to the same USB port on the Pi?

Can you ssh into the Pi when it's plugged in to car? If so, take a look at the output of `dmesg`, it should have something like this at the end:
```
[519587.637275] dwc2 20980000.usb: new device is high-speed
[519587.734891] dwc2 20980000.usb: new address 9
[519587.750439] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
```

clean-ev, 2019-05-05:
> Those who it works with again - are all running campercent at 100 or are they dividing the card? Just looking for other possible data points as to why it works for some and not others.

I use 100% for camera recording.
Just tested various settings for Camera and music -
75% cam, 25% music
50% cam, 50% music
25% cam, 75% music

Though last is not worth it, I believe, still completed to test this thru. Worked on 64GB card I am using. Plan to start using 128gb or 256gb once @marcone gets the code to save all recordings out

tmhalen, 2019-07-11:
Same issue - Model X w/ 2019.20.4.3. If I plug in to both USB ports on the Pi, it boots fine, but one cable & it drops into a reboot loop. Tried numerous cables - right down to a 6 inch shorty. Nothing seems to make a difference.

mattster98, 2019-07-13:
Interesting - I'm on 2019.20.4.3 in my X and it's still working.

tmhalen, 2019-07-15:
Still fighting with this.  I bought a little USB power monitor which connects inline between the USB port and whatever device whose power consumption you're trying to measure.

I re-installed teslausb onto two SD cards.  Thought maybe something in the one I used previously was causing the issue.  One is a 128GB, the other is 32GB - both are Samsung cards.  The install for both was a little hinky.  Both cards seemed to get stuck.  When I looked at teslausb-headless-setup.log, it only got to the line: "reading config from /root/teslausb_setup_variables.conf" and then stopped.  Running /etc/rc.local seemed to restart everything and it finished OK.  Both cards have the same reboot issue in the car.

Connected to my PC, the PI draws no more than 190mA and generally 0.6 - 0.7W.  The watts jump a bit to a max of 0.95 when the LED is flashing quickly, which I believe is when its connecting to wifi.  As soon as that is done (the LED changes to double-flashes), the power goes right back down to 100-120mA / 0.6W.

When I connect it in the car, the numbers don't change, but it never gets to the point where it associates with the wifi.  At around that point (just judging by time), the power just drops completely - the Pi reboots, the USB power meter reboots and even the backlighting on the vehicle USB port flickers.  And then the cycle restarts.

By contrast, my current TeslaCam device, a USB-C m.2 SATA SSD will draw 240mA and 1.2W and will work just fine in the car.

It seems like something is going on just around the time the wifi is trying to connect that is making the USB port drop power momentarily.  Any chance the wifi signal is inducing something in the USB circuit that would cause it to kick off for a moment?

henley-regatta, 2019-07-17:
I'm seeing this too, on a new-build image. Model S running 2019.20.4.2, teslausb built from scripts 2019-07-17. Raspberry Pi Zero W, 64GB card set to 90% camera, 10% music.

I don't have much to add except the following observations:

1. The Pi boots successfully with a single cable in the USB slot powered either by a PC or by a battery pack. I can ssh into it, the SAMBA functionality works, and (if attached to PC) both CAM and MUSIC partitions are available and writeable.
2. When plugged into a front port on the Tesla, the Pi's LED does the rapid-flicker I associate with a boot, there's a pause, two slow blinks, then the sequence repeats _ad infinitum_.  `/mutable/archiveloop.log` is not updated.
3. When plugged into either front port on the Tesla _and_ a second cable is attached to the PWR socket on the Pi (either from the 2nd port on the Tesla or via an external power source, boot completes as expected (the flickering LED gives way to a rapid 2-flash sequence as wifi connects, archive happens, then reverts to the slower 2-flash sequence indicating normal operation with wifi). `/mutable/archiveloop.log` 
3. On at least one occasion I've had success in booting the Pi using 2 cables, and then unplugging the power cable from the Pi after steady-state has been reached. Normal operation continued. However, I've also had the thing crash and bootloop per the 2nd case above so this can't be seen as a particularly reliable work-around!

The behaviour in the 2nd case leads me to think that `/etc/rc.local` isn't getting called - the reboot happens before then. Sadly, whatever's present in `dmesg` is not captured anywhere in this case and cannot be used to help debug what's going on. 

Just speculation but this looks to me like one of the modules being loaded causes a power spike that freaks out whatever's protecting the Tesla's USB provider. Which is odd given that it'll happily supply at least 1000mA to a phone that's plugged in and charging....

clean-ev, 2019-07-17:
I am not sure if this helps as my setup is a bit different where I am using Raspberry Pi Zero W USB-A Addon Board from here - 
https://www.amazon.com/MakerFocus-Raspberry-Required-Connector-Protective/dp/B07BK2BR6C?th=1
I have not encountered any issues with any version of Tesla software including but not limited to beta if you are in early access program.

isentropik, 2019-07-30:
I get the same issues as @henley-regatta regarding the boot loop with a single cable. Mine will only work with dual cables.

henley-regatta, 2019-08-02:
I've just re-tried my test with the same board @clean-ev is using to integrate the power and USB slots on the Pi Zero to a single USB-A plug. Same results as before: boot looping before rc.local gets called. Which I wasn't expecting: the pogo-pins on the adapter board are split - power goes to the power port, data to the data port. Works really well on a laptop, but no dice on the Tesla.

ratcliffeg, 2019-08-05:
Just been trying this, exactly same results. Works perfectly on PC but as soon as I plug into the Model S it goes into a boot loop. Tried various cables, all the same.
Tried cable to the power usb only (to test) and actually came up fine! Wifi working and can SSH into.
Seems as soon as I plug it into the port for data it goes into the loop.

AnnihilaT, 2019-08-22:
Can confirm the same behavior.. for me i even need to pull the second power only cable from the rear seat USB port for it to NOT boot loop.

fat-tire, 2019-09-14:
I too am having the cam die after 15 minutes or so, at random times, including when driving.  Using the buster version of raspian.  Aside from starting to use both USB slots to power/data, I'm also now tying a couple things to minimize power usage:

In `/boot/config.txt`:

Disable audio (don't load snd_bcm2835) by commenting out:

     # dtparam=audio=on

And added this to underclock slightly:

     arm_freq=700
     core_freq=250

You can also play with some of the other settings like sdram_freq etc... but I'm not doing that.

To turn off the LEDs entirely, add

     dtparam=act_led_trigger=none
     dtparam=act_led_activelow=on

I know this isn't using that much power, but illuminating a closed area seems wasteful to me.

Then in `/etc/rc.local`, I'm turning off the HDMI display circuitry by adding

     /usr/bin/tvservice -o

and moving the cpu governor from "ondemand" -> "powersave":

      echo "powersave" > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor 

Thanks to [here](https://www.jeffgeerling.com/blogs/jeff-geerling/controlling-pwr-act-leds-raspberry-pi) and [here](https://www.jeffgeerling.com/blogs/jeff-geerling/raspberry-pi-zero-conserve-energy).  Anyway, let's see if it makes any difference.  I have a feeling using both USB ports will be the biggest factor, but I'm trying it all at once, and if it works, I'll try just one USB and see if that's enough power.

If anyone has any other thoughts/tips for reducing raspberry pi zero power, let me know.  I guess I could kill the wifi, but if I can help it I'd rather not.

fat-tire, 2019-09-14:
Hmm this doesn't seem to be working.  For fun I hooked up the pi to the computer and just tried dumping /dev/random into CAM/file.txt to see if it could handle it.

First I plugged in the pie, and saw from the host computer's dmesg:

```
[744298.343098] sd 7:0:0:0: [sdd] Attached SCSI removable disk
    ^ this is  MUSIC
[744298.344622] sd 7:0:0:1: [sde] Attached SCSI removable disk
    ^ this is  CAM
[744298.885889] FAT-fs (sdd1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[744298.899548] FAT-fs (sde1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
     ^  these two lines appeared on mounting.  Shouldn't teslacam be fscking them?

Then, this happened when i did:

     dd if=/dev/urandom of=/media/fattire/CAM/random.txt bs=1 count=10M

On the host computer (not the pi).  Next, on the host computer, I saw this:

[744367.325083] FAT-fs (sde1): error, fat_free_clusters: deleting FAT entry beyond EOF
[744367.325087] FAT-fs (sde1): Filesystem has been set read-only
     ^ WHAAAT?
[744382.514267] scsi_io_completion_action: 5 callbacks suppressed
[744382.514271] sd 7:0:0:1: [sde] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[744382.514274] sd 7:0:0:1: [sde] tag#0 Sense Key : Medium Error [current] 
[744382.514275] sd 7:0:0:1: [sde] tag#0 Add. Sense: Write error - auto reallocation failed
[744382.514278] sd 7:0:0:1: [sde] tag#0 CDB: Write(10) 2a 00 00 0c 98 00 00 00 f0 00
[744382.514278] print_req_error: 2095 callbacks suppressed
[744382.514280] print_req_error: I/O error, dev sde, sector 825344 flags 104001
[744382.618267] sd 7:0:0:1: [sde] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[744382.618269] sd 7:0:0:1: [sde] tag#0 Sense Key : Medium Error [current] 
[744382.618270] sd 7:0:0:1: [sde] tag#0 Add. Sense: Write error - auto reallocation failed
[744382.618271] sd 7:0:0:1: [sde] tag#0 CDB: Write(10) 2a 00 00 0c 98 f0 00 00 f0 00
[744382.618273] print_req_error: I/O error, dev sde, sector 825584 flags 104001

```
Then the CAM mount (on /dev/sde1) locked up So it's like... something is messed up here.  The above occurred no matter what the freq settings were.  the local linux computer is remounting read only.

No errors are appearing on the pi side.

sarathi12345, 2019-09-29:
> My X is on 2019.12.1.1 4b1dd29 and does not work with one cable.

Does it works with 2 cable?

drlauridsen, 2020-02-22:
Have this issue.. TMS100D 2020.4.1. RPI Zero W.
PC: connected with single cable to USB port, drives show after 30 sec
In car: a) Single cable (same cable) connected to RP USB port results in bootloop and hence drive never becomes available. b) Single cable to RP PWR port - boots as expected, but (naturally) no data connection/no drive. c) Two cables, one to RP USB and one to RP PWR, same as "a" - even if PWR is supplied from powerbank and not car. d) Two cables, but on the cable going to RP USB port, + is isolated with tape so acting as data only cable - works perfectly.

So something happens with the RP USB port and power on boot, that does not happen on RP PWR port. And it happens even when both ports have power.. What could be the reason for this behaviour?

jasoncross, 2020-02-22:
> Have this issue.. TMS100D 2020.4.1. RPI Zero W.
> PC: connected with single cable to RP USB port, drives show after 30 sec
> In car: a) Single cable (same cable) connected to RP USB port results in bootloop and hence drive never becomes available. b) Single cable to RP PWR port - boots as expected, but (naturally) no data connection/no drive. c) Two cables, one to RP USB and one to RP PWR, same as "a" - even if PWR is supplied from powerbank and not car. d) Two cables, but on the cable going to RP USB port, + is isolated with tape so acting as data only cable - works perfectly.
> 
> So something happens with the RP USB port and power on boot, that does not happen on RP PWR port. And it happens even when both ports have power.. What could be the reason for this behaviour?

Interesting!  How do you isolate to act as just a data cable?

drlauridsen, 2020-02-22:
> Interesting! How do you isolate to act as just a data cable?
[image]

drlauridsen, 2020-02-22:
Could this be related: https://raspberrypi.stackexchange.com/questions/65733/pi-zero-w-reboots-when-connecting-to-wireless

pickerin, 2020-07-06:
Hate to necro this thread, but there hasn't been much discussion since February (maybe everyone stopped driving then!)
I just found this project, I JUST had my Model X 75D upgraded from 2.0 HW to 3.0 HW for FSD, so I just actually got TeslaCam functionality.  I immediately built a Pi-ZeroW.  Love all of the work that has gone into this, but alas, it doesn't work in my car with a single USB cable.  Took me a while to figure it out and I eventually found this thread.  

I then put a battery on the Pi-ZeroW Power port, and now the car sees it fine.  It does "power up" the Pi-Zero, but it never mounts it as a USB drive unless there is auxiliary power for the Pi.  I haven't confirmed what is actually happening on the Pi-Zero as it's also not on my wireless when it's trying to boot (the lights never go to flashing twice / second).

pickerin, 2020-07-06:
> Could this be related: https://raspberrypi.stackexchange.com/questions/65733/pi-zero-w-reboots-when-connecting-to-wireless

Wow, GREAT find.  I'd need to put together a 3.3v power supply I could test this with, but it should be easy to confirm.

pickerin, 2020-07-06:
> Have this issue.. TMS100D 2020.4.1. RPI Zero W.
> PC: connected with single cable to RP USB port, drives show after 30 sec
> In car: a) Single cable (same cable) connected to RP USB port results in [...]

For me, I'm able to power on the 5v port via a PowerBank, with the USB port plugged into the vehicle.  Vehicle sees the drive and TeslaCam works normally.  Without the 5v power from a PowerBank, it seems to boot loop (can't verify it yet, just know it never joins the wireless and never mounts in TeslaCam).

marcone, 2020-07-06:
> > Could this be related: https://raspberrypi.stackexchange.com/questions/65733/pi-zero-w-reboots-when-connecting-to-wireless
> 
> Wow, GREAT find. I'd need to put together a 3.3v power supply I could test this with, but it should be easy to confirm.

There's no need to power the Pi from 3.3V. It makes its own 3.3V from the 5V supplied by USB.

pickerin, 2020-07-07:
> There's no need to power the Pi from 3.3V. It makes its own 3.3V from the 5V supplied by USB.

Then why doesn't it work with a single USB cable?  There's clearly a lot of people having this problem...

I'll also add that the only way I can get mine to work in my vehicle (works flawlessly off of my computer) is:
1) Plug external power (battery pack) into USB Power port on RPIZ
2) Wait for the system to fully boot
3) Plug the USB Data port on RPIZ into the Tesla USB port - at this point my TeslaCam icon will show up on the MCU

Unfortunately, when I get out of the car and come back, it's "dead" again (icon no longer appears on MCU for TeslaCam).  I then have to:
1) Unplug from Tesla
2) Unplug from PowerPack
3) Plug back into PowerPack
4) Wait for RPIZ to power up
5) Plug back into Tesla

Essentially, this makes it useless, as the above process requires about 2 minutes to perform and my family isn't willing to have me screwing around with it before we leave every time we get into the car.

I'm open to suggestions and/or troubleshooting options.  It seemed that @drlauridsen found something that might explain the exact issue I was seeing (an apparent boot loop when powered off of the Tesla USB port), but you're telling us we're wrong.  I have a USB power meter I'm going to check to see if it's in fact providing a full 5V on the USB port.

marcone, 2020-07-07:
> I'm open to suggestions and/or troubleshooting options. It seemed that @drlauridsen found something that might explain the exact issue I was seeing (an apparent boot loop when powered off of the Tesla USB port), 

No, the post you referred to said the exact opposite of that: the Pi was working when powered by 5V/USB, and bootlooping when powered via 3.3V.

Have you tried using two USB cables to connect the Pi to the car? (one from the 'USB' port on the Pi to one of the car's ports, and the other from the "PWR" port on the Pi to the other port on the car)

pickerin, 2020-07-07:
> No, the post you referred to said the exact opposite of that: the Pi was working when powered by 5V/USB, and bootlooping when powered via 3.3V.
>
> Have you tried using two USB cables to connect the Pi [...]

You're of course correct, I misread the post, hoping it was something that would help.

No, it does not power up properly when plugged into both USB ports.

githubsean, 2020-07-07:
For what it's worth, in my Model 3, only one of the USB ports is working
(service appointment is scheduled). I've never had an issue powering the
Pi0 from one USB though.
But maybe you have some sort of issue with your USB ports?

drlauridsen, 2020-07-07:
The link I provided about a bootloop with 3.3V, was meant as a possible explanation, as to why the pi zero bootloops with some cars and "standard" use of only one usb port as power+data connection. It seems the usb port in some cars, for some reason drops just slightly when the pi zero is booting and possibly at the critical time where wifi is enabled - resulting in a bootloop.

I have this problem, in a TMS100D with AP2.5/MCU1.
I have no idea why I have this issue, and others do not.
I solved it by using two usb cables, one for the power port and one for the data port on the pi. I however had to isolate the power pin for the data port cable, to make it a data only cable (as shown in the photo in above post). If I simply connect two standard usb cables, it bootloops..
This works most of the time, but once in a while, I have to disconnect either cable to make it reconnect as a drive..

I just ordered this cable on aliexpress, maybe it will work, so I will only use one usb port in the car..
[image]

But @marcone..
Would it be possible to delay the wifi turning on, and maybe see if that could avoid the bootloop.
Eg a setting in config, where wifi is delayed x seconds?

pickerin, 2020-07-07:
@marcone @drlauridsen I seem to have the same issue, but mine isn't resolved by having two USB ports "powering" the PiZero.  No matter what I do, the only way I can get it to work in the car is to leave the USB Data port unplugged until the device is fully booted (blinking twice a second); then I can plug in the Data cable and it will mount properly.

When powered on only the Data USB port, it will never get to "fully booted".  I'm going to haul a display out to the car so I can see the console and understand what it's doing while booted on only the one USB data port.

drlauridsen, 2020-07-07:
> @marcone @drlauridsen I seem to have the same issue, but mine isn't resolved by having two USB ports "powering" the PiZero. No matter what I do, the only way I can get it to work in the car is to [...]

Just to make sure..
For the data port on the pi, are you using a "data only" cable, like this https://github.com/marcone/teslausb/issues/9#issuecomment-589959287
And then a normal cable for the power only port?
It does not work for me, when using two standard cables..

pickerin, 2020-07-07:
> > @marcone @drlauridsen I seem to have the same issue, but mine isn't resolved by having two USB ports "powering" the PiZero. No matter what I do, the only way I can get it to work in the car is to leave the USB Data port unplugged until the device is fully booted (blinking twice a second); then I can plug in the Data cable and it will mount properly.
> > When powered on only the Data USB port, it will never get to "fully booted". I'm going to haul a display out to the car so I can see the console and understand what it's doing while booted on only the one USB data port.
> 
> Just to make sure..
> For the data port on the pi, are you using a "data only" cable, like this [#9 (comment)](https://github.com/marcone/teslausb/issues/9#issuecomment-589959287)
> And then a normal cable for the power only port?
> It does not work for me, when using two standard cables..

I saw that, no they're both data+power, so that may be the difference.  I saw your post about the tape so I'll give that a try and see if it improves the behavior.  Thank you for the clarification.

marcone, 2020-07-07:
>  I'm going to haul a display out to the car so I can see the console and understand what it's doing while booted on only the one USB data port.

Note that teslausb turns off the hdmi port to save power, so you're going to have to edit the archiveloop script so it doesn't do that (by removing the line that says `/usr/bin/tvservice -o` from /root/bin/archiveloop)

pickerin, 2020-07-07:
> Note that teslausb turns off the hdmi port to save power, so you're going to have to edit the archiveloop script so it doesn't do that (by removing the line that says `/usr/bin/tvservice -o` from [...]

Thank you for the pointer!  Update below.

pickerin, 2020-07-07:
So, using a two-cable, one power, one data (modified to disable the 5V), my Pi will boot properly in the car.  However, it doesn't present the drives to the Tesla unless I then unplug the USB and re-attach it.  Here is my [diagnostics.txt](https://www.dropbox.com/s/dshxcjrguvc0ln6/diagnostics.txt?dl=0) from such a boot (the TeslaCam icon would not appear on the dash, but the PiZero was showing two flashing LEDs / second)

ninjanwong, 2020-07-17:
I have a similar experience, but I can only get teslausb working if I first power the pi from the vehicle usb, then wait for proper boot, then plug in the usb cable (plugged into the usb port on the pi) to the vehicle.  I tried using an external power source (RAVPOWER battery pack) and that did not work.  I've also tried isolating the power pin on the usb connector to make it data only, but that also doesn't seem to work for me.

ninjanwong, 2020-07-17:
I decided to start again, but instead set camsize to 100%.  On my Win10 box, I notice the usb interface is live during boot, then turns off, then is live again.  Is this normal behavior?

ninjanwong, 2020-07-28:
Setting camsize to 100% didn't work.  I bought 4" Ankur premium USB cables and those didn't work either.  The only way I can get it to work is to plug in the USB to the power port and let it boot (1 - 2m), then plug in the USB cable to the vehicle.  Camera logo shows up and stays up till the USB port is shut off, then I have to reboot the pi like above.  Any suggestions?

pickerin, 2020-07-28:
Mine is working "normally".  But I had to use a data-only USB cable for one of the ports (I actually used the "tape over power pin" method) and then a second cable in the power-only port on the Pi-zero.  This is working reliably, but it also burns BOTH of my USB ports in the front (eliminating my ability to charge my phone).  Note:  it will NOT work with two "normal" USB cables, as the Pi-zero doesn't seem to boot properly with power on both cables.

ninjanwong, 2020-07-28:
I am using the tape over method as well, but it still will not completely boot if both are plugged in at the same time.

[9 more replies]