marcone/teslausb #10: Missing in action:  /root/bin/remountfs_rw

ArthurD, 2019-04-07:
Installing this today via your provided image & auto-config (marcone/main-dev), I wanted to change it to use SFTP afterwards but that required non-readonly filesystem.  There was nothing at '/root/bin/remountfs_rw' however (and only two files in /root/bin). 

Easy fix was to run the following instead, which I presume is all that 'remountfs_rw' helper is doing.  
`mount / -o remount,rw`

Just wanted to let you know -- otherwise thank you for this, pretty cool & appreciate that you're maintaining it!  :)

PS -- The only other change I wanted to make was to secure SSH on the Pi;  for most users a password change would probably be fine (and I think that'd be worth at least suggesting, if not automating) though I personally wanted only key-based auth.  

That might be equally easy option to automate (key-based) though, else end up with the password sitting in plaintext on the Pi;  still probably better than leaving it default, though!  :)

marcone, 2019-04-08:
The prebuilt image is meant for (and only tested with) CIFS. If you don't configure it for a working CIFS server, then I think the initial setup, which includes downloading scripts like remountfs_rw, will fail.

marcone, 2019-04-09:
Closing this issue. The headless setup process will download the scripts as needed. For manual setup, you'll need to downloaded them yourself.
For ssh, when you first log into the Pi, it will tell you it is using the default password and to change it if you want to be more secure.

closed 2019-04-09 by marcone