marcone/teslausb #12: Institute version numbers?

leofaoro, 2019-04-08:
Would make it easier for others to help and debug also

marcone, 2019-04-09:
Can you elaborate?

thijsdejong, 2019-04-10:
He means what you are already doing here: https://github.com/marcone/teslausb/releases
@leofaoro check that link out

marcone, 2019-04-10:
Those are the releases for the prebuilt image though. After flashing one of those, setup will still pull the latest/current versions of the scripts from github, not the versions as they existed for that release. And I plan to keep it that way, because I don't want to have to do a prebuilt every time I change something. I'll only do new prebuilts when a change in one of the scripts requires it.

leofaoro, 2019-04-10:
To elaborate. Right now I don't even know what version I have. So it would be nice to have even in the setup.py a line specifying a version number. This would help with troubleshooting.

marcone, 2019-04-11:
There is no setup.py :)
I see your point, but I don't want to have version numbers in every file that need to be kept up to date. Perhaps I can add something to setup-teslausb that writes the date or a sha when it downloads the scripts. Might not cover all cases, but should give a general idea of whether the scripts are up to date.

raushel, 2019-05-18:
Somewhat related question, currently using a different fork (jspv), is there an easy way to switch to yours without re-flashing by updating REPO/BRANCH somewhere? Assuming no, but figured I'd ask.

marcone, 2019-05-18:
That depends on how much jsvp's fork differs. I took a quick peak and I suspect you can probably just point the config file at my fork and rerun setup to download the latest scripts.

alex-peters, 2019-05-18:
@marcone can you elaborate? The latest scripts are pulled on every startup or only the first boot after a flash?

marcone, 2019-05-18:
`/etc/rc.local` will pull the latest `/root/bin/setup-teslausb` once, normally on first boot after flash, and then `setup-teslausb` will download the remaining scripts every time you run it.

If you don't have (m)any customizations, I'd recommend just starting from scratch by flashing my prebuilt.

marcone, 2019-06-04:
I'm going to close this because setup-teslausb is now pretty good at being run again, so you can
```
/root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb
```
to update your install to the latest version, which makes the version number of the scripts less interesting (if you have a problem, update. If you still have a problem, you have a problem with the latest version).

closed 2019-06-04 by marcone