marcone/teslausb #14: Partition Issues

wwwb0n3zcom, 2019-04-15:
I have a 200GB SD card and get the following errors each attempt. The SD card is cleared each attempt and I use Rufus to image.

`Fri 12 Apr 02:54:09 BST 2019 : Starting setup.
Fri 12 Apr 02:54:09 BST 2019 : Updating package index files...
Fri 12 Apr 03:12:00 BST 2019 : Verifying that the requested configuration is valid...
Fri 12 Apr 03:12:01 BST 2019 : Downloaded /tmp/verify-configuration.sh ...
Fri 12 Apr 03:12:01 BST 2019 : Verifying that there is sufficient space available on the MicroSD card...
Fri 12 Apr 03:12:02 BST 2019 : STOP: The MicroSD card is too small: 104857600 bytes available.
Fri 12 Apr 03:12:02 BST 2019 : BYT;
/dev/mmcblk0:197GB:sd/mmc:512:512:msdos:SD SC200:;
1:4194kB:49.9MB:45.7MB:fat32::lba;
2:50.3MB:1816MB:1766MB:ext4::;
3:1816MB:197GB:195GB:ext4::;
4:197GB:197GB:105MB:ext4::;`

marcone, 2019-04-16:
The setup scripts expect to see 2 partitions (root and boot), with free space following those partitions, in which the scripts then create two more partitions. Your card is already fully partitioned, with no free space left.
I just tried Rufus myself, and it does leave the space after the root/boot partitions unallocated like the scripts expect, so given that your card is partitioned exactly how the setup script would partition it, with that large ext4 partition and the 105 MB ext4 partition at the end, it looks like you somehow ran the partitioning/setup script twice in a row.

wakitcha, 2019-04-18:
I've spent the last few days struggling with this too.  From another thread I found that the raspian initialization tries to expand the partition upon first boot too.  Removing that command lets the scripts run as expected.  In the boot partition, edit the cmdline.txt file.  Remove init=… to the end.

Note: once the setup scripts run once, even partially, there will likely be four partitions and running the setup script again will fail with the same error as above.  I'm not sure exactly how to remove them, but you'd have to "clean up" the partitions first and then run setup again.  Or, have the scripts recognize this state and skip over adding partitions a second time.

marcone, 2019-04-18:
I submitted a change that does just that (skip adding the partitions if they've already been added) to the 'test' branch a few hours ago. Give that a try.
I'm curious though: how did you get into a situation where you had to run the script a 2nd time? The headless setup should (at least if you use [my prebuilt](https://github.com/marcone/teslausb/releases)) just run through until the device is fully configured and ready to go. Did you not use my prebuilt, or did you interrupt the setup for some reason?

wakitcha, 2019-04-18:
A couple different things have caused the scripts to stop for me.  Most recently I miss-typed the archive name and it stopped.  More often I don't have the correct CIFS version selected.  Auto-select hasn't worked for me, and 2.1 originally worked for me but no longer does so.  I'm now only getting the script to run by hard coding it to v2.0.

marcone, 2019-04-20:
The scripts involved will now skip the partitioning step if the card was already partitioned previously.

closed 2019-04-20 by marcone