marcone/teslausb #15: Rclone move to GDrive corrupts MP4

travelton, 2019-04-26:
I've configured rclone to use Google Drive. However, I've noticed every uploaded file is corrupted (file size is 500kb each). 

Changing the archive-clips.sh file to use the copy command instead seems to fix this issue. 

```sh
rclone --config /root/.config/rclone/rclone.conf copy "$file_name" "$drive:$path" >> "$LOG_FILE" 2>&1 || echo ""
```

However, this does not clean up the files from the saved clips location. 

For now, I've added after the rclone. This can be dangerous though... If the rclone command is not successful, you may end up deleting without uploading. 
```sh
  rm "$file_name"
```

I will work on a PR to potentially validate the upload prior to removing the local copy.

marcone, 2019-04-26:
Does rclone print any error messages when it truncates the files likes this? (these would be in /mutable/archiveloop.log) Maybe add -v to the command if the default isn't verbose enough.
This seems like a bug in rclone and while it might be worth incorporating a workaround, you should probably also report it to the rclone maintainers.

marcone, 2019-04-26:
Out of curiosity, which version of rclone are you using?
I haven't tried rclone myself, but noticed that the teslausb docs say to use "rclone4pi", which hasn't been updated in about a year, while "sudo apt install rclone" on the pi will install rclone 1.35, which looks like it's over 2 years old.
The canonical rclone from rclone.org is on version 1.47, released 2 weeks ago, and seems to work on rpi out of the box.

travelton, 2019-04-26:
I followed the instructions here, https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md. Which seems to pull the latest rclone4pi. 
```
wget -O rclone.zip https://downloads.rclone.org/rclone-current-linux-arm.zip
``` 
Version output:
```
$ rclone version
rclone v1.47.0
- os/arch: linux/arm
- go version: go1.12.4
```

-v shows the files transferred, with no errors. However, today they show up with 0 kb (yesterday, ~500kb). 

[image]

I think you're right. Sounds like an rclone bug.

Another, sort of related issue, is that the dated folders are not cleaned up after moving. So each time archive-clips.sh is called, you get a false push notification that files were uploaded, when it just tried to sync empty folders. I solved this by deleting everything in the folder `rm -rf "$CAM_MOUNT"/TeslaCam/SavedClips/*`. 

archive-clips.sh:
```bash
#!/bin/bash -eu

log "Moving clips to rclone archive..."

source /root/.teslaCamRcloneConfig

NUM_FILES_MOVED=0

for file_name in "$CAM_MOUNT"/TeslaCam/saved* "$CAM_MOUNT"/TeslaCam/SavedClips/*; do
  [ -e "$file_name" ] || continue
  log "Moving $file_name ..."
  rclone -vvv --config /root/.config/rclone/rclone.conf copy "$file_name" "$drive:$path" >> "$LOG_FILE" 2>&1 || echo ""
  log "Moved $file_name."
  NUM_FILES_MOVED=$((NUM_FILES_MOVED + 1))
done

rm -rf "$CAM_MOUNT"/TeslaCam/SavedClips/*

log "Moved $NUM_FILES_MOVED file(s)."

/root/bin/send-pushover "$NUM_FILES_MOVED"

log "Finished moving clips to rclone archive"
```

marcone, 2019-04-27:
I did something similar in https://github.com/marcone/teslausb/commit/8be6a913d2d9c525cb2f3a336d59b1536282f7a6 for the CIFS archive script, but didn't add it for any of the other archive methods.

Looks to me like your rclone script won't do the right thing when the archive operation is interrupted though. I expect the rclone operation will fail when you get out of wi-fi range while archiving is still in progress, and then that "rm -rf" will delete all your recordings including the ones that haven't been backed up yet.

eranarbel, 2019-06-04:
I'm also trying to configure the pi to archive into GDrive using rclone. What's the right sequence to accomplish that? 

The pre-built image / one step instruction is solving for CFIS sync, and following the step-by-step instructions results in an "outdated" error on the USB storage setup stage (which was predicted to happen per the instructions).

marcone, 2019-06-04:
I can't help you with rclone setup, but what "outdated" error are you seeing?

thedriveman, 2019-06-04:
See the error I referred to below. Looks like verify-configuration.sh looked for setup-teslausb under /root/bin, but the files exists under /root, not /root/bin. In the meantime, I've fixed this manually to complete the installation. 
---
root@raspberrypi:~# ./setup-teslausb
Updating package index files...
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]                         
Get:3 http://archive.raspberrypi.org/debian stretch/main armhf Packages [221 kB]                   
Get:4 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [45.0 kB]                     
Get:5 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]                
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]            
Fetched 12.1 MB in 24s (492 kB/s)                                                                   
Reading package lists... Done
Verifying that the requested configuration is valid...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1829  100  1829    0     0   3600      0 --:--:-- --:--:-- --:--:--  3607
Downloaded /tmp/verify-configuration.sh ...
grep: /root/bin/setup-teslausb: No such file or directory
setup-teslausb is outdated, attempting update
STOP: failed to update setup-teslausb

marcone, 2019-06-04:
Oh I see, you did a manual setup, and put setup-teslausb in /root/ instead of /root/bin/ where the headless setup puts it.

thedriveman, 2019-06-04:
Correct. I invoked the scripts on the default directory (/root) following login; IRRC the manual setup does not require a specific directory. Perhaps this can be fixed in the instructions?

appark, 2019-06-20:
So if I use an older version of rclone with this temp will this fix the problem?

travelton, 2019-06-20:
@appark I don't think there's a solid solution for this problem yet. There's really only one option right now, set the rclone command to "copy" instead of "move". Then manually clean up your recordings when the drive fills up. I have not had time to submit a pull request for this situation.

appark, 2019-06-21:
I was able to install an older version of rclone and it works

marcone, 2019-06-22:
@appark: can you clarify which version you used, and how to go about installing it? It might help others running into the same issue with rclone.

closed 2019-11-12 by marcone