marcone/teslausb #17: Add an (optional) delay after detecting archive is available

twrecked, 2019-04-26:
Thanks for all these fixes, I've been using your branch for 2 weeks and it's been working great.

I did have one issue, the first time I arrived home the FAT partition was too corrupted to fix, it couldn't be mounted and the images couldn't be transferred. I think the problem was the device connected to my WiFi and detected my archive while I was still pulling into the drive and unmounted the USB drive while the car was still writing a video file.

I added a small delay after the archive is detected to give me time to park and the car time to write the last recording.

I thought it might be useful to support an optional delay.

marcone, 2019-04-26:
Yes, I plan on adding something like that (and in fact have been running my setup like that for a while now), though mainly because I don't like my USB music cutting out as soon as I pull into my driveway :)

Rykemapo, 2019-04-26:
I would also like this feature since I live on a blind curve and my highest probability of being struck is within WiFi range.

marcone, 2019-05-08:
There is now a default delay of 20 seconds, but you can configure it to whatever you like.

closed 2019-05-08 by marcone