marcone/teslausb #19: Is it possible to save all recordings

clean-ev, 2019-05-02:
Title says it all. 
It is known that the car will overwrite after 59-60 minutes. Is it possible to code where this current files are moved to ‘Saved’ folder every 59-60 minutes to have a continuous recording.
I understand that this will require larger SD cards.
Thank you

marcone, 2019-05-03:
That's something I'm working on

Vipercat, 2019-05-28:
Short question to @marcone: how do I mount the cam_disk.bin file to /mnt/cam via ssh to check the contents of the RecentClips and SavedClips folders? I've once seen the mount to be active, but I don't seem to be able to manually mount the file.

marcone, 2019-05-28:
If it's currently archiving, then it should already be mounted on /mnt/cam.
If it's not archiving, `sudo mount -o ro /mnt/cam` should to the trick. To be extra safe you may want to `sudo modprobe -r g_mass_storage` before that, so that the car isn't writing to the image at the same time.

Vipercat, 2019-05-28:
Thanks! 

BTW: after you've updated your scripts to use ext4 instead of btrfs for the 'backingfiles' I haven't had any issue with the partition filling up and the RPI is working like a charm ever since. The only issue I have is the alarm going off occasionally. Only at home though when connected to WiFi. I will investigate further to try and analyse the exact cause. I do want to thank you for all the work done on developing this system with the RPI! Love the functionality.

notrhj, 2019-06-12:
If I log in to Pi while its connected to car and reboot
Boom Car goes into Alarm. USB port in car gets disabled (no power).
Need to do a steering wheel reset to get the USB port back.