marcone/teslausb #21: Running setup script produces error

macbentosh, 2019-05-05:
after running the setup script the final error is posted 
`/dev/mmcblk0p4 contains a ext4 file system labelled 'mutable'
	created on Sun May  5 08:13:49 2019
Proceed anyway? (y,N) y
Discarding device blocks: done
Creating filesystem with 103424 1k blocks and 25896 inodes
Filesystem UUID: a6b29797-65c4-4782-a3c1-403b7d0f48e2
Superblock backups stored on blocks:
	8193, 24577, 40961, 57345, 73729

Allocating group tables: done
Writing inode tables: done
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done

Mounting the partition for the backing files...
Mounted the partition for the backing files.
Creating backing disk files.
create-backingfiles: starting
create-backingfiles: starting
create-backingfiles: Allocating 0K for /backingfiles/cam_disk.bin...
create-backingfiles: Allocating 0K for /backingfiles/cam_disk.bin...
fallocate: invalid length value specified`

marcone, 2019-05-05:
I highly recommend using the one-step headless setup method instead of manual setup.
If you want to run manual setup anyway, please list the exact commands you ran, and include the full output of each of those commands.

macbentosh, 2019-05-05:
I ran the auto setup and no drives were created that would mount on the machine. Also where would the logs be for the auto setup. Everything else works fine. Ssh WIFI etc just no drive.

marcone, 2019-05-05:
The setup-log is `/boot/teslausb-headless-setup.log`

macbentosh, 2019-05-05:
All that is mentioned there is about the wpa_sup file.

marcone, 2019-05-05:
Can you attach the entire teslausb-headless-setup.log here?

macbentosh, 2019-05-05:
nano teslausb_setup_variables.conf

export ARCHIVE_SYSTEM="cifs"
export archiveserver="192.168.0.1.60"
export sharename="Tsentry"
export shareuser="tez"
export sharepassword="password"
export campercent=100
export pushover_enabled=true
export pushover_user_key=data
export pushover_app_key=data
export TESLAUSB_HOSTNAME=ModelXSentry
export REPO=marcone
export BRANCH=main-dev

Sanitized obvoiusly

marcone, 2019-05-05:
Thanks, but I meant the log, not the configuration file :)
The setup log is /boot/teslausb-headless-setup.log

One thing that seems obviously with your configuration file though is the IP address: it has 5 parts instead of 4. Perhaps you meant 192.168.1.60 ? (you can also use the host name of your server)

macbentosh, 2019-05-05:
the ip is false it was for sanitization..

I am running it again to check and will post that log.  I wanted to show you what I was starting with to see if there was anything obvious.

macbentosh, 2019-05-05:
cat /boot/teslausb-headless-setup.log 
Sun 21 Apr 18:21:55 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 21 Apr 18:21:55 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sun 21 Apr 18:21:55 BST 2019 : Rebooting...
Sun 21 Apr 18:22:19 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 21 Apr 18:22:19 BST 2019 : Grabbing main setup file.

Shouldn't take more than 20 min right?

marcone, 2019-05-05:
No, "Grabbing main setup file" means /etc/rc.local is trying to download the setup-teslausb script that continues the setup. That should take at most a few seconds, and then it should continue with: "Starting setup."
So it looks like you have connectivity problems. What does 'ifconfig' say if you ssh into the pi?

macbentosh, 2019-05-05:
I moved on to the troubleshooting step
at 
Try sudo -i and then run /etc/rc.local. The scripts are fairly resilient to restarting and not re-running previous steps, and will tell you about progress/failure.

macbentosh, 2019-05-05:
looks like running rc.local did the trick...Not sure why just booting it up didn't work. guess this can be closed.

closed 2019-05-05 by marcone