marcone/teslausb #23: "Can't find btrfs tools."

jasoncross, 2019-05-07:
Ran an older version for the past several months without issue until v9 (like most).  Recently decided to update to see if it would now work with my X again so rebuilt my TeslaUSB using the latest one-step install.  After running, I found that the CAM / MUSIC folders would no longer show up even on my Mac.

This morning I decided to do the manual install to see if that worked any better.  I got to the step for running ./setup-teslausb and it ran for a short time before erroring with the following:

```
Downloaded /root/configure.sh ...
~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
Starting to create backing files partition...
/tmp/create-backingfiles-partition.sh: line 21: hash: mkfs.btrfs: not found
Can't find btrfs tools. Please install manually or use the correct prebuilt
```

marcone, 2019-05-07:
As it says there: `Please install manually or use the correct prebuilt`

I recommend you use the latest prebuilt image from https://github.com/marcone/teslausb/releases, which has everything needed already.
If you want to work off your own image, you should simply install btrfs-tools: `sudo apt install btrfs-tools`

jasoncross, 2019-05-07:
I used the prebuilt teslausb-20190505 first and it didn't appear to be creating the partitions either.

marcone, 2019-05-07:
Do you have any error messages or logs from that first attempt?

marcone, 2019-05-07:
Odd that several of those lines are printed twice...
Looks like you either explicitly specified the size of the music drive to be 0, or there is a rounding error somewhere that makes it end up being zero. What size sdcard are you using, and how big is the /backingfiles partition?

For now, setting camsize to something smaller will probably work around the issue. Try "90%".

marcone, 2019-05-07:
Sorry, our messages crossed there :)
On closer look at the scripts, I think there's a bug: if musicsize is not set, it uses '10' by default, i.e. 10 bytes. Try explicitly setting it to 50% (or whichever size you want)

jasoncross, 2019-05-07:
> Odd that several of those lines are printed twice...
> Looks like you either explicitly specified the size of the music drive to be 0, or there is a rounding error somewhere that makes it end up being [...]

It's a 32GB SD card.  As I said above, I set camsize=50% and did not explicitly set the music size to anything.

Here are the partitions:

```
/dev/mmcblk0p1          8192    96042    87851 42.9M  c W95 FAT32 (LBA)
/dev/mmcblk0p2         98304  3522559  3424256  1.6G 83 Linux
/dev/mmcblk0p3       3522560 62316543 58793984   28G 83 Linux
/dev/mmcblk0p4      62316544 62521343   204800  100M 83 Linux
```

jasoncross, 2019-05-07:
> Sorry, our messages crossed there :)
> On closer look at the scripts, I think there's a bug: if musicsize is not set, it uses '10' by default, i.e. 10 bytes. Try explicitly setting it to 50% (or whichever size you want)

Will do.  Do I need to restart from scratch to do so?

marcone, 2019-05-07:
You should be able to just run setup again. The script that creates the image files will delete them before recreating them.

jasoncross, 2019-05-07:
> You should be able to just run setup again. The script that creates the image files will delete them before recreating them.

Ok will give it a go!

I wonder if this is why the one-step wasn't working for me either, as I wasn't setting the music size explicitly there either.

marcone, 2019-05-07:
Probably. That used to work, but it looks like I broke it somewhere along the way. Sorry about that.

jasoncross, 2019-05-07:
Ok, so setup completed, but it still doesn't want to show either CAM or MUSIC drives mounted to my Mac.  I'm going to retry the one-step setup setting both music and cam sizes explicitly and see if that works better.

jasoncross, 2019-05-07:
Ok, I was able to get it to work with the one-step by setting music explicitly.

closed 2019-05-07 by jasoncross

marcone, 2019-05-08:
The music size issue should be fixed now. If unspecified it will use the remainder of the space for music, as documented.

the-scott-davis, 2019-05-14:
> If you want to work off your own image, you should simply install btrfs-tools: `sudo apt install btrfs-tools`

I ran this portion and was able to get it working. Essentially, just re-ran the ./setup-teslausb item after I installed btrfs-tools