marcone/teslausb #25: Secret to connecting over wifi

acraigl, 2019-05-10:
Just tried to get this going. My router can see the device as a wifi client but I can't see how to access it over wifi. I tried going to \\teslausb and it's assigned IP address (10.0.0.12) but I can't 
access it. The only config changes I made was adding my SSID and password.

I'm sure I'm doing something wrong, but I'm stuck?

marcone, 2019-05-10:
It sounds like you're trying to access the Raspberry Pi as a network drive from your Windows PC.
While that would certainly be a very interesting feature to add, the way it's supposed to work is that you configure it to connect to your wifi (sounds like you got that part covered), and set it up so that once it is connected it backs up your car's saved recordings to a network share on your network, or to a cloud storage provider. It sounds like you didn't do that last part, and if you ssh into the device, you'll probably find that the 'archiveloop' script isn't running.

acraigl, 2019-05-10:
Yes! That's exactly what happened, and what my misunderstanding was. Thank you!
I'll probably start with the windows share, but cloud backup is not as obvious to me... do I need to use rClone (https://github.com/cimryan/teslausb/blob/master/doc/SetupRClone.md) or is that handled in one-step as well? I'm thinking Amazon ClouldDrive if that matters.

Also, is there any logic in play for when the drive fills up (i.e. auto-removing the oldest files before that happens)?

marcone, 2019-05-10:
I have only tried CIFS (windows share) myself, so I can't really comment on rclone. I think you have to set that up manually, because it requires some manual steps to get an authentication token for the service, which isn't handled by the headless setup.

As for the card filling up: the recordings are deleted after backup, so if you're regularly within wi-fi range and the card is big enough, it should never fill up. If you're out of wifi range for a long time and save lots of recordings, then the car itself has logic to remove older files, though I don't think it has logic for removing manually saved recordings, so if you keep pressing the dashcam icon to save clips, you *can* fill up the card.

senf42, 2019-05-10:
Do the recordings get deleted only if they finish uploading? I'm wondering what the process is if there's a file in the middle of being uploaded when the car goes to sleep, and there's no battery backup.

acraigl, 2019-05-10:
@marcone thanks so much for the help, and for supporting this project. Much appreciated!

marcone, 2019-05-10:
@senf42 it should only delete a file once it has been completely backed up. If the upload is interrupted in the middle of a file, it will retry it the next time it is within wifi range.
That being said, I use an OBD2 power adapter to keep the rpi permanently powered, so I don't have to mess with leaving climate control or sentry mode on in order to keep the rpi powered.

wakitcha, 2019-05-11:
@marcone can you share a link to the adapter you have?  I've been playing with an rpi battery UPS, but always powered may be preferable.

marcone, 2019-05-11:
I got this: https://www.ebay.com/itm/OBD-charging-cable-5V-3A-micro-USB-power-adapter-with-switch-button-connecto-Jh/143150452836
Before that I used a RAVPower 22000 mAh portable charger, which would last several days before it needed to be charged again.

closed 2019-05-15 by marcone