marcone/teslausb #26: I/O Issues

akarnani, 2019-05-11:
I just tried using the manual install to setup an rclone to dropbox teslausb.  However no writes to the provided FAT mounts are persisted.  In fact, mac os complains that it cannot repair either `CAM` or `MUSIC` and that they are read only.  I see similar behavior when I mount the backingfiles in raspbian as well.

I found these errors in the dmesg:
```[ 1392.998546] loop: Write error at byte offset 1049088, length 512.
[ 1392.998578] print_req_error: I/O error, dev loop0, sector 1
[ 1392.998598] Buffer I/O error on dev loop0, logical block 1, lost async page write
[ 1393.009947] loop: Write error at byte offset 14516224, length 512.
[ 1393.009977] print_req_error: I/O error, dev loop0, sector 26304
[ 1393.009998] Buffer I/O error on dev loop0, logical block 26304, lost async page write
[ 1403.078890] loop: Write error at byte offset 1049088, length 512.
[ 1403.078912] print_req_error: I/O error, dev loop0, sector 1
[ 1403.078926] Buffer I/O error on dev loop0, logical block 1, lost async page write
[ 1448.517260] loop: Write error at byte offset 14516224, length 512.
[ 1448.517291] print_req_error: I/O error, dev loop0, sector 26304
[ 1448.517312] Buffer I/O error on dev loop0, logical block 26304, lost async page write```

I also tried running `rpi-update`.  This happens on both the 4.14 and 4.19 kernels.

Dennis4b, 2019-05-11:
Fundamentally the problem is that you really don't want to be using a loopback file as the backend for the USB volumes, especially not on a COW filesystem such as btrfs which needs lots of free space to work efficiently.

If you allocate all the available space the backing filesystem has no room for the housekeeping tasks in the backing partition, changes then can't be written to the USB volumes and you get errors such as the ones mentioned above.

You can work around this by leaving some percentage unallocated (i.e. camstorage+musicstorage should not be 100%).

You want to ideally create raw partitions for music/cam, and export those (make /mutable the 3rd partition, then a 4th as an extended partition to fill up the SD and then create 2 logical partitions in there). Then in /etc/fstab and archiveloop you don't need the loopback mounts.

Then you can use the whole sd-card without worry and unneeded abstractions. There are a couple of other improvements (fsck before mount, not after) that could be added too.

marcone, 2019-05-11:
@Dennis4b Very interesting, thanks for the analysis. I probably never ran in to that myself because I leave plenty of free space (32GB cam, 1GB music on my 64GB and 128GB cards I use for testing), and even at 100% there is usually a bit of free space left due to rounding.
I was able to replicate the problem with a different card just now though, and it went away when limiting the size of the cam file and leaving just over 1GB of free space on the backingfiles partition. Not sure how much free space is really needed, but I'll update the setup script to always leave some free space.

Dennis4b, 2019-05-11:
What I tried:

Use all the free space on the SD card for a single primary partition (one is /boot, one is /, one is /mutable).

`losetup -fP /dev/mmcblk0p4`  (or p3 depending on setup)
`fdisk /dev/loop0`
`# Create 2 partitions according to cam/music ratio, use the whole device`
`#(write down the start cylinders, see below)`
`mkfs.vfat /dev/loop0p1`
`mkfs.vfat /dev/loop0p2`

and export the single device (in `/etc/modprobe.d/g_mass_storage.com`)
`options g_mass_storage file=/dev/mmcblk0p4 removable=1 ro=0 stall=0 iSerialNumber=123456`

This works, as a USB disk with 2 partitions gets exported, without worrying about the free space, indirection of a nested filesystem, or fsync/sync issues.

You can then also just mount/fsck /dev/loop0p1 and /dev/loop0p2 _while not exporting the disk_

However this gave me a kernel oops after a while, in process `file-storage`! 

So currently I am trying to do this without a permanent loopback device (so the Pi also doesn't see the partitions all the time and tries to do who knows what).

(You still need to create one to do the initial fdisk/mkfs.vfat, but only once)

First attempt was to do `losetup /dev/loop0 /dev/mmcblk0p4` and then `losetup -d /dev/loop0` on demand, but the detaching somehow takes forever / hangs so this didn't work.

So the second attempt is to do similar to the backingstore idea, changing fstab like so:
```
/dev/mmcblk0p4  /mnt/music vfat noauto,users,umask=000,offset=1048576
/dev/mmcblk0p4  /mnt/cam vfat noauto,users,umask=000,offset=12885950464
```
(offsets based on start cylinder reported by fdisk multiplied by 512, mine is for ~12G/50G music/cam)

Then a `/dev/loop0` gets created only when mounting `/mnt/cam` or `/mnt/music` which also explains why the archiveloop script first mounts and then fscks.

You can not mount `/mnt/cam` and `/mnt/music` on the Pi simultaneously, this doesn't matter for the functionality, and I think I didn't have to change the archiveloop script to make sure this doesn't happen anyway.

So now I'm trying it out and waiting to see if it Oopses again or now works :)

marcone, 2019-05-11:
I've updated the setup script to leave 1GB of unused space. Probably more than is needed, but I figure storage is cheap, and better safe than sorry.

patelhiren, 2019-05-13:
This is insightful. I was running into similar problems when I updated to the _teslausb-20190505_ build. I realized the backingfile was changed from ext4 to btrfs. I reverted the script to use ext4 on my fork and it worked well. I was out on a three day trip and the dashcam + sentry mode features worked well with the pi and it synced the videos when I got home without any problems. Is btrfs more better than ext4 for the usecase we have?

I am using a 32gb card with no music drive. Just 100% allocated to CAM.

Dennis4b, 2019-05-13:
Honestly, and without casting any shadow on the great discovery work done in this and the original repo, you don't want to use a filesystem at all. Unfortunately `btrfs`, while generally a good filesystem, is an especially unlucky choice because of its COW design needing lots of free space. 

You actually want to export volumes directly, that way you can use all the free space, not worry about sync, and avoid the indirection.

(I "fixed" my random crashes with "over_voltage=4" and "force_turbo=1", so retried the steps I listed earlier).

Easiest to do (assuming all free space is in the 4th partition):
```
losetup -P /dev/loop0 /dev/mmcblk0p4
fdisk /dev/loop0
# within fdisk create 1 partition for music and 1 for cam
mkfs.vfat /dev/loop0p1   # cam
mkfs.vfat /dev/loop0p2   # music
mount /dev/loop0p1 /mnt/cam
mkdir /mnt/cam/TeslaCam
umount /mnt/cam

mount /dev/loop0p2 /mnt/music
# Copy your music
umount /mnt/music
```
and then in `g_mass_storage.conf` export a single `/dev/loop0` - regardless of using only one cam partition or a cam and a music partition.

You'll want to add `losetup -P /dev/loop0 /dev/mmcblk0p4` to the start of `/etc/rc.local`.

(Even with one partition only you still need to `fdisk` and create one partition and use `/dev/loop0p1` as the Tesla does not accept an exported partition only, it wants to see a partition table)

What I have working at the moment:
Create an LVM space and create an LVM volume leaving a few gigabytes free. Then export the LVM volume as above, just using `losetup -P /dev/loop0 /dev/VG/TeslaUSB`. However, now you can take a snapshot of the LVM volume and "peek" inside the Cam volume while it is exported, for example to copy files without having to "unplug" the virtual USB volume, or to for example decide there's nothing to do (no new SavedClips) so no need to unplug, or decide to monitor the free space and only unplug once it's time to clean up. You just need to keep in mind that Tesla might keep writing so you don't want the snapshot around too long (this is what the free gigabytes within the LVM space are for) but it's fine to just copy everything off.

Takes a bit of tinkering, but seems to work well (I have a 64GB high endurance SD). Needs some more scripting to get it all automated.

Will write it all up in a gist when I get around to it.

My only issue at the moment is that the front USB ports are pretty finicky about supplying power and sometimes do not like the Pi booting and will stop, after which they will happily talk to the Pi as a USB stick as long as you supply external power from the cigarette lighter, but then you need to manage the cabling. From reports it seems this is a recent problem so maybe it gets better in a future update.

marcone, 2019-05-13:
The reason I'm using btrfs is so I can take a quick snapshot of the disk
image while it is in use, fsck that snapshot, and then copy any videos off
of it before they are expired by the car. Being able to get a
complete/continuous set of recordings that covers more than 1 hour is the
main reason I created this fork. If that can be done better/safer with LVM,
that's fine with me too. Just seemed simpler with btrfs.

Dennis4b, 2019-05-13:
Ah yes that's also possible with btrfs. You copy also the RecentClips? Do you have a second storage space for that? That's also an interesting usecase, you could get a whole trip archived that way.

I didn't mean to hijack the discussion, with btrfs you do need to leave enough free space (more than with ext4, which was probably the problem with this issue 26) - which you also need for the snapshots, but otherwise if it works it works!

It's easy to go in and find small things that could be better, the hard part is getting everything running and documented, for which I thank you and upstream, wouldn't have gotten this far without it.

marcone, 2019-05-13:
> Ah yes that's also possible with btrfs. You copy also the RecentClips? Do you have a second storage space for that? That's also an interesting usecase, you could get a whole trip archived that way.

Yes, I was planning on basically extending the RecentClips past one hour, and then optionally also backing them up.
It always seemed odd to me that Tesla decided to delete everything older than an hour, even if there is still plenty of space left on the card. On a 128GB card, /backingfiles ends up being 118GB, which is enough for over 20 hours of video.

marcone, 2019-05-17:
Per https://github.com/marcone/teslausb/issues/28 it seems that even leaving a gigabyte of free space on the partition is not enough. I've changed the scripts back to use ext4 for now.

marcone, 2019-05-19:
What I've found is that btrfs loses space whenever the drive is disconnected from the car while it's being written to. The space can be partially recovered by doing a `btrfs balance` as long as there is still enough free space, however it only takes a few disconnect/reconnect cycles to lose enough free space that recovering becomes impossible.

closed 2019-05-19 by marcone