marcone/teslausb #28: Deleted All Videos, Saved Nothing

xnaas, 2019-05-17:
Hey so Sentry Mode told me it had 4 events so it should've saved something...but it saved absolutely nothing. Yes the camera icon was there and should've been working.

What logs would be helpful in figuring out why it would've not saved files or just deleted them?

marcone, 2019-05-17:
"there's nothing left in /mnt/cam" sounds like the normal state after a successful backup: after a backup it will unmount the drive from /mnt/cam, and reconnect it to the car, i.e. at that point /mnt/cam will just be the empty mountpoint, but the car will see the drive.
You can check /mutable/archiveloop.log, which should tell you exactly if it backed up any files and if so which ones.
If it backed up something, you should see something like:
```
Wed 15 May 18:32:23 PDT 2019: Moving clips to archive...
Wed 15 May 18:32:23 PDT 2019: Creating output directory ''
Wed 15 May 18:32:23 PDT 2019: Creating output directory '2019-05-15_16-17-03'
Wed 15 May 18:32:23 PDT 2019: Moving '2019-05-15_16-17-03/2019-05-15_16-16-front.mp4'
Wed 15 May 18:36:34 PDT 2019: Moved '2019-05-15_16-17-03/2019-05-15_16-16-front.mp4'
Wed 15 May 18:36:34 PDT 2019: Moving '2019-05-15_16-17-03/2019-05-15_16-16-left_repeater.mp4'
Wed 15 May 18:40:50 PDT 2019: Moved '2019-05-15_16-17-03/2019-05-15_16-16-left_repeater.mp4'
Wed 15 May 18:40:50 PDT 2019: Moving '2019-05-15_16-17-03/2019-05-15_16-16-right_repeater.mp4'
Wed 15 May 18:47:44 PDT 2019: Moved '2019-05-15_16-17-03/2019-05-15_16-16-right_repeater.mp4'
Wed 15 May 18:47:44 PDT 2019: Moving '2019-05-15_16-17-03/2019-05-15_16-17-front.mp4'
Wed 15 May 18:49:08 PDT 2019: Moved '2019-05-15_16-17-03/2019-05-15_16-17-front.mp4'
Wed 15 May 18:49:08 PDT 2019: Moving '2019-05-15_16-17-03/2019-05-15_16-17-left_repeater.mp4'
Wed 15 May 18:51:56 PDT 2019: Moved '2019-05-15_16-17-03/2019-05-15_16-17-left_repeater.mp4'
Wed 15 May 18:51:56 PDT 2019: Moving '2019-05-15_16-17-03/2019-05-15_16-17-right_repeater.mp4'
Wed 15 May 18:53:11 PDT 2019: Moved '2019-05-15_16-17-03/2019-05-15_16-17-right_repeater.mp4'
Wed 15 May 18:53:11 PDT 2019: Moved 6 file(s), failed to copy 0, deleted 0.
Wed 15 May 18:53:11 PDT 2019: Finished moving clips to archive.
```
If there was nothing to back up, it will show something like this instead:
```
Thu  3 Nov 10:17:16 PDT 2016: Moving clips to archive...
Thu  3 Nov 10:17:16 PDT 2016: Creating output directory ''
Thu  3 Nov 10:17:16 PDT 2016: Creating output directory ''
Thu  3 Nov 10:17:16 PDT 2016: Moved 0 file(s), failed to copy 0, deleted 0.
Thu  3 Nov 10:17:16 PDT 2016: Finished moving clips to archive.
```

Also if you set up your Pi between May 5 and 11 and used 100% of the available space, you may have to redo setup. There was a problem with using all the available space that could prevent the car from writing to the drive at all.

xnaas, 2019-05-17:
This is a snippet of the archiveloop from the relevant time:

    Thu 16 May 16:45:15 CDT 2019: Archive is reachable.
    Thu 16 May 16:45:35 CDT 2019: Archiving...
    Thu 16 May 16:45:35 CDT 2019: Disconnecting usb from host...
    Thu 16 May 16:45:35 CDT 2019: Disconnected usb from host.
    Thu 16 May 16:45:36 CDT 2019: Ensuring cam archive is mounted...
    Thu 16 May 16:45:36 CDT 2019: /mnt/archive is already mounted.
    Thu 16 May 16:45:36 CDT 2019: Ensured cam archive is mounted.
    Thu 16 May 16:45:36 CDT 2019: Starting...
    Thu 16 May 16:45:36 CDT 2019: Ensuring cam file is mounted...
    Thu 16 May 16:45:36 CDT 2019: Mounting /mnt/cam...
    Thu 16 May 16:45:36 CDT 2019: Mounted /mnt/cam.
    Thu 16 May 16:45:36 CDT 2019: Ensured cam file is mounted.
    Thu 16 May 16:45:36 CDT 2019: Running fsck on /mnt/cam...
    fsck from util-linux 2.29.2
    fsck.fat 4.1 (2017-01-24)
    0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
     Automatically removing dirty bit.
    FATs differ but appear to be intact. Using first FAT.
    Reclaimed 35857 unused clusters (1174962176 bytes) in 46 chains.
    Free cluster summary wrong (3726586 vs. really 3690750)
      Auto-correcting.
    Performing changes.
    /dev/loop0: 121 files, 85567/3776317 clusters
    Thu 16 May 16:45:39 CDT 2019: Finished fsck on /mnt/cam.
    Thu 16 May 16:45:39 CDT 2019: Moving clips to archive...
    Thu 16 May 16:45:39 CDT 2019: Creating output directory ''
    Thu 16 May 16:45:39 CDT 2019: Creating output directory ''
    Thu 16 May 16:45:39 CDT 2019: Moved 0 file(s), failed to copy 0, deleted 0.
    Thu 16 May 16:45:39 CDT 2019: Finished moving clips to archive.
    Thu 16 May 16:45:39 CDT 2019: Unmounting /mnt/cam...
    Thu 16 May 16:51:41 CDT 2019: Unmounted /mnt/cam.
    Thu 16 May 16:51:41 CDT 2019: Finished archiving.
    Thu 16 May 16:51:41 CDT 2019: No music archive configured
    Thu 16 May 16:51:41 CDT 2019: Connecting usb to host...
    Thu 16 May 16:51:41 CDT 2019: Connected usb to host.

But if Sentry Mode had 4 events, it definitely should've saved something. Any other logs I can look it to see maybe why it didn't save something?

Note: It's a 128GB microSD card that's basically empty.

marcone, 2019-05-17:
what does `df /backingfiles` say?

xnaas, 2019-05-17:
root@teslausb:~# df /backingfiles
    Filesystem     1K-blocks      Used Available Use% Mounted on
    /dev/mmcblk0p3 122983424 122125792         0 100% /backingfiles
    root@teslausb:~#

marcone, 2019-05-17:
Odd, you seem to be "missing" over 800 megabytes of space: /backingfiles is 117.3 gigabytes, of which 116.5 gigabytes are used, yet there is 0 available space.
What does `btrfs fi df /backingfiles/` say?

xnaas, 2019-05-17:
```
root@teslausb:~# btrfs fi df /backingfiles/
Data, single: total=116.27GiB, used=116.27GiB
System, single: total=4.00MiB, used=16.00KiB
Metadata, single: total=1.01GiB, used=99.61MiB
GlobalReserve, single: total=98.84MiB, used=0.00B
root@teslausb:~#
```

marcone, 2019-05-17:
Any errors in dmesg?

xnaas, 2019-05-17:
An _insane_ amount of lines like this:

```
[248663.943651] print_req_error: I/O error, dev loop0, sector 50211
[248663.943667] Buffer I/O error on dev loop0, logical block 50211, lost async page write
[248668.838630] lo_write_bvec: 408 callbacks suppressed
[248668.838646] loop: Write error at byte offset 26966016, length 512.
[248668.838669] print_req_error: 408 callbacks suppressed
[248668.838678] print_req_error: I/O error, dev loop0, sector 50620
[248668.838694] buffer_io_error: 408 callbacks suppressed
[248668.838704] Buffer I/O error on dev loop0, logical block 50620, lost async page write
[248668.857937] loop: Write error at byte offset 26966528, length 512.
[248668.857967] print_req_error: I/O error, dev loop0, sector 50621
[248668.857989] Buffer I/O error on dev loop0, logical block 50621, lost async page write
[248668.870072] loop: Write error at byte offset 26967040, length 512.
```
This goes on basically indefinitely.

marcone, 2019-05-17:
The "lost async write" thing is the same error as https://github.com/marcone/teslausb/issues/26, which I thought I fixed by leaving about a gigabyte of free space. Either that's not enough, or something else went wrong. When did you set up this device?

xnaas, 2019-05-17:
The 13th.

marcone, 2019-05-17:
At this point I would recommend redoing setup and possibly also selecting a smaller camsize. Note that the car will only keep the last hour of non-saved recordings, or about 5.5 GB, so using 116 GB for recordings is enough for the last 1 hour, plus another 20 hours of saved recordings. I doubt you press the save button that much between backups :)

I changed the scripts back to using ext4 for the backingfiles partition for now, until I can sort out the btrfs issue.
Give it a few minutes to propagate on Github, then rerun setup. You don't need to download a new release, just reflash 20190505, and it will download the current setup scripts.

xnaas, 2019-05-17:
So in the `.conf` I should just set the size to like 50% or something?

marcone, 2019-05-17:
Now that it's back to using ext4 I don't think it really matters and 100% should be fine.
I usually use "32G" myself.

xnaas, 2019-05-17:
Oh, you can specify a size as well? Guess I missed that!

Cat forced me to bed, but I’ll give the updated stuff a try in the morning. Thanks for the quick replies/resolution!

Btw: any reason for aiming for btrfs over ext4 in general? I know Synology uses a hybrid solution of btrfs and ext4, but their reasoning involves btrfs’ issues with RAID. Just curious which btrfs features you’re looking to take advantage of!

marcone, 2019-05-17:
The plan was to use btrfs to take snapshots of the disk image while it's in use by the car, so I can get past the artificial 1-hour time window and keep more recordings.

xnaas, 2019-05-17:
Ahhh, very cool. I wish you luck in that endeavor! Worst case scenario, I guess btrfs just needs a boatload of free space. 🙃

Thanks again. ♥️

xnaas, 2019-05-17:
Didn't get a chance to set this up again this morning, but I just set it back up after getting home. Since it's already been rolled back to ext4, I'll go ahead and close out this issue and monitor #26. :)

Thanks once again!

closed 2019-05-17 by xnaas

xnaas, 2019-05-18:
Well, not sure what I fucked up this time around, but this doesn't seem right:
```
Fri 17 May 19:23:31 CDT 2019: Mounting /mnt/archive...
mount: can't find /mnt/archive in /etc/fstab
Fri 17 May 19:23:31 CDT 2019: Failed to mount /mnt/archive.
Fri 17 May 19:23:31 CDT 2019: Attempts exhausted.
Fri 17 May 19:23:31 CDT 2019: Failed to mount cam archive.
Fri 17 May 19:23:31 CDT 2019: Couldn't connect archive, skipping archive step
```
Is the formatting of my `.conf` incorrect?
```
export ARCHIVE_SYSTEM=cifs
export archiveserver=192.168.1.150
export sharename=Media/Videos/TeslaCam
```

Could've sworn this worked previously?

marcone, 2019-05-18:
Looks OK to me, assuming you also specified the username and password to use.
To be honest I've never used a "share/folder/subfolder" type path, I've always used "share/folder".
The fact that /mnt/archive isn't in fstab indicates something went wrong during setup, so check the setup log in /boot.

xnaas, 2019-05-18:
Hmm. It's complaining about the CIFS version, but there's not a chance that's right.

marcone, 2019-05-18:
Does/did your password have special characters in it, and you forgot to quote it?

xnaas, 2019-05-18:
Are files supposed to save to /mnt/cam before going elsewhere? I’m trying to test the thing again after getting it setup but hitting the save icon on the Tesla I done see videos anywhere on the device and it doesn’t back anything up. /mutable/archiveloop.log doesn’t see any videos either.