marcone/teslausb #29: Headless setup 80% video 20% music

XmemphistoX, 2019-05-17:
Can't seem to get it going through.  Wifi is configured correctly and I am able to SSH in but checking the teslausb-headless-setup.log its stuck at Grabbing main setup file.

/boot/TESLAUSB_SETUP_STARTED is there but not TESLAUSB_SETUP_FINISHED

marcone, 2019-05-17:
When you're ssh'd in, but does `ifconfig` say?

XmemphistoX, 2019-05-17:
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.196  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 2601:646:8980:6810:a898:4eb:200e:5d02  prefixlen 64  scopeid 0x0<g

XmemphistoX, 2019-05-17:
Full ifconfig.

`
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.196  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 2601:646:8980:6810:a898:4eb:200e:5d02  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::469a:16ef:52df:c9ba  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:08:68:72  txqueuelen 1000  (Ethernet)
        RX packets 1654  bytes 232803 (227.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 989  bytes 125512 (122.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

`

marcone, 2019-05-17:
OK, so you're actually on wifi and not on the usb ethernet that is active during the first stage of setup.
I don't know why it wouldn't download the script then. Maybe Github had an outage when you ran setup, or your firewall is blocking access or something.
That log messages comes from /etc/rc.local, which will download the main setup script and run it.
You can try just running rc.local manually and see if you get any further error messages.

XmemphistoX, 2019-05-17:
Thanks, ran rc.local and its all good now.

marcone, 2019-05-17:
I'll just assume this was a github outage or something then.

closed 2019-05-17 by marcone