marcone/teslausb #30: music sync CIFS / Not all coming across

XmemphistoX, 2019-05-17:
I'm trying to sync my music over and its semi working.  Out of 100 songs, its only syncing across around 30.  Tried restarting the PI to start sync again and it says its finished copying even though it's not 1:1 synched.

Fri 17 May 20:02:50 BST 2019: Finished fsck on /mnt/music.
Fri 17 May 20:02:50 BST 2019: Copying music from  archive...
Fri 17 May 20:03:08 BST 2019: Unmounting /mnt/music...
Fri 17 May 20:03:09 BST 2019: Unmounted /mnt/music.
Fri 17 May 20:03:09 BST 2019: Finished copying music.

marcone, 2019-05-17:
I suspect it encountered an error while copying, and then silently failed.
Try running `/root/bin/copy-music.sh` manually to see what goes wrong.

XmemphistoX, 2019-05-17:
Getting this spammed. 

cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory
cp: cannot create regular file '/mnt/music/./tmp': Read-only file system
mv: cannot stat '/mnt/music/./tmp': No such file or directory

marcone, 2019-05-17:
So either the music drive is not mounted, or something went wrong with the underlying filesystem.
What do `df`, `mount` and `dmesg` say?

XmemphistoX, 2019-05-17:
pi@teslausb:~ $ df
Filesystem                   1K-blocks       Used  Available Use% Mounted on
/dev/root                      1668536    1072972     492756  69% /
devtmpfs                        217244          0     217244   0% /dev
tmpfs                           221568          0     221568   0% /dev/shm
tmpfs                           221568       5968     215600   3% /run
tmpfs                             5120          4       5116   1% /run/lock
tmpfs                           221568          0     221568   0% /sys/fs/cgroup
tmpfs                           221568         12     221556   1% /tmp
tmpfs                           221568         12     221556   1% /var/log
tmpfs                           221568          0     221568   0% /var/tmp
/dev/mmcblk0p1                   44529      22696      21834  51% /boot
/dev/mmcblk0p3                59421620   55324860    1048568  99% /backingfiles
/dev/mmcblk0p4                   96052       1671      87142   2% /mutable
//192.168.1.117/TeslaVideos 5860520956 1524687316 4335833640  27% /mnt/archive
//192.168.1.117/TeslaMusic  5860520956 1524687940 4335833016  27% /mnt/musicarchive
/dev/loop0                    11042480    6956664    4085816  63% /mnt/music
tmpfs                            44312          0      44312   0% /run/user/1000

XmemphistoX, 2019-05-17:
pi@teslausb:~ $ mount
/dev/mmcblk0p2 on / type ext4 (ro,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=217244k,nr_inodes=54311,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=31,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,relatime)
tmpfs on /var/log type tmpfs (rw,nosuid,nodev,relatime)
tmpfs on /var/tmp type tmpfs (rw,nosuid,nodev,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /backingfiles type ext4 (rw,noatime)
/dev/mmcblk0p4 on /mutable type ext4 (rw,relatime)
//192.168.1.117/TeslaVideos on /mnt/archive type cifs (rw,relatime,vers=3.0,cache=strict,username=hai,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.117,file_mode=0777,dir_mode=0777,soft,nounix,serverino,mapposix,rsize=1048576,wsize=1048576,echo_interval=60,actimeo=1)
//192.168.1.117/TeslaMusic on /mnt/musicarchive type cifs (rw,relatime,vers=3.0,cache=strict,username=xxx,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.117,file_mode=0777,dir_mode=0777,soft,nounix,serverino,mapposix,rsize=1048576,wsize=1048576,echo_interval=60,actimeo=1)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=44312k,mode=700,uid=1000,gid=1000)

XmemphistoX, 2019-05-17:
pi@teslausb:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.37+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1216 Thu May 2 13:25:42 BST 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:5D:3D:27 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=f06e2317-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2 quiet fastboot noswap ro
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434492K/458752K available (6931K kernel code, 633K rwdata, 2068K rodata, 456K init, 797K bss, 16068K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (6933 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 456 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 634 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25040 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000074] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000173] bcm2835: system timer (irq = 27)
[    0.000761] Console: colour dummy device 80x30
[    0.000789] console [tty1] enabled
[    0.000853] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060333] pid_max: default: 32768 minimum: 301
[    0.060872] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060893] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062297] CPU: Testing write buffer coherency: ok
[    0.063605] Setting up static identity map for 0x8200 - 0x8238
[    0.064989] devtmpfs: initialized
[    0.076243] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.076647] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.076677] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.078016] pinctrl core: initialized pinctrl subsystem
[    0.079585] NET: Registered protocol family 16
[    0.082289] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.089927] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.089944] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.090094] Serial: AMBA PL011 UART driver
[    0.093752] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.138341] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.140890] SCSI subsystem initialized
[    0.141341] usbcore: registered new interface driver usbfs
[    0.141450] usbcore: registered new interface driver hub
[    0.141629] usbcore: registered new device driver usb
[    0.161141] raspberrypi-firmware soc:firmware: Attached to firmware from 2019-03-27 15:45, variant start
[    0.171162] raspberrypi-firmware soc:firmware: Firmware hash is 2e98b31d18547962e564bdf88e57b3df7085c29b
[    0.183665] clocksource: Switched to clocksource timer
[    0.286829] VFS: Disk quotas dquot_6.6.0
[    0.286956] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.287272] FS-Cache: Loaded
[    0.287629] CacheFiles: Loaded
[    0.306580] NET: Registered protocol family 2
[    0.307906] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.307973] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[... 177 lines cut ...]
[   22.955089] CIFS VFS: cifs_mount failed w/return code = -101
[   23.979676] random: crng init done
[   23.979704] random: 7 urandom warning(s) missed due to ratelimiting
[   30.692604] Mass Storage Function, version: 2009/09/11
[   30.692624] LUN: removable file: (no medium)
[   30.692796] LUN: removable file: /backingfiles/music_disk.bin
[   30.692887] LUN: removable file: /backingfiles/cam_disk.bin
[... cut ...]
[  102.648136] LUN: removable file: /backingfiles/music_disk.bin
[  102.648229] LUN: removable file: /backingfiles/cam_disk.bin
[  102.648238] Number of LUNs=2
[  102.657408] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  102.657428] g_mass_storage gadget: g_mass_storage ready
[  102.657444] dwc2 20980000.usb: bound driver g_mass_storage
[  102.847736] dwc2 20980000.usb: new device is high-speed
[  102.877209] dwc2 20980000.usb: new address 33
[  103.593347] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage

XmemphistoX, 2019-05-17:
I do see this, however i see my files mounted in /mnt/musicarchive

[   22.936474] CIFS VFS: Error connecting to socket. Aborting operation.
[   22.936506] CIFS VFS: cifs_mount failed w/return code = -101
[   22.955057] CIFS VFS: Error connecting to socket. Aborting operation.
[   22.955089] CIFS VFS: cifs_mount failed w/return code = -101

marcone, 2019-05-17:
The reason you're not seeing anything in /mnt/music is because the drive is not mounted on the rpi, it's exposed as a usb drive to the car/PC
What's odd though is that `df` still shows the music drive:
```
/dev/loop0 11042480 6956664 4085816 63% /mnt/music
```
but `mount` does not.
Try restarting the pi again, it should try to copy the remaining files.
If that still doesn't work, you can manually mount the music drive:
```
modprobe -r g_mass_storage     # stops exporting the usb drives
mount /mnt/music               # mounts the music drive locally on the rpi
```
then you can run `/root/bin/copy-music.sh` to copy over the music files and see any errors.

XmemphistoX, 2019-05-17:
Got it.  Will try in a bit and report back.  Thanks for the help!

XmemphistoX, 2019-05-17:
I went ahead and started from scratch - when its automated, it doesn't copy all the files over, just around 30-40% of them and says its finished.  However if I run copy-music.sh manually, it pops a one line error?  It's skipping this one flac file but continuing with the rest.

I don't mind doing the sync manually if I can update it from my laptop - just an extra step.

'mv: cannot move '/mnt/music/./tmp' to a subdirectory of itself, '/mnt/music/LP.flac'

Thanks again!

marcone, 2019-05-18:
Do you have a file and/or folder named "tmp" in the music folder you're syncing?
The copy script first copies each file to "tmp", then renames it to its final name.
I should probably have picked a name that's less likely to clash...

XmemphistoX, 2019-05-18:
Yeah, I had a filename with a special character that i think was having issues renaming.  Leaving the TMP file behind in the folder.  Think this may be why.

Röyksopp

marcone, 2019-05-18:
I added some better logging and error checking to the music copy script, so it will skip over problematic files, and log the problem.

closed 2019-05-18 by marcone

Vipercat, 2019-06-01:
I noticed that files with specific characters in it's name (e.g. "ë" in Tiësto or "é" in Céline) stops the copying of files from the share to the RPI.

marcone, 2019-06-01:
@Vipercat looks like your setup is older (specifically, older than 15 days), since the current version of the music copying script is more error resilient, and while it may fail to copy files, it will just log the failure and continue on with the remaining files. I've verified that it in fact does so with files that contain "ë" and "é" characters.
I'll see if I can convert the filenames to something that works with a FAT filesystem.

To update your rpi to get the latest scripts, you should be able to just do:
```
sudo -i
killall archiveloop
modprobe -r g_mass_storage
/root/bin/setup-teslausb
reboot
```