marcone/teslausb #33: Creation of USB worked, but Tesla won't save files

RoxyWau, 2019-05-20:
Created the TeslaUSB with teslausb-20190505, which worked as it should. Can log into thru WiFi and if I connect the TeslaUSB to a computer I will see the folder.

When I connect the drive to the Tesla the icon on the MCU shows the normal red dot. When I press the icon, the icon changes to the download/save icon, but then there will popup the x icon. After a few seconds the red dot is present again. Nothing was saved on the TeslaUSB.

When I insert a normal USB stick everything works as it should. How can I troubleshoot this?
Can I post/check any logfiles?

marcone, 2019-05-20:
The main log is `/mutable/archiveloop.log`
The output of `dmesg` might also be relevant.

mikehaney24, 2019-05-21:
I'm having the what looks like the same issue--it worked the day I got everything up and running, and sometimes my car thinks it's writing something to disk, but there's nothing actually in storage.

main log output: https://pastebin.com/nN2kGbsR
dmesg output: https://pastebin.com/wLVfV7pa

marcone, 2019-05-22:
Your setup is still using a btrfs partition, which has been shown to cause problems, especially if you're using all the available space for recordings.
Flash the latest release and redo setup, and everything should work again.

mikehaney24, 2019-05-22:
I'm pretty sure I flashed the 5/5 release.

marcone, 2019-05-22:
It's not the release itself that matters, it's that flashing the release erases the partition table and cached setup scripts, allowing the newly downloaded setup scripts to recreate the partitions with the correct type.

mikehaney24, 2019-05-22:
Just for clarity, you're saying I should flash using HEAD, rather than the
5/5 release? Because that's the only release I've ever used.

marcone, 2019-05-22:
No, I'm saying that you should flash 20190505 again and go through the normal setup process after that. The main setup script will then download the latest versions of all the other scripts.

mikehaney24, 2019-05-22:
Ah, got it, I'll give it a shot.

mikehaney24, 2019-05-22:
It looks like the new setup scripts addressed it, but it worked the first day last time too. I'll update here if it fails after a couple days. Thanks!

AboveNBeyondCode, 2019-05-23:
used the headless setup here are my log files.
it seems the car will not save to the teslausb. 

I noticed the following:
first the icon hat the red dot. pressing it, the download icon appears. After a while shortly the x was displayed then the red dot again. Tying again the download was finished with the green arrow.

After a while the icon switched to the x again.

here are my log files:
/boot/teslausb-headless-setup.log  https://pastebin.com/1v01SwYu
/mutable/archiveloop.log  https://pastebin.com/76mdFxne
dmsg https://pastebin.com/QCLUdkFF

marcone, 2019-05-24:
I can't tell from the logs why you're getting the X icon or why nothing was saved.  The logs all look normal to me, and indicate there was nothing to back up.

marcone, 2019-06-03:
Nothing actionable here

closed 2019-06-03 by marcone