marcone/teslausb #34: Setup say to put WiFi password in single quotes, but example shows sharepassword

mobileagellc, 2019-05-21:
_Be sure that your WiFi password is enclosed in single quotes, and that if it contains one or more single quote characters you replace each single quote character with a backslash followed by a single quote character.

For example, if your password werea'b
you would have
export sharepassword='a\'b'_

Should this be:
**export WIFIPASS='a\'b'**
?

marcone, 2019-05-21:
Yes, the doc talks about the wifi password, but then uses the cifs share password in the example (they both have the same constraints though)

marcone, 2019-05-21:
Fixed

closed 2019-05-21 by marcone

mobileagellc, 2019-05-21:
Thank you. If the wifi network name has a space in it should it be placed inside single quotes?

marcone, 2019-05-21:
Yes, any password with special characters should be quoted.