marcone/teslausb #35: Problems? Read this first.

marcone, 2019-05-21:
If you're having trouble configuring teslausb, please check [the troubleshooting document on the wiki](https://github.com/marcone/teslausb/wiki#troubleshooting) before reporting an issue.
When you do file an issue, it's helpful to attach your `/boot/teslausb-headless-setup.log`, `/mutable/archiveloop.log` and the output of `dmesg`

mattblumberg, 2019-05-22:
Seems like the 'main setup files' are not available...

Mon  6 May 01:08:52 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:08:52 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Mon  6 May 01:08:52 BST 2019 : Rebooting...
Mon  6 May 01:09:18 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:09:18 BST 2019 : Grabbing main setup file.
Tue 21 May 06:17:16 BST 2019 : Detecting whether to update wpa_supplicant.conf
Tue 21 May 06:17:16 BST 2019 : Grabbing main setup file.

marcone, 2019-05-22:
At that point it's just trying to download the setup file from Github, so perhaps a Github outage?
If you run /etc/rc.local from the shell it might show some more information.

marcone, 2019-05-22:
Locking this bug because I don't want to start any discussions here. Please file new issues instead, if the troubleshooting document doesn't help resolve your issue.