marcone/teslausb #36: The headless setup instructions should mention that the Pi only works on 2.4 Ghz WiFi

coredumperror, 2019-05-21:
I just blew half my Sunday and a hefty chunk of today banging my head against the wall trying to set up my new Raspberry Pi using the instructions provided for this project. I finally discovered that the problem was that I have a two-band WiFI router, and the 2.4 and 5 Ghz networks use different SSIDs. I had not read *anywhere* (in this project, on Amazon where I bought the thing, or even on the main RaspberryPI site) that the Raspberry Pi Zero W only supports 2.4Ghz, so I'd been using the SSID that all my devices use, and had absolutely no luck with headless setup. 

I eventually managed to figure it out by hooking my Pi up to my monitor and keyboard and logging in directly, then futzing around with `raspi-config`. And don't get me started on the absurdity of the default keyboard config being UK instead of US, which completely breaks your ability to type pipe characters and double-quotes. Ugh that was a pain to find a fix for.

Anyway, seeing as how this project is likely to be encountered primarily by RaspberryPi newbies like myself, I'd simply like to request that it's instructions should mention the Pi Zero W's 2.4 Ghz WiFi band limitation.

I'd also love to know if you have any advice to help get `ssh pi@raspberrypi.local` to work. I've installed Bonjour, but that didn't help. I still get "Could not resolve hostname raspberrypi.local: No such host is known." The only way I can get into the Pi is by ssh'ing directly to it's WiFi IP (192.168.1.111 in my case). Which actually makes me wonder if the "USB Ethernet/RNDIS Gadget" is even doing anything. Or are the instructions regarding that just a hint about when the Pi has finished booting up?

Anyway, now that I've finally got networking functional, I'm going to keep trying to set up my Pi for TeslaUSB. Thanks for making such a cool tool!

marcone, 2019-05-21:
That's a good thing to know, thanks. I guess most people either already know it only supports 2.4 GHz, or use the same SSID for both their 2.4 and 5 GHz networks (I'm in the latter category myself), which is why yours is the first report of this. I'll add a note in the docs.

As for Bonjour and the .local address (not) working: note that the default name is "teslausb.local", not "raspberrypi.local". If you're using raspberrypi.local because you started with Raspbian, I highly recommend starting with [my latest prebuilt](https://github.com/marcone/teslausb/releases) instead, since that contains everything you need and is known to work.
AFAIK you don't need to install anything special anywhere to get the .local address to work. At least, it worked for me as-is when accessing the rpi from Windows, Mac and Linux machines.

Finally, I can't take credit for this project, since it is a fork of a (now abandoned/neglected) upstream project. I'm just trying to keep it up to date and make it better.

coredumperror, 2019-05-21:
I sure do wish that prebuilt image had actually worked when I tried it last night. Not sure yet what may have gone wrong, though, so I haven't got anything specific to report. Will try again after work tonight.

closed 2019-05-21 by coredumperror

coredumperror, 2019-05-23:
Update on install: success!

Turns out I'd *misspelled the teslausb config filename*. Ugh.

Still can't get the `teslausb.local` or `raspberrypi.local` hostnames to work, but I can SSH in via the IP address, so it's not a big deal.