marcone/teslausb #37: Feature Request: Use Tesla API to keep car awake until sync is complete

millercentral, 2019-05-22:
I'm a new teslausb user (thank you for all the work on this!), so take this suggestion with a grain of salt:

My typical use case (which I assume is typical for most) is that the teslausb will connect to my home network as I get home and park the car. The challenge is it seems like there is a good chance that the car could power down and go to sleep while teslausb is still syncing content, and I'd rather not mess with keeping Climate on, or Sentry mode while at home in the garage.

It seems like it would be interesting if teslausb was more aware of the state of the car, and could prevent it from going to sleep until the sync completes. Would integrating the Tesla car API (using something like this:https://www.teslaapi.io/) to understand the car state ("Am I parked?") and keep it awake while syncing (state poll, or maybe sending WakeUp command?). This could lead potentially to other interesting features like only syncing while charging, etc.

Anyway, perhaps this has been thought of / tried before, but seems like it would be a pretty cool addition to the project...

Thanks for considering!

marcone, 2019-05-22:
It's an interesting idea, and I'm happy to consider pull requests that implement this.
Seems easier to use a battery or OBD-2 power adapter though.

millercentral, 2019-05-22:
Is there a recommended battery with passthrough charging support? I don't believe the US version of the Tesla Model 3 has an OBD-2 port...

marcone, 2019-05-22:
I've never used a battery with passthrough, so can't offer any recommendations there. I *have* used a RAVPower 22000 mAh battery, which was enough to power the pi for about 3-4 days before needing to be charged again.

coredumperror, 2019-05-23:
This idea seriously intrigues me... No promises, but a PR might be on the way this weekend.

Any advice you can give for detecting when the Pi is actively archiving over WiFi would be appreciated. 

My first thought is to look for the existence of the `/mnt/cam/TeslaCam` folder, since that doesn't seem to be there unless the drive is currently archiving.

millercentral, 2019-05-23:
Just in case it helps, there is a pretty robust node.js wrapper for the Tesla API set... https://www.npmjs.com/package/teslajs

Let me know if you need a tester. :)

marcone, 2019-05-23:
@coredumperror you'd probably want to add it to the main logic in /root/bin/archiveloop, which basically does something like:

forever {
  1 wait for wifi
  2 disconnect drive from car and mount drive on rpi
  3 archive files
  4 unmount drive from rpi and connect drive to car
  5 wait for wifi to go away
}

step 2 can take a bit of time since it checks the drive for errors, and the TeslaCam folder won't be available until that is done, so you'd want to start making sure the car doesn't turn off after step 1. Then after step 4 you can let the car know it's OK to turn off.

senf42, 2019-05-23:
I've had great luck with the Voltaic V15 in always-on mode. It charges while I drive or use Sentry/climate, and has enough to keep the pi going when I get home before it runs out.

coredumperror, 2019-05-26:
Making good progress on this. I expect to have a working version ready for PR by the end of the day.

coredumperror, 2019-05-28:
Dangit, due to several unexpected delays, I haven't quite finished yet. Hopefully I'll be able to get the last little bit done soon. Though weekdays aren't the best days for me, when it comes to free time.

marcone, 2019-05-28:
I completely understand. I've been working on this for months, and *still* haven't gotten around to the thing I actually wanted to work on :)

marcone, 2019-06-03:
I'm going to close this issue because I won't be working on this, but pull requests are of course still welcome.

closed 2019-06-03 by marcone

marcone, 2019-06-05:
This is now available in the main-dev branch. Thanks, @coredumperror !

millercentral, 2019-07-30:
@coredumperror - I'm trying to debug an authorization error in my tesla_api use. In the logs I see "Unable to create access token", which I can replicate interactively on the pi:

`root@teslausb:~/bin# ./tesla_api.py --debug is_car_locked
Mon 29 Jul 18:09:00 PDT 2019: Loading mutable data from disk...
Mon 29 Jul 18:09:00 PDT 2019: Retrieving new API token...
Mon 29 Jul 18:09:00 PDT 2019: Unable to create access token:
Mon 29 Jul 18:09:00 PDT 2019: {'response': 'authorization_required_for_txid_``'}`

As far as I can tell, my credentials are correct in teslausb_setup_variables.conf:
`# TeslaUSB will only send your credentials to the Tesla API itself.
export tesla_email=james@millercentral.com
export tesla_password='tqkfWgCjFYe4'
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
export tesla_vin=# TeslaUSB will only send your credentials to the Tesla API itself.
export tesla_email=user@emailaddr.com    // obfuscated for post
export tesla_password='supersecret'           // obfuscated for post
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
export tesla_vin=5YJ3E1EXXXXXXXXXX`     // obfuscated for post

Any advice you can provide on how to debug this? Thanks!

smcasagrande, 2019-07-30:
I had the same problem - turns out I had a typo in the VIN.  VINs don't have the letter I (i) letter O (o), or letter Q (q) in order to avoid confusion.  If you have a 1 or 0 in your VIN and thought they were the letters above, fix your VIN in the config file and try again.

coredumperror, 2019-08-05:
Did you actually wrap your password in single-quotes? If so, that's the problem. I haven't gotten around to updating the code to parse that .conf file "correctly", so including single quotes in that env var definition makes the API code include the single quotes in your password.

Ironically, I think Tesla changed the way the car's sleep behavior affects the USB power, at least on the Model 3, shortly after I coded this whole API thing. I don't have my own API code running on the PI in my car (I developed it on a separate Pi), but it still manages to stay awake long enough to transfer all my recordings each night, even if there are a ton of them. Which is why I've been unmotivated to do further work on this.