marcone/teslausb #38: Feature Request

dmachadol, 2019-05-22:
While moving files and usb massivestorage are stopped, mount secondary massivestorage to able sentry mode to write new images. Until moving files are moved, swap massivestorage and perform again loop.

marcone, 2019-05-24:
Changing to a new drive while recording is active will cause the recording that's in progress at the time to be lost, plus it takes a little bit of time before the car will start recording on the new drive, so you might lose even more. So strictly from a recording perspective, it would be better to have a single drive of size X instead of two drives of size X/2.
Can you explain more about your use case? What are you actually trying to achieve?

dmachadol, 2019-05-24:
During the movement of files the unit is disconnected from the vehicle, therefore it is not possible to register new images until the copying process is finished and the unit is reassembled (in my case it has taken 10 minutes sometimes). The idea is that while this process lasts, have a vehicle with a disk to continue recording images.
Maybe an alternative is to mount 2 discs at a time with the TeslaCam folder, if one disappears the vehicle imagine that it would record images on the other disc.

marcone, 2019-05-24:
So you want to keep recording continuously, even while parked in wifi range. Do you simply want to record everything, always, or are there moments where it's OK for the car not to record?

XmemphistoX, 2019-05-24:
I don't see this happening in any meaningful way.  The disconnect only happens when you're at home connected to your wifi.  Even if you could write this in, I'm curious if the SD/USB2.0 write bandwidth is even enough for all that movement.

closed 2019-05-27 by marcone

dmachadol, 2019-06-03:
> So you want to keep recording continuously, even while parked in wifi range. Do you simply want to record everything, always, or are there moments where it's OK for the car not to record?
I want to record even in my community car park, where I have Wi-Fi range, the problem is that it tries to make movement of files constantly, even when I have new ones. That causes a disconnection of the USB in the vehicle that takes about 30 seconds to be available again, making impossible the recording of new images, since the file movement loop is executed constantly without a delay.

marcone, 2019-06-03:
That's not how teslausb works though. It only transfers files upon connection to wifi. When it is done transferring, it waits until it loses wifi, before starting the cycle again.
If you're seeing it constantly backing up files while it is parked, that means it is constantly disconnecting and reconnecting to wifi.

dmachadol, 2019-06-03:
Yes, its true. Solved this problem the second question.
Into loop, wen are connected and wait to disconnect wifi and archive folder, USB stay connected to car?

marcone, 2019-06-03:
Yes, while it is connected to wifi and waiting to disconnect from wifi, the USB drive is connected to the car.