marcone/teslausb #39: Archiveloop stuck?

xnaas, 2019-05-24:
This is my archiveloop for today:

```
Fri 24 May 16:05:09 CDT 2019: Archive is reachable.
Fri 24 May 16:05:29 CDT 2019: Archiving...
Fri 24 May 16:05:29 CDT 2019: Disconnecting usb from host...
Fri 24 May 16:05:29 CDT 2019: Disconnected usb from host.
Fri 24 May 16:05:29 CDT 2019: Ensuring cam archive is mounted...
Fri 24 May 16:05:29 CDT 2019: /mnt/archive is already mounted.
Fri 24 May 16:05:29 CDT 2019: Ensured cam archive is mounted.
Fri 24 May 16:05:29 CDT 2019: Starting...
Fri 24 May 16:05:29 CDT 2019: Ensuring cam file is mounted...
Fri 24 May 16:05:29 CDT 2019: Mounting /mnt/cam...
Fri 24 May 16:05:29 CDT 2019: Mounted /mnt/cam.
Fri 24 May 16:05:30 CDT 2019: Ensured cam file is mounted.
Fri 24 May 16:05:30 CDT 2019: Running fsck on /mnt/cam...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 389 files, 312620/3538228 clusters
Fri 24 May 16:05:32 CDT 2019: Finished fsck on /mnt/cam.
Fri 24 May 16:05:32 CDT 2019: Moving clips to archive...
Fri 24 May 16:05:32 CDT 2019: Creating output directory ''
Fri 24 May 16:05:32 CDT 2019: Creating output directory '2019-05-24_08-08-11'
Fri 24 May 16:05:32 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_07-58-57-front.mp4'
Fri 24 May 16:06:01 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_07-58-57-front.mp4'
Fri 24 May 16:06:01 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_07-58-57-left_repeater.mp4'
Fri 24 May 16:06:25 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_07-58-57-left_repeater.mp4'
Fri 24 May 16:06:25 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_07-58-57-right_repeater.mp4'
Fri 24 May 16:06:51 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_07-58-57-right_repeater.mp4'
Fri 24 May 16:06:51 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_07-59-57-front.mp4'
Fri 24 May 16:07:09 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_07-59-57-front.mp4'
Fri 24 May 16:07:09 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_07-59-57-left_repeater.mp4'
Fri 24 May 16:07:28 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_07-59-57-left_repeater.mp4'
Fri 24 May 16:07:28 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_07-59-57-right_repeater.mp4'
Fri 24 May 16:07:55 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_07-59-57-right_repeater.mp4'
Fri 24 May 16:07:55 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-00-57-front.mp4'
Fri 24 May 16:08:25 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-00-57-front.mp4'
Fri 24 May 16:08:25 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-00-57-left_repeater.mp4'
Fri 24 May 16:08:43 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-00-57-left_repeater.mp4'
Fri 24 May 16:08:43 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-00-57-right_repeater.mp4'
Fri 24 May 16:08:56 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-00-57-right_repeater.mp4'
Fri 24 May 16:08:56 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-01-58-front.mp4'
Fri 24 May 16:09:07 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-01-58-front.mp4'
Fri 24 May 16:09:07 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-01-58-left_repeater.mp4'
Fri 24 May 16:09:20 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-01-58-left_repeater.mp4'
Fri 24 May 16:09:20 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-01-58-right_repeater.mp4'
Fri 24 May 16:09:32 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-01-58-right_repeater.mp4'
Fri 24 May 16:09:32 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-02-58-front.mp4'
Fri 24 May 16:09:47 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-02-58-front.mp4'
Fri 24 May 16:09:47 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-02-58-left_repeater.mp4'
Fri 24 May 16:10:02 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-02-58-left_repeater.mp4'
Fri 24 May 16:10:02 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-02-58-right_repeater.mp4'
Fri 24 May 16:10:14 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-02-58-right_repeater.mp4'
Fri 24 May 16:10:14 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-03-58-front.mp4'
Fri 24 May 16:10:26 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-03-58-front.mp4'
Fri 24 May 16:10:26 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-03-58-left_repeater.mp4'
Fri 24 May 16:10:38 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-03-58-left_repeater.mp4'
Fri 24 May 16:10:38 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-03-58-right_repeater.mp4'
Fri 24 May 16:10:50 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-03-58-right_repeater.mp4'
Fri 24 May 16:10:50 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-04-58-front.mp4'
Fri 24 May 16:11:01 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-04-58-front.mp4'
Fri 24 May 16:11:01 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-04-58-left_repeater.mp4'
Fri 24 May 16:11:15 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-04-58-left_repeater.mp4'
Fri 24 May 16:11:15 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-04-58-right_repeater.mp4'
Fri 24 May 16:11:32 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-04-58-right_repeater.mp4'
Fri 24 May 16:11:32 CDT 2019: Moving '2019-05-24_08-08-11/2019-05-24_08-05-59-front.mp4'
Fri 24 May 16:11:50 CDT 2019: Moved '2019-05-24_08-08-11/2019-05-24_08-05-59-front.mp4'
[... 33 lines cut ...]
[505777.677020] ffc0: 00000000 00000005 00037ef0 0000015c 00038838 00000000 00037ef0 bedfc4ac
[505777.677032] ffe0: b6e53834 bedfc120 0001ef30 b6e53854
[505900.557062] INFO: task mv:2257 blocked for more than 120 seconds.
[505900.557084]       Tainted: G         C        4.19.37+ #1216
[505900.557091] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[505900.557101] mv              D    0  2257   2091 0x00000000
[505900.557165] [<c06c6890>] (__schedule) from [<c06c6d6c>] (schedule+0x50/0xa8)
[505900.557202] [<c06c6d6c>] (schedule) from [<c004e3e0>] (io_schedule+0x20/0x48)
[505900.557237] [<c004e3e0>] (io_schedule) from [<c0104d14>] (wait_on_page_bit+0x108/0x150)
[505900.557262] [<c0104d14>] (wait_on_page_bit) from [<c0104e54>] (__filemap_fdatawait_range+0xf8/0x148)
[505900.557284] [<c0104e54>] (__filemap_fdatawait_range) from [<c0104ecc>] (filemap_fdatawait_range+0x28/0x38)
[505900.557307] [<c0104ecc>] (filemap_fdatawait_range) from [<c01073a4>] (filemap_write_and_wait+0x68/0x9c)
[505900.557971] [<c01073a4>] (filemap_write_and_wait) from [<bf5c01c4>] (cifs_setattr+0x90/0x86c [cifs])
[505900.558397] [<bf5c01c4>] (cifs_setattr [cifs]) from [<c019005c>] (notify_change+0x1c4/0x40c)
[505900.558427] [<c019005c>] (notify_change) from [<c01a6dec>] (utimes_common+0xc0/0x198)
[505900.558451] [<c01a6dec>] (utimes_common) from [<c01a6fe0>] (do_utimes+0x11c/0x160)
[505900.558468] [<c01a6fe0>] (do_utimes) from [<c01a71c4>] (sys_utimensat+0x80/0xc0)
[505900.558490] [<c01a71c4>] (sys_utimensat) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[505900.558501] Exception stack(0xca87ffa8 to 0xca87fff0)
[505900.558517] ffa0:                   00000000 00000005 00000005 00000000 bedfc19c 00000000
[505900.558533] ffc0: 00000000 00000005 00037ef0 0000015c 00038838 00000000 00037ef0 bedfc4ac
[505900.558545] ffe0: b6e53834 bedfc120 0001ef30 b6e53854
```

Is it safe to reboot and keep all of my files? Can I manually restart the archiveloop script or some such? Tried looking through previous issues/FAQ, but I might be blind!

marcone, 2019-05-25:
It should be safe to reboot. It will resume where it left off.

xnaas, 2019-05-25:
Well, rebooted it and cannot ssh into it now. Still doesn't seem to be backing up. I'll have to manually power cycle the thing probably.

xnaas, 2019-05-25:
It seems to get to the final clip for a folder and then hangs. I have to keep rebooting it.

marcone, 2019-05-25:
What does the status LED show?
(one single blink per second means it's not connected to wifi, one double-blink per second means it's connected to wifi but not archiving, continuous rapid blink means it's checking the drive for errors and/or archiving)

xnaas, 2019-05-25:
Even when my ssh connections were being refused, the LED was blinking rapidly constantly. I stayed remoted into it this time typing `cat /mutable/archiveloop.log` periodically and it seems to be going through everything finally. Quite a few files today, so it's taking awhile, but at least it's going now.

I'll just chock it up to random unlucky fluke, I guess. ¯\\\_(ツ)_/¯

closed 2019-05-25 by xnaas