marcone/teslausb #40: Is it possible to setup just a shared network drive?

kcharwood, 2019-05-25:
I don't have any good options to sync and archive the footage, but I would like the ability to just be able to connect over wifi and look through the files and delete them remotely.

Is this possible in the current setup?

marcone, 2019-05-25:
Not currently, but I'm working on providing read-only access via a network share so you can at least browse the files. That should be available in the next few days.
After that I'll see if it's possible to easily extend that to read-write access.

kcharwood, 2019-05-26:
Great! Happy to be a tester of that!

marcone, 2019-05-26:
This is now available in the "samba" branch.
Doing the headless setup with these three lines in the config file might work for you:
```
export BRANCH=samba
export SAMBA_ENABLED=true
export CONFIGURE_ARCHIVING=false
```
(that last line should skip configuring archiving, though I've never tried this)

kcharwood, 2019-05-26:
Not seeing it come up as a USB drive at the moment.

Log File

```
Sun 26 May 21:21:05 BST 2019 : Verifying that the requested configuration is valid...
Sun 26 May 21:21:06 BST 2019 : Downloaded /tmp/verify-configuration.sh ...
Sun 26 May 21:21:06 BST 2019 : Verifying that there is sufficient space available on the MicroSD card...
Sun 26 May 21:21:08 BST 2019 : There is sufficient space available.
Sun 26 May 21:21:16 BST 2019 : Downloading additional setup scripts.
Sun 26 May 21:21:17 BST 2019 : Downloaded /tmp/create-backingfiles-partition.sh ...
Sun 26 May 21:21:18 BST 2019 : Downloaded /tmp/create-backingfiles.sh ...
Sun 26 May 21:21:18 BST 2019 : Downloaded /tmp/make-root-fs-readonly.sh ...
Sun 26 May 21:21:19 BST 2019 : Downloaded /tmp/configure.sh ...
Sun 26 May 21:21:19 BST 2019 : Fixing the modules-load parameter in /boot/cmdline.txt...
Sun 26 May 21:21:19 BST 2019 : Fixed cmdline.txt.
Sun 26 May 21:21:50 BST 2019 : Checking existing partitions...
Sun 26 May 21:21:52 BST 2019 : Modifying partition table for backing files partition...
Sun 26 May 21:21:52 BST 2019 : Modifying partition table for mutable (writable) partition for script usage...
Sun 26 May 21:21:53 BST 2019 : Writing updated partitions to fstab and /boot/cmdline.txt
Sun 26 May 21:21:53 BST 2019 : Formatting new partitions...
Sun 26 May 21:22:03 BST 2019 : Mounting the partition for the backing files...
Sun 26 May 21:22:03 BST 2019 : Mounted the partition for the backing files.
Sun 26 May 21:22:03 BST 2019 : Creating backing disk files.
Sun 26 May 21:22:03 BST 2019 : create-backingfiles: starting
Sun 26 May 21:22:03 BST 2019 : create-backingfiles: cam: 100%, music: 100%, mountpoint: /backingfiles
Sun 26 May 21:22:04 BST 2019 : create-backingfiles: Allocating 120984304K for /backingfiles/cam_disk.bin...
Sun 26 May 21:22:04 BST 2019 : create-backingfiles: Creating filesystem with label 'CAM'
Sun 26 May 21:22:07 BST 2019 : create-backingfiles: updated /etc/fstab for /mnt/cam
Sun 26 May 21:22:07 BST 2019 : create-backingfiles: created camera backing file
Sun 26 May 21:22:11 BST 2019 : create-backingfiles: done
Sun 26 May 21:22:20 BST 2019 : Main setup completed.
Sun 26 May 21:22:20 BST 2019 : skipping configure.sh
Sun 26 May 21:22:21 BST 2019 : Downloaded /tmp//configure-samba.sh ...
Sun 26 May 21:22:22 BST 2019 : Downloaded /root/bin/smb-on.sh ...
Sun 26 May 21:22:22 BST 2019 : Downloaded /root/bin/smb-off.sh ...
Sun 26 May 21:21:40 BST 2019 : Detecting whether to update wpa_supplicant.conf
```

marcone, 2019-05-26:
That's where the "I've never tried this" is relevant :)
What probably happened is that without configuring an archive method, it doesn't download and start the archiveloop script, which is normally the thing that activates the USB drive emulation.
Try ssh'ing into the device and running `modprobe g_mass_storage`

kcharwood, 2019-05-26:
That seems to have worked in terms of making it appear as a USB drive on my machine.

Can't seem to figure out how to setup a smb://  link to the drive.

Should it just be smb://ip_address? I could also make it available to you to remotely ssh into if you wanted to use it as a test target.

marcone, 2019-05-26:
On Mac and Linux you should be able to access smb://ip_address or smb://hostname, and on Windows with \\\\ip_address or \\\\hostname
It should then show a TeslaCam share, and if you try to access that it will ask you to log in. The default user/pass are the same as for ssh: pi/raspberry. You can also enable guest mode (no password needed) by adding `export SAMBA_GUEST=true` to the .conf file.

kcharwood, 2019-05-26:
Ya that’s what I was thinking, but neither of the smb links work (ip or hostname)

marcone, 2019-05-26:
Does "service --status-all" show that the smbd service is active?

kcharwood, 2019-05-26:
Stepped away. Will check tonight!

kcharwood, 2019-05-27:
```
service --status-all
 [ + ]  alsa-utils
 [ + ]  avahi-daemon
 [ + ]  bluetooth
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  dbus
 [ + ]  dhcpcd
 [ + ]  dphys-swapfile
 [ + ]  fake-hwclock
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  networking
 [ - ]  nfs-common
 [ - ]  nmbd.dpkg-new
 [ - ]  paxctld
 [ - ]  plymouth
 [ - ]  plymouth-log
 [ + ]  procps
 [ + ]  raspi-config
 [ ? ]  rng-tools
 [ - ]  rpcbind
 [ - ]  rsync
 [ + ]  rsyslog
 [ - ]  samba-ad-dc.dpkg-new
 [ - ]  samba.dpkg-new
 [ - ]  smbd.dpkg-new
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  triggerhappy
 [ + ]  udev
```

marcone, 2019-05-27:
hmm, why is your smbd called "smbd.dpkg-new" ? Mine's just "smbd":
```
 [ + ]  alsa-utils
 [ + ]  avahi-daemon
 [ + ]  bluetooth
 [ - ]  busybox-klogd
 [ + ]  busybox-syslogd
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  dbus
 [ + ]  dhcpcd
 [ - ]  exim4
 [ + ]  fake-hwclock
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  networking
 [ - ]  nfs-common
 [ + ]  nmbd
 [ + ]  ntp
 [ - ]  paxctld
 [ - ]  plymouth
 [ - ]  plymouth-log
 [ + ]  procps
 [ + ]  raspi-config
 [ ? ]  rng-tools
 [ - ]  rpcbind
 [ - ]  rsync
 [ + ]  samba
 [ - ]  samba-ad-dc
 [ + ]  smbd
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  udev
```
You could try starting your version manually: `service smbd.dpkg-new start` and see if that works for you.

kcharwood, 2019-05-27:
Tried `smbd.dpkg-new`, and it failed.

Tried `smbd`, and got some errors:

```
May 27 13:53:19 teslausb sudo[16830]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/sbin/service smbd.dpkg-new start
May 27 13:53:19 teslausb sudo[16830]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
May 27 13:53:21 teslausb sudo[16830]: pam_unix(sudo:session): session closed for user root
May 27 13:53:34 teslausb sudo[16860]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/sbin/service smbd start
May 27 13:53:34 teslausb sudo[16860]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
May 27 13:53:36 teslausb systemd[1]: Starting Samba SMB Daemon...
-- Subject: Unit smbd.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit smbd.service has begun starting up.
May 27 13:53:38 teslausb systemd[1]: smbd.service: Main process exited, code=exited, status=1/FAILURE
May 27 13:53:38 teslausb systemd[1]: Failed to start Samba SMB Daemon.
-- Subject: Unit smbd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit smbd.service has failed.
-- 
-- The result is failed.
May 27 13:53:38 teslausb systemd[1]: smbd.service: Unit entered failed state.
May 27 13:53:38 teslausb systemd[1]: smbd.service: Failed with result 'exit-code'.
May 27 13:53:38 teslausb sudo[16860]: pam_unix(sudo:session): session closed for user root
```

To clarify, I downloaded the latest image from the repo, flashed with etcher, then created the config file with the branch name set to `samba` and did the setup.

Let me know if I should try again from scratch or try anything else.

marcone, 2019-05-27:
Where's that log from? (I don't seem to have any logs on my device that look like that)
When you say "the latest image from the repo", you mean https://github.com/marcone/teslausb/releases/tag/v1.2a ?
Might be worth flashing that again, and if it still doesn't work, attach /boot/teslausb-headless-setup.log here.

kcharwood, 2019-05-27:
Yep, that's the image I'm using.

To get the log, I ran the following:

```
sudo service smbd start
```

It returned the following error:

```
Job for smbd.service failed because the control process exited with error code.
See "systemctl status smbd.service" and "journalctl -xe" for details.
```

I used `systemctl status smbd.service` to see that error.

I'll try flashing from scratch again and report back.

kcharwood, 2019-05-27:
Reflashed and re-setup. I see `TESLAUSB_SETUP_FINISHED`.

The USB drive did not automatically start up as noted above.

The log is as follows:

```
Mon  6 May 01:08:53 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:08:53 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Mon  6 May 01:08:53 BST 2019 : Rebooting...
Mon  6 May 01:09:23 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:09:23 BST 2019 : Grabbing main setup file.
Mon  6 May 01:09:24 BST 2019 : Starting setup.
Mon  6 May 01:09:24 BST 2019 : reading config from /root/teslausb_setup_variables.conf
Mon  6 May 01:09:24 BST 2019 : Updating package index files...
Mon 27 May 16:14:37 BST 2019 : Verifying that the requested configuration is valid...
Mon 27 May 16:14:38 BST 2019 : Downloaded /tmp/verify-configuration.sh ...
Mon 27 May 16:14:38 BST 2019 : Verifying that there is sufficient space available on the MicroSD card...
Mon 27 May 16:14:39 BST 2019 : There is sufficient space available.
Mon 27 May 16:14:48 BST 2019 : Downloading additional setup scripts.
Mon 27 May 16:14:49 BST 2019 : Downloaded /tmp/create-backingfiles-partition.sh ...
Mon 27 May 16:14:49 BST 2019 : Downloaded /tmp/create-backingfiles.sh ...
Mon 27 May 16:14:50 BST 2019 : Downloaded /tmp/make-root-fs-readonly.sh ...
Mon 27 May 16:14:51 BST 2019 : Downloaded /tmp/configure.sh ...
Mon 27 May 16:14:51 BST 2019 : Fixing the modules-load parameter in /boot/cmdline.txt...
Mon 27 May 16:14:51 BST 2019 : Fixed cmdline.txt.
Mon 27 May 16:15:26 BST 2019 : Checking existing partitions...
Mon 27 May 16:15:28 BST 2019 : Modifying partition table for backing files partition...
Mon 27 May 16:15:28 BST 2019 : Modifying partition table for mutable (writable) partition for script usage...
Mon 27 May 16:15:28 BST 2019 : Writing updated partitions to fstab and /boot/cmdline.txt
Mon 27 May 16:15:28 BST 2019 : Formatting new partitions...
Mon 27 May 16:15:38 BST 2019 : Mounting the partition for the backing files...
Mon 27 May 16:15:39 BST 2019 : Mounted the partition for the backing files.
Mon 27 May 16:15:39 BST 2019 : Creating backing disk files.
Mon 27 May 16:15:39 BST 2019 : create-backingfiles: starting
Mon 27 May 16:15:39 BST 2019 : create-backingfiles: cam: 100%, music: 100%, mountpoint: /backingfiles
Mon 27 May 16:15:39 BST 2019 : create-backingfiles: Allocating 120984304K for /backingfiles/cam_disk.bin...
Mon 27 May 16:15:40 BST 2019 : create-backingfiles: Creating filesystem with label 'CAM'
Mon 27 May 16:15:42 BST 2019 : create-backingfiles: updated /etc/fstab for /mnt/cam
Mon 27 May 16:15:42 BST 2019 : create-backingfiles: created camera backing file
Mon 27 May 16:15:47 BST 2019 : create-backingfiles: done
Mon 27 May 16:15:56 BST 2019 : Main setup completed.
Mon 27 May 16:15:56 BST 2019 : skipping configure.sh
Mon 27 May 16:15:57 BST 2019 : Downloaded /tmp//configure-samba.sh ...
Mon 27 May 16:15:57 BST 2019 : Downloaded /root/bin/smb-on.sh ...
Mon 27 May 16:15:58 BST 2019 : Downloaded /root/bin/smb-off.sh ...
```

⚠️ I noticed that maybe the configure-samba script is the wrong path? It has double slashes. **Possibly part of the problem?**

`Mon 27 May 16:15:57 BST 2019 : Downloaded /tmp//configure-samba.sh ...`

Running `sudo service --status-all` now shows the "correct" smbd:

```
 [ - ]  alsa-utils
 [ + ]  avahi-daemon
 [ + ]  bluetooth
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  dbus
 [ + ]  dhcpcd
 [ + ]  fake-hwclock
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  networking
 [ - ]  nfs-common
 [ + ]  nmbd
 [ - ]  paxctld
 [ - ]  plymouth
 [ - ]  plymouth-log
 [ + ]  procps
 [ + ]  raspi-config
 [ ? ]  rng-tools
 [ - ]  rpcbind
 [ - ]  rsync
 [ + ]  rsyslog
 [ - ]  samba
 [ - ]  samba-ad-dc
 [ - ]  smbd
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  udev
```

Attempting to start smbd:

```
sudo service smbd start
Job for smbd.service failed because the control process exited with error code.
See "systemctl status smbd.service" and "journalctl -xe" for details.
```

Looking at the error as suggested:

```
pi@teslausb:/boot $ systemctl status smbd.service
● smbd.service - Samba SMB Daemon
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-05-27 16:20:59 BST; 35s ago
     Docs: man:smbd(8)
           man:samba(7)
           man:smb.conf(5)
  Process: 4536 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=exited, status=1/FAILURE)
 Main PID: 4536 (code=exited, status=1/FAILURE)

May 27 16:20:57 teslausb systemd[1]: Starting Samba SMB Daemon...
May 27 16:20:59 teslausb systemd[1]: smbd.service: Main process exited, code=exited, status=1/FAILURE
May 27 16:20:59 teslausb systemd[1]: Failed to start Samba SMB Daemon.
May 27 16:20:59 teslausb systemd[1]: smbd.service: Unit entered failed state.
May 27 16:20:59 teslausb systemd[1]: smbd.service: Failed with result 'exit-code'.
```

kcharwood, 2019-05-27:
looking through `configure-samba.sh`, i see the generated files from the script in their right places, so I think that script is running up until the the start call, then failing. I'm not seeing any other logging from there.

Is there is an easy way to "re-run" setup without having to re-flash?

marcone, 2019-05-27:
This should be fixed now. You can either re-flash, or run these commands to fix your existing install:
```
sudo -i
mkdir /tmp/foo
mount --bind -o rw / /tmp/foo/
mv /tmp/foo/mutable/varlib/samba /mutable/varlib/
reboot
```
Sorry, this isn't one of the things that can be fixed by running setup again.

kcharwood, 2019-05-27:
That worked!!

Now I just had to run `modprobe g_mass_storage` to get it to kick into the USB mode. What do I need to do get that going automatically?

marcone, 2019-05-27:
Probably just put that at the end of /etc/rc.local

kcharwood, 2019-05-28:
Add that to the bottom of the file, and its looking good!!!

Next step is to make it writable via smb to remotely delete things?

marcone, 2019-05-28:
Yes, though to be honest that's low priority for me, and I have other things I want to work on first.

kcharwood, 2019-05-28:
No worries. If you can point me in the right direction on where I should do that, happy to dig in. This issue has already been a good exercise at understanding some things under the covers.

marcone, 2019-05-28:
Maybe something like this would work:
- in /etc/samba/smb.conf, make the samba share read-write
- in /root/bin/smb-on.sh, get a list of all the files in the snapshot about to be exported
- in /root/bin/smb-off.sh, get a list of all the files still in the snapshot
- diff the two lists to build a new list of all the files that were deleted
- remove those files from the actual disk image

The tricky part is to ensure that that last step does not interfere with any archiving activity going on in parallel (not something that applies to your situation, but something that would have to be handled if you're considering turning this into a pull request)

marcone, 2019-06-03:
read-only samba access is now available in the main branch, and you can set the archive method to "none" in the headless setup config file if you don't want to configure an archive method. The rpi will just function as a USB drive that can also be accessed as a network share in that case.

closed 2019-06-03 by marcone

rggnkmp, 2019-08-18:
I am lost here, hence re-opening it.
Re-flashed and redid the setup a gazillion times now, also tried all the steps above. And with all this manual intervention, I get it to work (= mounting /CAM on my Mac; on occasion also with SMB active AND working).
However, on reboot … back to square one.

/etc/rc.local says 
`Setup doesn't seem to have completed, there is no /root/bin/archiveloop.`

Anything I need to change / add to config to make this work right away?

Am using:
```
export CONFIGURE_ARCHIVING=false
export ARCHIVE_SYSTEM=none
```
and 
```
export SAMBA_ENABLED=true
export SAMBA_GUEST=true
```

Am left clueless …

marcone, 2019-08-18:
Try:
```
export CONFIGURE_ARCHIVING=true
export ARCHIVE_SYSTEM=none
```