marcone/teslausb #41: Would it be possible to make the timestamps be more accurate in the archiveloop.log file?

coredumperror, 2019-05-26:
I have no idea if there's any realistic way to fix this, but I've found that `/mutable/archiveloop.log` has completely wrong timestamps when the Pi first boots up. The system thinks it's November 2016 for some reason. 

These timestamps fix themselves eventually during the archiving process (`ntp` finally runs, I guess?), but I often find myself SSHing into my pi when I get home to confirm that it's working, and the timestamps being totally wrong make it impossible to tell if the log lines are new or old.

marcone, 2019-05-26:
The Raspberry Pi Zero W does not have a battery backed hardware clock, so it tries to preserve some semblance of correct time by saving the current time to a file on shutdown, and restoring it on boot.
Unfortunately, that mechanism doesn't seem to actually work correctly. It doesn't bother me too much so it hasn't been a priority for me. Fixes are welcome. (see `man fake-hwclock` for an overview of how it's supposed to work)

marcone, 2019-05-26:
For what it's worth: when I looked into it a little bit just now I found that /mutable/etc/fake-hwclock.data does get updated, but it gets updated with the wrong time. For example:
```
~# cat /mutable/etc/fake-hwclock.data 
2016-11-03 17:17:02
~# service fake-hwclock restart
root@teslausb:~# cat /mutable/etc/fake-hwclock.data 
2019-05-26 16:24:33
~# reboot
...
~# cat /mutable/etc/fake-hwclock.data 
2016-11-03 17:17:01
```

closed 2019-05-26 by marcone

coredumperror, 2019-05-27:
Odd. I don't have any files at all in `/mutable/etc/` (the folder exists, though). I do have `/etc/fake-hwclock.data`, though I have no idea how it's being written to, since the fs is supposed to be readonly... right?

marcone, 2019-05-27:
Right, which is why the setup moves /etc/fake-hwclock.data to /mutable/etc/fake-hwclock.data, then creates a symbolic link from one to the other.
The only way I can see how it would create the /mutable/etc folder but then not move the .data file and create the link is if /etc/fake-hwclock didn't exist during setup, but I don't know how that could happen.

coredumperror, 2019-05-28:
Is there any logging I could look for to diagnose this issue?

marcone, 2019-05-28:
There's the setup log at /boot/teslausb-headless-setup.log, but other than that I don't know.