marcone/teslausb #42: Failed mounting in archiveloop, after one step flash

seandoesgis, 2019-05-26:
```
Sun 26 May 18:16:38 BST 2019: Mounting /mnt/archive...
mount error(22): Invalid argument
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Sun 26 May 18:16:38 BST 2019: Failed to mount /mnt/archive.
Sun 26 May 18:16:38 BST 2019: Sleeping before retry...
```
 How would I find the invalid argument?  Only thing I've been able to find online is the CIFS version may need to be changed to something like 2.1.  How would I be able to try that.

myquealer, 2019-05-26:
Make sure the folder you're mounting has no spaces or other special characters in its name. This tripped me up for a while, and I think this is the error I was getting.

The documentation should make this requirement clear, or just surround each mounting argument in quotes so it works with spaces.

marcone, 2019-05-26:
Does /mnt/archive exist?
Does it appear in /etc/fstab?
Are the username and password in /root/.teslaCamArchiveCredentials correct?

seandoesgis, 2019-05-26:
> Does /mnt/archive exist?
> Does it appear in /etc/fstab?
> Are the username and password in /root/.teslaCamArchiveCredentials correct?

```
root@teslausb:~# cd /mnt/archive
root@teslausb:/mnt/archive# ls -a
.  ..
root@teslausb:/mnt/archive# cd /etc/fstab
-bash: cd: /etc/fstab: Not a directory
root@teslausb:/mnt/archive# 
```

marcone, 2019-05-26:
/etc/fstab is a file, not a directory. Check the content of that file and see if there's an entry for /mnt/archive in there

seandoesgis, 2019-05-26:
> /etc/fstab is a file, not a directory. Check the content of that file and see if there's an entry for /mnt/archive in there

`//192.168.1.20/TeslaCam /mnt/archive cifs vers=3.0,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0`

marcone, 2019-05-26:
That looks fine. You say your share is on a Mac though. Another user had problems with shared drive on MacOS too (see https://github.com/marcone/teslausb/issues/27), and while it should in theory be as easy as telling the Mac to use SMB instead of AFP, I couldn't get it to work myself either.

seandoesgis, 2019-05-26:
> That looks fine. You say your share is on a Mac though. Another user had problems with shared drive on MacOS too (see #27), and while it should in theory be as easy as telling the Mac to use SMB instead of AFP, I couldn't get it to work myself either.

Doh...thanks marcone, I'll have to try rclone4pi or something.

closed 2019-05-27 by marcone

myquealer, 2019-05-27:
I set up TeslaUSB with a Mac and it is saving to a Mac. I lowered gradually  CIFS version to 1.0, and then realized other problems were preventing it from working. It looks like you're still on CIFS 3.0. You might try 2.1, 2.0 or 1.0 and see if it works for you. I'm still on 1.0, since it's working for me.

seandoesgis, 2019-05-28:
> I set up TeslaUSB with a Mac and it is saving to a Mac. I lowered gradually CIFS version to 1.0, and then realized other problems were preventing it from working. It looks like you're still on CIFS [...]

I actually got it working today on the Mac with the default CIFS version 3.0.  When entering in variables for the config I simply used no quotes for the password or others.  I also made sure the folder have caps.

`export ARCHIVE_SYSTEM=cifs
export archiveserver=192.168.1.20
export sharename=teslacam
export shareuser=ssssss
export sharepassword=xxxxxx
export camsize=95%
export SSID=p00p
export WIFIPASS='111111111'
export HEADLESS_SETUP=true
export timezone=America/New_York
`