marcone/teslausb #43: Alarm going off every morning at 7:00 AM

myquealer, 2019-05-26:
For the past couple days, since upgrading my Tesla to 2019.16.2 and auto-enabling Sentry Mode my Tesla alarm has gone off at 7:00 AM with no external disturbances.

I strongly suspect TeslaUSB is unmounting the USB drive from the Tesla at this time, perhaps a daily restart or some other cron-driven maintenance. The car is seeing the USB unmount and triggering the alarm with horn blaring and music blasting.

If I run 'date' when sshed to TeslaUSB it returns a time 8 hours ahead of me with a timezone of BST, but the created/modified time of files saved to my server are seven hours behind, so the alarm is going off right when the timestamp on the files would be midnight.

If there's no way to avoid the unmounting, perhaps TeslaUSB could be updated to use the Tesla API to turn off Sentry Mode before the restart and reenable it after the restart.

marcone, 2019-05-26:
Teslausb doesn't have an automated restart or anything like that. Can you try ssh'ing into the device and running `sudo modprobe -r g_mass_storage` while sentry mode is active, and see if that triggers the alarm too?

marcone, 2019-05-26:
Does "service --status-all" show that the smbd service is active?

myquealer, 2019-05-27:
@marcone Yes, several seconds after running `sudo modprobe -r g_mass_storage` the alarm sounds if Sentry Mode is enabled.

marcone, 2019-05-27:
I don't have 2019.16.2 yet, and it doesn't happen for me with 2019.12.1.2.
Does your rpi actually restart when this happens at 7am? (i.e. if you ssh in afterwards, does `uptime` show it just booted?)
Have you tried if it happens with just a regular usb drive?
Does the problem still happen if you set your timezone correctly, or does it happen at a different time?

myquealer, 2019-05-27:
I can't check `uptime` until next time it happens since the Pi turned off after I turned Sentry Mode off. 
I've not checked with a regular USB drive, but I think there would be widespread reports of this if that were the case. Everyone in the Reddit thread who it's happened to is also using TeslaUSB. Since it only happens once a day I'm limited in the testing I can do.
I just changed my timezone using raspi-config. I'll let you know what time it happens next.

schmug, 2019-05-27:
I suspect this is happening to me too but I was planning on rebuilding it first. Checking with Tesla support they could not find a reason for why it went off. They did mention "USB being removed too quickly" but when I mentioned a custom pi drive they dismissed it and thought the sensitivity was too high for 16.2.

marcone, 2019-05-27:
How are you both powering the Pi? Just USB, battery, hardwired?

senf42, 2019-05-27:
I haven't had my alarm go off at a set time, but it has gone off suspiciously over the last few days. I've either had no recording show up, or just a blank folder with the time. I'm powering mine with a battery pack (Voltaic V15 in always-on).

myquealer, 2019-05-27:
Mine is just USB powered, no battery.

ken830, 2019-05-27:
I also commented on the reddit thread previously. Mine happened at exactly 10pm PDT. Also note I'm running a modified version of the original cimryan. Mine has the data USB cable plugged into the USB port and the "power" USB cable plugged into the cig-adapter USB power.

mattster98, 2019-05-28:
Model of car may be a relevant variable here.  My X running 2019.16.2 hasn't had this problem happening spontaneously but I discovered it today when I decided to clean up the saved clips folder remotely.  Alarm went off a few seconds after I removed the mass storage module.  Came here to share in case it was a fresh discovery (obviously not).  So WAI from my perspective.

acraigl, 2019-05-28:
Glad it's not just me (Model 3 2019.16.2), though it's not at any set time. Rebooting the pi remotely while in sentry mode (or even sudo modprobe) will set off the alarm. This morning. I disabled sentry mode, waited a bit, then re-enabled. I was shocked that set off the alarm as well, much to my wife's dismay.

No external power on mine -- just when the front USBs are hot (i.e. car is on/sentry mode).

Vipercat, 2019-05-28:
I've noticed several times that the alarm went off when I 1) unmounted the TeslaCam USB drive (sudo modprobe -r g_mass_storage) and 2) I'm connected to my home WiFi. My setup is the data USB cable plugged into the USB port and the "power" USB cable plugged into the cig-adapter USB power. I'm on 2019.16.2. I didn't have any alarm go off when I was on 2019.12.1.2

schmug, 2019-05-28:
> How are you both powering the Pi? Just USB, battery, hardwired?

I have a nomad, splitter to pi. USB

mattster98, 2019-05-28:
Single USB cable between the car and the USB port on the raspi.

Vipercat, 2019-05-28:
Dual USB Cable: The USB Cable for 'cam and music' connected to one of the regular front USB ports (Model S) and the 'power usb port' connected to the cigarette lighter outlet. Made sure that I first connected the 'power usb port' and then the 'cam and music USB port. The reason for connecting in this order is that I noticed that connecting in ad different order resulted in the 'power usb port's power not being used (still following?)

Vipercat, 2019-05-28:
Update: I've been able to reproduce the alarm off (or Sentry being triggered). I'm using the app Rasbi Check where you're able to connect with the RPI using your shared WiFi netwerk and you can configure commands. As soon as Sentry is enabled and I unmount the 'cam/music USB disk' (via sudo modprobe -r g_mass_storage) the alarm is triggered approximately 10 seconds after. So depending on the mechanism the TeslaUSB program on the RPI is checking WiFi and/or footage to be copied from the /mnt/cam it could trigger the Sentry alarm. Not sure. Marcone is the one that knows best how the archiveloop process works, but I've definitely been able to trigger the alarm. Therefore I'm a bit concerned on keeping Sentry on while in range of my WiFi network, because I don't want the alarm to trigger in the middle of the night. One thing is for sure: I had Sentry enabled yesterday while parked in another city and Sentry was activated by someone walking by (no alarm) and nothing happened. No alarm triggered.

Vipercat, 2019-05-28:
Typo: the App is called RasPi Check.

myquealer, 2019-05-28:
After fixing the timezone on my Pi the alarm has not triggered at a particular time. Whatever was causing the USB to dismount at 7 AM seems to be fixed by that (unless I was driving when the unmount happened, though it's been home most of the time).

acraigl, 2019-05-28:
> After fixing the timezone on my Pi the alarm has not triggered at a particular time. Whatever was causing the USB to dismount at 7 AM seems to be fixed by that (unless I was driving when the unmount happened, though it's been home most of the time).

I'm not convinced that changing timezone would help, but just in case, how did you do it. :)

marcone, 2019-05-29:
> I'm not convinced that changing timezone would help, but just in case, how did you do it? :)

/etc/localtime is a link to the timezone file under /usr/share/zoneinfo/
Simply point it at the right one, e.g. `ln -sf /usr/share/zoneinfo/America/Los_Angeles /etc/localtime`

marcone, 2019-05-29:
Also for what it's worth: I've had my teslausb rpi connected to a computer for the past 30 hours, and it has neither rebooted nor caused the computer to see the drive disconnect.

I wonder if there could be two different things at play here: the car could cause the USB port to become unpowered briefly, which would cause a USB-powered rpi to reboot, or the rpi could lose and immediately regain wifi connectivity, which would cause a battery-powered rpi to disconnect the drive to start archiving.

phtp, 2019-05-29:
I had my alarm randomly go off the other day and didn't even think the two could be related. Will keep following.

gummby8, 2019-05-29:
> Update: I've been able to reproduce the alarm going off (or Sentry being triggered). I'm using the app RasPi Check where you're able to connect with the RPI using your shared WiFi netwerk and you can configure commands. As soon as Sentry is enabled and I unmount the 'cam/music USB disk' (via sudo modprobe -r g_mass_storage) the alarm is triggered approximately 10 seconds after. So depending on the mechanism the TeslaUSB program on the RPI is checking WiFi and/or footage to be copied from the /mnt/cam it could trigger the Sentry alarm. Not sure. Marcone is the one that knows best how the archiveloop process works, but I've definitely been able to trigger the alarm. Therefore I'm a bit concerned on keeping Sentry on while in range of my WiFi network, because I don't want the alarm to trigger in the middle of the night. One thing is for sure: I had Sentry enabled yesterday while parked in another city and Sentry was activated by someone walking by (no alarm) and nothing happened. No alarm triggered.

I see an issue with this in a few situations, but 1 in particular. Since the Pi unmounts the volume to copy files the tesla thinks someone stole the USB stick and thus triggers the alarm. 

The pi automaticly triggers a sync when it detects the wifi goes away, and then comes back. Untill teh wifi goes away again, it won't trigger a sync, and unmount the volume, and trigger the alarm

So suppose your wifi cuts out in the middle of the night, firmware update, power  outage ect...
Your car alarm is going to go off....

So yes somehow we need the pi to disable sentry mode during a sync or this is going to be an ongoing issue. Or we all need batteries to keep the pi active for a few min in the event the USB shut off because sentry mode is no longer active.

acraigl, 2019-05-29:
@gummby8 I agree with you. 
I'm planning on disabling Sentry while at home, and using Tasker to enable it for 20 minutes, then disable it again. Likely at 10pm or something so that it doesn't wake me up in the middle of the night. That way it can sync without additional power and won't go off accidentally (hopefully).

myquealer, 2019-05-29:
So much for changing the timezone making a difference. It just went off again at 7:00.

@gummby8 said the Pi triggers a sync when Wi-Fi goes away and comes back and that it unmounts the volume to copy the files. I think that's the answer.  I have two wi-fi networks setup using the same hardware. One network is active 24 hours a day, the other shuts off between 10:30 PM and 7:00 AM (or else my kids would be active 24 hours a day...). The TeslaPi is on the 24 hour network. I'm guessing the WAPs turning the other network on causes enough of a hiccup on the 24 hour network that the Pi triggers a sync.

Using the Tesla API to disable Sentry mode before starting a sync, and then enable it after the sync seems like the solution.

BTW, to close loose ends, `uptime` shows the Pi has been up for 46 hours, the alarm went off 14 minutes ago, so it is not restarting daily as I initially theorized.

myquealer, 2019-05-29:
archiveloop.log made it more clear what's going on and I think allowed me to fix the 7:00 issue. The computer I'm saving files to was on the kids' network, so losing network connection at night until 7:00 AM. When it came back online, TeslaUSB saw it was reachable again so started a sync, which unmounts the USB without warning to the Tesla.

Vipercat, 2019-05-29:
I don't understand all the feedback on wifi connect and/or disconnect, and that this would trigger the alarm? In the situation I described the RPI stays connected to wifi all the time and the alarm is only triggered by disconnecting the cam/music USB (manually or via TeslaUSB process). The RPI stays powered on all the time as well because I'm using both the front USB port and the power usb port (via cigarette lighter power) and the uptime (via RasPi Check) shows 22 hours or more. Depending on a reboot I gave manually.

Like **gummby8** reported earlier, it looks like the tesla thinks the USB disk is being stolen and then triggers the alarm. Indeed the solution would be to disable Sentry via an API before the RPI starts to sync cam footage and by doing that disconnects the cam/music USB from the car.

marcone, 2019-05-29:
@Vipercat: teslausb **only** syncs when it (1) starts up or (2) regains wifi connection after having lost it. If it's continuously connected to wifi, it will **not** sync.

If your rpi has an uptime of 22 hours, and the alarm went off during that time, check archiveloop.log to see if it tried to archive anything during that time because it thought it lost/gained wifi connectivity.

Vipercat, 2019-05-29:
@marcone, thanks for the info. Regarding the 22 hours uptime; this was just a response to your earlier remark where you said that '....the car could cause the USB port to become unpowered briefly, which would cause a USB-powered rpi to reboot...." and clarify that in my situation the RPI keeps powered on all the time due to the fact that I use both USB ports. Again, the only (reproduced) trigger in my situation for an alarm is the manual disconnect of the cam/music USB disk. If the TeslaUSB only disconnects due to syncing when it 1) it starts up or 2) it regain wifi connection then I understand could be triggered by those events when Sentry is active.

Based on this info I did some additional testing and discovered that a reboot of the RPI, while Sentry is active, does indeed trigger the alarm. A wifi disconnect and then reconnect did not trigger the alarm. So I wonder why some of us experience an alarm trigger when the wifi disconnect and reconnect happens and not me?

marcone, 2019-05-29:
How did you disconnect/reconnect wifi? What matters isn't so much the wifi connection itself, but whether archiveloop notices that it lost and then gained connectivity. If it goes down only very briefly, then archiveloop might not detect it.

Vipercat, 2019-05-29:
I forced the disconnect/reconnect by disabling my home WiFi for a minute or so. Maybe not long enough. Will test and reproduce tomorrow because I don't want to disturb the neighbor's (11:30pm right now)

acraigl, 2019-05-30:
> I'm planning on disabling Sentry while at home, and using Tasker to enable it for 20 minutes, then disable it again. Likely at 10pm or something so that it doesn't wake me up in the middle of the night. That way it can sync without additional power and won't go off accidentally (hopefully).
Worked a treat last night. I'll keep folks posted if this works consistently. If anyone (has Android) wants the Tasker script let me know (not that it's rocket science).

acraigl, 2019-06-03:
My tasker script worked perfectly for 4 days, and last night it set the alarm off again. Presumably when it turned on (with Sentry enabled) and then began to archive.  I wonder what could have been different last night? Other than that, the device/script is working well.

gummby8, 2019-06-03:
> My tasker script worked perfectly for 4 days, and last night it set the alarm off again. Presumably when it turned on (with Sentry enabled) and then began to archive. I wonder what could have been different last night? Other than that, the device/script is working well.

Do you need sentry mode on at home? Does enabling the AC power the USB ports as well. Perhaps that would be a better option. Would be a bit more of a power draw, but you wouldn't risk enabling a sound grenade in your garage.

acraigl, 2019-06-03:
> Do you need sentry mode on at home? Does enabling the AC power the USB ports as well. Perhaps that would be a better option. Would be a bit more of a power draw, but you wouldn't risk enabling a [...]

Definitely don't need sentry but I do need a way to keep the ports live. Outside of starting the car (which I could do) I was under the impression that the other things (AC, unlock, etc) would timeout on their own. I'll play with a few other options this evening. Good suggestion!

acraigl, 2019-06-04:
@gummby8 Good call. Enabling the AC seemed to work, at least based on a sample of one night. Will keep watching... thanks again for the suggestion.

shackrock, 2019-06-11:
Is there a planned fix to the code base for this one, long term?

marcone, 2019-06-11:
Depends on what you mean by "this one". There are two different things mentioned in this issue: the pi turning off because the USB ports turn off, and the recording drive being disconnected because the pi gets disconnected and reconnected to wifi.
For the first issue, there is support now for using the Tesla API to keep the car awake during archiving. Alternatively, you can leave climate control on, or use a battery.
For the second issue, I'm working on making it so that the recording drive does not need to be disconnected from the car while archiving.

schmug, 2019-06-12:
I've done some testing and so far disabling automatic sentry mode and only using the app/api/icon on screen does not seem to break autopilot or trigger the alarm

ScottESanDiego, 2019-06-12:
I haven't yet put my TeslaUSB together, but I use WebCore to enable/disable sentry w/SmartThings, and it seems like we could change the "keep car awake" Tesla API script to also call .../command/set_sentry_mode to "off" - do the sync - then "on", so that the rPI is doing the sentry disable/enable dance when it needs to, right?

marcone, 2019-06-12:
That seems like it should work, but what I want to do instead is to not disconnect the drive from the car while archiving (since that can take quite a bit of time). I believe this will work, because this is already how CIFS access works (it takes a snapshot of the disk image, then mounts that snapshot and serves it up to the client). Additionally, it appears that disconnecting the drive from the car only briefly doesn't trigger the alarm, so it's still possible to free up space on the drive this way by swapping in an empty disk image at some point.

phtp, 2019-06-12:
> That seems like it should work, but what I want to do instead is to not disconnect the drive from the car while archiving (since that can take quite a bit of time). I believe this will work, because [...]

Keep up the good work, that sounds like a good solution. I hope you'll be able to accomplish this.

marcone, 2020-04-18:
It appears that current Tesla software no longer sounds the alarm when teslausb makes the drive unavailable to the car while Sentry mode is active, so I'm going to close this issue.
(I still plan on making it so the drive can stay available to the car while teslausb is archiving)

closed 2020-04-18 by marcone