marcone/teslausb #44: /etc/fstab not updated when changing sharename

cjwang18, 2019-05-26:
After changing my sharename in `/boot/teslausb_setup_variables.conf` and re-running `/etc/rc.local`, I noticed that it was failing to mount the archive. Thanks to #42 and #27, I checked `/etc/fstab` and noticed that it was still pointing to the old sharename I had specified. I reran the whole setup process but it still didn't change. For now I've manually updated `/etc/fstab`. Ideally this file should get updated/recreated when the setup process happens again but I haven't delved into the details yet. Hope this helps for anyone who's running into the same issue.

marcone, 2019-05-27:
Fixes are welcome, but it's not a goal or priority for me to make it so setup can be rerun again to change aspects of the configuration. While there is some support for running setup again in case something went wrong the first time, I generally consider setup to be a one time thing. If you need to change something, it's safest to start over from scratch.

closed 2019-05-27 by marcone

phillq, 2020-11-05:
Thanks @cjwang18, this solved it for me!