marcone/teslausb #50: Setup does not complete [Grabbing main setup file]

mcowger, 2019-05-28:
Good morning - 

I've found a couple closed issues with this problem, but the ones I saw all 'magically' fixed themselves.   

In short, after burning the image to my  SD, I boot the Pi0W and wait.  it does some activity for about 4 minutes, and then stops.  I'm able to SSH to the Pi (which tells me the network parameters are fine).

On the Pi, I see in `/boot` the `WIFI_ENABLED` and `TESLAUSB_SETUP_STARTED` files, but it never finishes setup.   

The setup log never progresses past this stage:

```
Mon  6 May 01:08:52 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:08:52 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Mon  6 May 01:08:52 BST 2019 : Rebooting...
Mon  6 May 01:09:18 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:09:18 BST 2019 : Grabbing main setup file.
```

Thoughts?

marcone, 2019-05-28:
Were you actually using wifi to ssh in? Note that during early setup there's also a network interface on the usb port, so if you were ssh'ing in from the computer that the pi was attached to, you may have been using ethernet over usb.

mcowger, 2019-05-28:
I was using wifi to SSH as I powered the Pi from a wall plug with no connection to my laptop.

smcasagrande, 2019-05-28:
I ran into this same thing.  I was able to continue by ssh into the pi, then running /etc/rc.local manually.  There were a couple of places where I needed to say "yes" to a prompt for creating filesystems, which I suspect was the reason it was hanging.  Perhaps a holdover of a previous installation?

mcowger, 2019-05-28:
I'll give it a shot.   

There isn't a previous install.  Brand new microSD and pi0w

mcowger, 2019-05-28:
Indeed you were correct...it was seeing something old:

```
/dev/mmcblk0p3 contains a ext4 file system labelled 'backingfiles'
	last mounted on /backingfiles on Tue May 28 01:17:09 2019
Proceed anyway? (y,N) y
Discarding device blocks: done
Creating filesystem with 3418368 4k blocks and 855120 inodes
Filesystem UUID: 5560da63-be07-4738-b469-1884d388a0f4
Superblock backups stored on blocks:
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208
```
that I had to answer yes to.

Seems like theres a need to force this @marcone

marcone, 2019-05-28:
Odd. The script that creates the partition checks whether the partition already exists. The only way to get into the situation described above is if the previous run was interrupted after the backingfiles partition was created but before the mutable partition.

mcowger, 2019-05-28:
That did happen in my case.    Seems like a first run of the script (eg not having written a success marker) can pretty safely wipe out anything that's there.

marcone, 2019-05-28:
Yes, the script that creates the partitions could be more robust. Fixes welcome.

marcone, 2019-06-03:
This should work better now

closed 2019-06-03 by marcone