marcone/teslausb #53: grep: /root/bin/setup-teslausb: No such file or directory (STOP: failed to update setup-teslausb)

CmSCMU, 2019-05-29:
I'm attempting a manual install with GDrive and Pushover. I keep running into the same error on the last step when I try to run the TeslaUSB setup. Specifically, I get:

grep: /root/bin/setup-teslausb: No such file or directory
setup-teslausb is outdated, attempting update
STOP: failed to update setup-teslausb

I have looked over the closed and open issues and don't see anyone who is having a similar issue. I was wondering if you could tell me where I'm going wrong. Here is the log from my setup:

PS C:\Users\chads> ssh pi@raspberrypi
The authenticity of host 'raspberrypi (fe80::57c4:d551:6557:a7a8%20)' can't be established.
ECDSA key fingerprint is SHA256:S5vf5pIwnKt+NY5HxbNRBDkNcM8xCTNNdIB54meku+4.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'raspberrypi,fe80::57c4:d551:6557:a7a8%20' (ECDSA) to the list of known hosts.
pi@raspberrypi's password:
Linux raspberrypi 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ passwd
Changing password for pi.
(current) UNIX password:
Enter new UNIX password:
Retype new UNIX password:
passwd: password updated successfully
pi@raspberrypi:~ $ sudo -i

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

root@raspberrypi:~# curl -L https://raw.github.com/pageauc/rclone4pi/master/rclone-install.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  2508  100  2508    0     0   3451      0 --:--:-- --:--:-- --:--:--  3451
bash ver 1.6 written by Claude Pageau
-------------------------------------------------------------------------------
--2019-05-29 05:01:27--  https://downloads.rclone.org/rclone-current-linux-arm.zip
Resolving downloads.rclone.org (downloads.rclone.org)... 2a02:24e0:8:61f9::1, 5.153.250.7
Connecting to downloads.rclone.org (downloads.rclone.org)|2a02:24e0:8:61f9::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9447065 (9.0M) [application/zip]
Saving to: ‘rclone.zip’

rclone.zip                                          100%[==================================================================================================================>]   9.01M  2.00MB/s    in 5.2s

2019-05-29 05:01:33 (1.74 MB/s) - ‘rclone.zip’ saved [9447065/9447065]

unzip rclone.zip to folder rclone-tmp
Archive:  rclone.zip
  inflating: rclone-tmp/rclone.1
  inflating: rclone-tmp/README.txt
 extracting: rclone-tmp/git-log.txt
  inflating: rclone-tmp/README.html
  inflating: rclone-tmp/rclone
Install files and man pages
Purging old database entries in /usr/share/man...
Processing manual pages under /usr/share/man...
Purging old database entries in /usr/share/man/fr.UTF-8...
Processing manual pages under /usr/share/man/fr.UTF-8...
Purging old database entries in /usr/share/man/sl...
Processing manual pages under /usr/share/man/sl...
Purging old database entries in /usr/share/man/sv...
[... 305 lines cut ...]
Saving to: ‘setup-teslausb’

setup-teslausb                                      100%[==================================================================================================================>]   8.00K  --.-KB/s    in 0.003s

2019-05-29 05:06:33 (2.39 MB/s) - ‘setup-teslausb’ saved [8192/8192]

root@raspberrypi:~# chmod +x setup-teslausb
root@raspberrypi:~# ./setup-teslausb
Updating package index files...
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [221 kB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [45.0 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 12.1 MB in 26s (464 kB/s)
Reading package lists... Done
Verifying that the requested configuration is valid...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1804  100  1804    0     0   4382      0 --:--:-- --:--:-- --:--:--  4400
Downloaded /tmp/verify-configuration.sh ...
grep: /root/bin/setup-teslausb: No such file or directory
setup-teslausb is outdated, attempting update
STOP: failed to update setup-teslausb

marcone, 2019-05-29:
I think you just need to download setup-teslausb to /root/bin instead of to /root

CmSCMU, 2019-05-29:
Crap. Thanks.

closed 2019-05-29 by CmSCMU