marcone/teslausb #54: Mounting /mnt/archive fails

gummby8, 2019-05-29:
from my archiveloop.log

> Wed 29 May 06:02:45 BST 2019: Mounting /mnt/archive...
mount: can't find /mnt/archive in /etc/fstab
Wed 29 May 06:02:45 BST 2019: Failed to mount /mnt/archive.
Wed 29 May 06:02:45 BST 2019: Sleeping before retry...
Wed 29 May 06:02:46 BST 2019: Retrying...

thoughts?

gummby8, 2019-05-29:
I may have fixed this. I think it was having issues mounting the shared drive from my PC. I set the permission son the share for "EVERYONE" and figured it would not need a username/password
After I created a new user and password and gave it permissions to the share. I flashed the SD card again and it appears to have worked. I guess specify in the setup instructions that you MUST give a valid username and password, so you must have a user with a password set on the windows machine. 

I did have to use /etc/rc.local as it seemed my previous partitions did not get wiped by windows

marcone, 2019-05-29:
I'm using a "guest mode" share myself, and simply have this in the config file:
```
export shareuser=
export sharepassword=
```
works for that. My share is on a linux machine though, not a Windows machine.

How did you flash the sd card? If done right, the flashing process should create only 2 partitions, and then when booted the setup scripts will create another 2 partitions.

closed 2019-05-31 by marcone