marcone/teslausb #55: MicroSD Card is too small (manual install)

CmSCMU, 2019-05-29:
Marcone,

I have been looking through the closed issue threads and saw that you put some code into the one-step install to check for previous partitions. I am trying to install manually (trying to use gdrive), but am installing for the second time. Therefore when I try to run: 

wget https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb
chmod +x setup-teslausb
./setup-teslausb

I get:

 Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small: 524288 bytes available.
BYT; /dev/mmcblk0:31.9GB:sd/mmc:512:512:msdos:SD SU32G:; 1:4194kB:49.2MB:45.0MB:fat32::lba; 2:50.3MB:1804MB:1753MB:ext4::; 3:1804MB:31.8GB:30.0GB:ext4::; 4:31.8GB:31.9GB:105MB:ext4::;

Is there anyway to fix this for the manual folks like you did for the one-step installs? Is there any workaround for this?

Finally, can I throw you a donation for answering my questions the last two days?

marcone, 2019-05-29:
Can you run `fdisk -l /dev/mmcblk0` and `blkid` ?
It looks like your sd card was partioned before, but the partitions weren't labeled, and so the setup script doesn't recognize them. Probably easiest to delete the 3rd and 4th partition (the one that is ~30 GB and the one that is 105 MB) and then running setup again.

marcone, 2019-05-31:
I've reworked the free space check a bit.

closed 2019-05-31 by marcone

lowfuel, 2019-06-22:
I'm getting the exact same error. No idea what I'm doing wrong. Following the instructions carefully. 
```
root@raspberrypi:~/bin# fdisk --l /dev/mmcblk0
Disk /dev/mmcblk0: 59.5 GiB, 63864569856 bytes, 124735488 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5ba58c2a

Device         Boot Start       End   Sectors  Size Id Type
/dev/mmcblk0p1       8192     96042     87851 42.9M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 124735487 124637184 59.4G 83 Linux
root@raspberrypi:~/bin# blkid
/dev/mmcblk0p1: LABEL="boot" UUID="16D2-035F" TYPE="vfat" PARTUUID="5ba58c2a-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="d065e631-6b9d-48c0-a8fe-e663b42828e0" TYPE="ext4" PARTUUID="5ba58c2a-02"
/dev/mmcblk0: PTUUID="5ba58c2a" PTTYPE="dos"

```
My SD card is initially unformatted/partitioned, using Etcher to write the zip file to it. Not sure if it is the issue, but the "boot" partition I get after booting from the card is 42MB.

marcone, 2019-06-22:
@lowfuel which zipfile did you flash? The "card too small" message generally happens when there is not enough free space after the 2nd partition, and in your case the 2nd partition is 59.4 GB, instead of the usual 1.7 GB, leaving no free space at all.
With the current image, the 1st partition should be 44.3 MB, and the 2nd partition 1.7 GB.

lowfuel, 2019-06-22:
@marcone I was following the instructions on the main page, using the Raspbian Stretch Lite zipfile. Not sure, but I suspect something might be wrong with those instructions, because I never get the 2nd partition you mentioned.

However, after posting the above message, my morning coffee finally kicked in and I spotted the link for the "One step setup instructions", and followed that using the teslausb zip file. Worked flawlessly.

marcone, 2019-06-22:
You did have a 2nd partition in your fdisk and blkid output.
I added a note to the instruction emphasizing that the prebuilt image is preferred.