marcone/teslausb #57: failed to mount archive / no apparent entries for stab and credentials

vaeriecom, 2019-05-31:
Hi, I've been so freaking excited about this and it's absolutely amazing how easy this has been so far. I haven't really touched code in this way since middle school, but here I am relearning bash. I hope you can help me muddle through this, I'm obviously not the best with this stuff!

I've been using the latest prebuilt image 20190505 and the one step config file.
It works fine as far as the Tesla is concerned, and the Cam partition mounts (there is always a dirty eject error on my Mac when I plug it in there, but then it remounts fine)

It connects to wifi just fine as well, but when I check the archive loop, it always fails. 
Checking fstab, there was no apparent entry for my server. This is what was in my one step config. I just hacked together the last line based on reading previous issues here. I read in a previous issue that removing the single quotes worked for someone, and I've tried it both ways, etching a new prebuilt image and using a new .conf.

One thing I've noticed is that in my case, the script always seems to hang at fetching main config file in archive loop, and I have to run rc scripts again to get it to finish setting up.

Back to fstab - I also have no .teslaCamArchiveCredentials either so I'm playing a guessing game as to how that should be formatted.

**fstab, last line is hacked in - no idea how it should be**
```
proc            /proc           proc    defaults          0       0
PARTUUID=b4293f65-01  /boot           vfat    defaults          0       2
PARTUUID=b4293f65-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=backingfiles /backingfiles ext4 auto,rw,noatime 0 2
LABEL=mutable /mutable ext4 auto,rw 0 2
/backingfiles/cam_disk.bin /mnt/cam vfat noauto,users,umask=000,offset=1048576 0 0
/homeserver.local/Storage /mnt/archive cifs vers=1.0,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0
```

**blank .teslaCamArchiveCredentials, I have no idea what should be in here**
```
shareuser=pi
sharepassword=raspberry
```

**contents of teslausb_setup_variables.conf**
```
#####################################################################
# SAMPLE CONFIGURATION FILE FOR TESLAUSB Pi Setup
#
# Example config file for teslausb_setup. Lines with "#" are comments/ignored.
# Remove the "#" before "export" to activate a line. Be sure to rename this file
# to "teslausb_setup_variables.conf" and place it in the "boot" folder of your
# SD card. 
#
######################################################################

# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export archiveserver=homeserver.local
export sharename=Storage
export shareuser=pi
export sharepassword=raspberry
export camsize=100%
export timezone=America/Chicago

# if you set camsize to less than 100%, but don't want music to use
# all the remaining space, you can specify how much space you want
# music to use by setting this variable
#export musicsize=4G

# If you want to automatically copy music from a CIFS share every time
# the Pi connects to wifi, set the following variable. The share is
# assumed to exist on the same server as the archive share. It can
# be the same share as the share used for backing up recordings, but
# the folder needs to be different.
# export musicsharename=your_music_share/music_folder

# Wifi setup information
export SSID=wifi
export WIFIPASS='password'

# If doing a headless (i.e. automatic) setup
export HEADLESS_SETUP=true

# Uncomment if setting up Pushover push notifications 
# export pushover_enabled=false
# export pushover_user_key=user_key
# export pushover_app_key=app_key

# Uncomment and change if you want setup scripts to be pulled
# from a different repo than github.com/marcone/teslausb
# export REPO=marcone

# Uncomment and change if you want a different branch than main-dev
# export BRANCH=main-dev

```

marcone, 2019-05-31:
You said that "the script always seems to hang at fetching main config file in archive loop". What do you mean by that? archiveloop does not fetch anything. The setup script does though, and from what you said it looks like setup didn't finish. Can you attach /boot/teslausb-headless-setup.log here?

As for the rest, the entry in fstab needs to have a double-/ at the beginning, i.e:
```
//homeserver.local/Storage /mnt/archive cifs vers=1.0,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0
```
Check that the cifs version is actually correct.

/root/.teslaCamArchiveCredentials should contain the username and password needed for accessing the share on your file server, using this format:
```
username=yourusername
password=yourpassword
```
If the share does not require authentication, then
```
username=
password=
```
will do.

vaeriecom, 2019-05-31:
Thank you, I'll try those fixes.

Every time I setup a newly etched prebuilt image with config file, the log always stops on fetching main config file. It will sit there for 3 hours if I don't run rc.local and then it continues or restarts and then finishes setup.

marcone, 2019-05-31:
Can you try this after a stuck/failed setup:
```
sudo -i
mkdir /tmp/root
mount --bind / /tmp/root
cat /tmp/root/var/log/boot.log
```
That should show the output of rc.local and the setup scripts during setup, and may reveal why it doesn't finish.

closed 2019-06-02 by marcone

vaeriecom, 2019-06-03:
I've implemented all the changes, and am now getting permission denied. I've checked the login info and it works when mounting the network volume on my laptop. (both the server and laptop are Macs)

```
proc            /proc           proc    defaults          0       0
PARTUUID=b4293f65-01  /boot           vfat    defaults          0       2
PARTUUID=b4293f65-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=backingfiles /backingfiles ext4 auto,rw,noatime 0 2
LABEL=mutable /mutable ext4 auto,rw 0 2
/backingfiles/cam_disk.bin /mnt/cam vfat noauto,users,umask=000,offset=1048576 0 0
//homeserver.local/Storage /mnt/archive cifs vers=3.0,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0
```

```
Mon  3 Jun 12:48:09 CDT 2019: Mounting /mnt/archive...
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Mon  3 Jun 12:48:09 CDT 2019: Failed to mount /mnt/archive.
Mon  3 Jun 12:48:09 CDT 2019: Sleeping before retry...
Mon  3 Jun 12:48:10 CDT 2019: Retrying...
Mon  3 Jun 12:48:10 CDT 2019: Mounting /mnt/archive...
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Mon  3 Jun 12:48:10 CDT 2019: Failed to mount /mnt/archive.
Mon  3 Jun 12:48:10 CDT 2019: Attempts exhausted.
Mon  3 Jun 12:48:10 CDT 2019: Failed to mount cam archive.
Mon  3 Jun 12:48:10 CDT 2019: Couldn't connect archive, skipping archive step
Mon  3 Jun 12:48:10 CDT 2019: Connecting usb to host...
Mon  3 Jun 12:48:11 CDT 2019: Connected usb to host.
Mon  3 Jun 12:48:11 CDT 2019: Waiting for archive to be unreachable...
```