marcone/teslausb #59: Moved Files Are Protected and Not in the Correct Folders

barjohn, 2019-06-01:
The application is moving files to my Shared drive.  The shared drive is named TeslaShare and it has a folder TeslaCam with two sub folders SavedClips and RecentClips.  The moved files appear to be moved in bathes that very in size and are themselves enclosed in a date folder.  They are moved to the root, not into TeslaCam nor into the appropriate subdirectory in TeslaCam.  Additionally, they are only openable but the username I gave for the shared drive.  This occurs even though in my sharing I have given admins and the car as a user read/write access (permissions) and the shared drive and all subfolders on the drive have that admins and auto as user with read/write permission.  (I suspect when it creates the date sub-folder it only gives itself permission).  Why it doesn't move files to the appropriate sub-folder isn't clear.  I could change the user to myself to resolve the issue of permissions but I'm not sure how to fix it creating date subfolders and not putting things in the correct subfolder.  The reason this is desirable is that the App called DashCam Viewer (Downloadable from the Apple App store) expects to find files in the proper sub directories in TeslaCam and it allows you to view 3 cameras simultaneously which is a nice feature.  Thanks for guidance on how to fix this.

barjohn, 2019-06-01:
I also found that the clips on the USB CAM drive are all 595 byte files in the RecentClips folder.  See 
[image]

marcone, 2019-06-01:
The location of the archived files is working as intended. If you want to have the files archived to the "TeslaCam" folder on your share, then you should specify "TeslaShare/TeslaCam" as the share name during setup. (or in your case it sounds like you want "TeslaShare/TeslaCam/SavedClips"
The 595 byte files is a problem with the car, perhaps rebooting the MCU will fix it.

closed 2019-06-01 by marcone

barjohn, 2019-06-01:
I understood that but you don't move the files, you move the files into date folders that are not on the original USB drive.  I would like recentClips to go to RecentClips on the share and SavedClips to go to SavedClips on the Share.  Is there a way to do this?  This cuts down on the time it takes to find a desired clip because the events that create them are different.

barjohn, 2019-06-01:
I guess another way to put it is I would like the files structure on the USB to be mimicked on the shared folder.  If that isn't possible, then can I modify the save so it just saves the file without grouping into date folders?

marcone, 2019-06-01:
The date folders *are* on the original drive. That's how the car saves them in the SavedClips folder.
The RecentClips folder does not have subfolders, but teslausb currently only backs up the SavedClips folder, not the RecentClips folder.