marcone/teslausb #61: Adding Music from CIFS Share stops when strange characters are in the filename

Vipercat, 2019-06-01:
I noticed that files with specific characters in it's name (e.g. "ë" in Tiësto or "é" in Céline) stops the  files from being copied from the share to the RPI. When copied via Windows and reviewed later on the RPI those characters are displayed as "?" So I expect that the 'cp' command is not able to copy files with these characters in it's name. 

Is there a way to overcome this limitation, for example by putting the files names between brackets while copying?

Thanks

kdvlr, 2019-06-02:
Interesting.. I was about to post this as well. 

Here's what I see in archiveloop log
```
couldn't make directory /mnt/music/Music/Beyoncé/I Am...Sasha Fierce
Sun  2 Jun 00:57:04 BST 2019: Couldn't move to /mnt/music/Music/Compilations/Billboard Top 100 of 2015/19 Where Are Ü Now_.mp3
```
When I had my own version of pi that I build. I mounted using 
```sudo mount /dev/loop0p2 /mnt/music -o shortname=mixed -o utf8```

and this solved the problem for me.

marcone, 2019-06-02:
Yes, I was just about to add the utf8 option

marcone, 2019-06-02:
This is fixed now. You can either add 'utf8' to the mount options in /etc/fstab, or update your install by running
```
sudo -i
rm /backingfiles/cam_disk.bin
/root/bin/setup-teslausb
```

closed 2019-06-02 by marcone

Vipercat, 2019-06-02:
Thanks, I've updated my install