marcone/teslausb #63: TeslaUSB having trouble disconnecting from network sometimes

hoopsta1423, 2019-06-04:
Seems like my teslaUSB sometimes has trouble if I leave while it is trying to archive files. My pi was archiving files this evening and I needed to run an errand (~8 PM
 EST). I was gone for ~30-45 mins and the drive never mounted in the car but connected back to my home network as soon as I got home and continued archiving. This has happened a few times now and I've just reset the pi to get it back but I'm not entirely sure where to start troubleshooting. Logs attached, hope they are helpful

Let me know
Thanks!

[archiveloop.log](https://github.com/marcone/teslausb/files/3250097/archiveloop.log)
[dmesg.txt](https://github.com/marcone/teslausb/files/3250105/dmesg.txt)
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/3250106/teslausb-headless-setup.log)
[diagnostics.txt](https://github.com/marcone/teslausb/files/3250107/diagnostics.txt)

hoopsta1423, 2019-06-04:
also, having issues running the selfupdate command from the 'read this first' post
[selfupdate.txt](https://github.com/marcone/teslausb/files/3250148/selfupdate.txt)

raushel, 2019-06-04:
Need to run the readwrite_fs script first to reload the file system to write. 

Assuming this could be added to the self update script?

marcone, 2019-06-04:
@hoopsta1423 I'm a little confused as to what you say happened. The archiveloop.log ends at 01:06 BST, which is 20:06 EST, right around the time you say you left home for a bit.
If that line is really the last in the log, then that would mean it just got stuck there and never recovered, however you said that when you came back, it connected back to your network and continued archiving.

Your dmesg does show cifs hanging, but since that's down in the kernel/filesystem itself, I don't know how to fix that. Maybe adding something like `echo_interval=5` to the mount options for /mnt/archive would help.

hoopsta1423, 2019-06-04:
Hmm ok, I'm going to try recreating the issue tonight. Basically it seems
like the pi gets hung up if I leave while it's moving files. Ill have to
unplug it to have it remount the drives and begin recording again.

I'll post logs if it does this again tonight

hoopsta1423, 2019-06-04:
Yup, I left the house at 7:37 and just pulled into the grocery store at
7:51. The drive never remounted and still blinking fast like it's trying to
continue copying files. Will pull logs when I'm back home

marcone, 2019-06-06:
Did you try adding the "echo_interval=5" mount option in /etc/fstab ? (you'll also need to reboot after)

hoopsta1423, 2019-06-06:
Sorry, must've missed that suggestion earlier. I'll add that today and see
what happens

hoopsta1423, 2019-06-07:
> Did you try adding the "echo_interval=5" mount option in /etc/fstab ? (you'll also need to reboot after)

No luck with this. I added that in /etc/fstab and rebooted last night. Just left the house while it was archiving (~3:50 EST) and the drive never successfully disconnected/remounted the cam/music drives (~30mins of drive time)

Have you never had this issue? Should I just start from scratch and see what happens? It's not usable in its current form as it would require my wife to reboot the drive everytime she went out

[diagnostics.txt](https://github.com/marcone/teslausb/files/3267510/diagnostics.txt)

[archiveloop.log](https://github.com/marcone/teslausb/files/3267518/archiveloop.log)

smcasagrande, 2019-06-07:
hoopsta1423, you're not alone.  I'm seeing the same thing on occasion, but haven't spent too much time to diagnose, just reboot the pi after a 5-10 minutes of driving.

hoopsta1423, 2019-06-08:
> hoopsta1423, you're not alone. I'm seeing the same thing on occasion, but haven't spent too much time to diagnose, just reboot the pi after a 5-10 minutes of driving.

gah, was hoping it was just me. Rebooting the unfortunately isn't an option in my case. I love my wife, but I can't count on her noticing it's not recording and then trying to unplug/replug the pi

This script is almost perfect, but I'll have to look at other options if I can't figure this issue out

xnaas, 2019-06-08:
I happen to notice with no discernible pattern that the Pi doesn’t mount some mornings when leaving for work. Unfortunately, I don’t notice until I go to hit record... My backups always complete within minutes of being home, so I know that’s not the issue in my case. 

I did just run a selfupdate a few minutes ago, I guess we’ll see if anything changes.

marcone, 2019-06-08:
@xnaas Note that "setup-teslausb selfupdate" only updates the setup-teslausb script itself. To update the install, you then need to run that updated setup-teslausb.

kdvlr, 2019-06-09:
I have been having the same issues and I submitted a PR to fix that (which I then closed due to further research needed). 
Occasionally, the drives don't get mounted. A modprobe -r, fsck and modprobe usually fixes that. 
Some other times, the tesla itself doesn't recognize the drive till the tesla is restarted. I have seen this happen when the pi is attached to a power bank and it keeps disconnecting and reconnecting. 
I haven't been able to resolve the second one. I'll go ahead and resubmit my previous fix in the hope it addresses this.

hoopsta1423, 2019-06-09:
> I have been having the same issues and I submitted a PR to fix that (which I then closed due to further research needed).
> Occasionally, the drives don't get mounted. A modprobe -r, fsck and modprobe [...]

Interesting, I do have mine plugged into a small power bank as well which may be part of the issue

smcasagrande, 2019-06-09:
Not me,the PI is just plugged in to the USB port on the Tesla.

xnaas, 2019-06-10:
> Note that "setup-teslausb selfupdate" only updates the setup-teslausb script itself. To update the install, you then need to run that updated setup-teslausb.

Really? Damn. Because that thing ran for a good 45 minutes and seemed to do *a lot*.

marcone, 2019-06-10:
If this is the first time you ran selfupdate since I added that functionality, then it will take longer.
Still shouldn't take 45 minutes though, unless the pi's internet connection is really slow.

kdvlr, 2019-06-10:
Submitted PR: https://github.com/marcone/teslausb/pull/74 in case the gadgets is not mounted directly.

xnaas, 2019-06-10:
It took that long because it definitely ran the full upgrade after the first step. I can tell because it asked me that "(yes/no/cancel)" question.

hoopsta1423, 2019-06-13:
I broke down and started from scratch, but it doesn't appear to have fixed the issue for me. I'm not really sure where to go from here

kdvlr, 2019-06-18:
I think  I  got  something like this today. Not sure if it's the same  issue.  I  came home and left 15 mins later and  my driver never mounted again. I have attached the log  files here, but haven't looked too much  into it yet. 
[archiveloop.log](https://github.com/marcone/teslausb/files/3299850/archiveloop.log)
[dmesg.txt](https://github.com/marcone/teslausb/files/3299851/dmesg.txt)

kdvlr, 2019-06-21:
OK. This seems to be reproducible on my end. 
1. Have some files in SavedClips. 
2. Let teslausb connect to archive and start moving files.
3. Turn off the wifi or drive the car away so that it loses wifi connection. 

The archiveloop pauses at the last file being moved and does not move beyond that. Also, it doesn't error out and give control back to archiveloop, so it can mount the USB gadget again. 

If anyone else can confirm it is not just me, that would be much appreciated.

hoopsta1423, 2019-06-21:
> OK. This seems to be reproducible on my end.
> 
> 1. Have some files in SavedClips.
> 2. Let teslausb connect to archive and start moving files. Don't let it complete moving the files
> 3. Turn off the wifi or drive the car away so that it loses wifi connection.
> 
> The archiveloop pauses at the last file being moved and does not move beyond that. Also, it doesn't error out and give control back to archiveloop, so it can mount the USB gadget again.
> 
> If anyone else can confirm it is not just me, that would be much appreciated.

Yup, this is exactly how it acts on my end as well

kdvlr, 2019-06-22:
I am about to submit a PR for a timeout on rsync which is what I am using. I don't know if it will be as effective for the other methods. I am just wondering what the correct timeout period should be. I think I should lean towards making the drive available again as soon as possible. I think 120s is reasonable.

hoopsta1423, 2019-06-22:
> I am about to submit a PR for a timeout on rsync which is what I am using. I don't know if it will be as effective for the other methods. I am just wondering what the correct timeout period should [...]

Not sure if this is the right place to ask, but is it possible to use rsync if I'm trying to back up to a drive on my win10 server?

kdvlr, 2019-06-22:
Actually I have a fix for both rsync and cifs mount method. I am going to test it today and have a PR submitted.

kdvlr, 2019-06-24:
Submitted PR https://github.com/marcone/teslausb/pull/94 for this fix. I have been trying this for two days and it seems to have fixed the problem. One caveat being that file transfer speeds seem to be quite affected by distance. I have set a timeout of 90s as each file transfer seems to complete within 60 seconds and gives about 30s of buffer. 

If more than 3 transfers fail consecutively, I am assuming that wifi is broken and it exits the loop to try again later. 

It takes about 4 minutes after wifi disconnection for the drive to be available again.

hoopsta1423, 2019-06-26:
> Submitted PR #94 for this fix. I have been trying this for two days and it seems to have fixed the problem. One caveat being that file transfer speeds seem to be quite affected by distance. I have [...]

This is great, I actually edited it a bit just because my wifi signal is weak in the driveway. I upped the transfer time to 120s and changed the number of failures to >2. Ends up being the same length of time, but allows for my slower transfers

Thanks for adding this!

marcone, 2019-06-27:
I've been trying to reproduce this problem, but when I go out of wifi range the mv just fails with: "mv: error writing '/mnt/archive/...': Host is down"

kdvlr, 2019-06-28:
@marcone - So you don't have the same problem that we have? 
If the car loses wi-fi while the files are being transferred, it fails gracefully and connects back to the car ?

marcone, 2019-06-29:
I was going to reply "yes that's how it works for me", but then I gave it another try and of course then I had the same/similar problem.
What happened was that the in-progress transfer failed after some time, then it tried the next file, which failed after 20 seconds, then the next file, which failed after 10 seconds, and then every file after that failed after 10 seconds, until it had tried and failed to transfer them all. After that it connected the drive back to the car.
Next time it connected back to wifi, it started transferring the files that previously failed.
The 10-second failure time is probably because I had added `echo_interval=10` to the /etc/fstab entries for the CIFS server (the default value is 60 seconds, and it looks like it may take up to twice the interval to fail (which would explain the initial 20-second failure).

So assuming the transfer doesn't outright hang indefinitely, what might work is:
1. set the echo_interval to a relative short interval, like 10 seconds
1. after every failed transfer, check whether it's still connected to wifi and abort the transfer loop if not

kdvlr, 2019-06-29:
I have echo_interval set to 5 and it still doesn't recover. @hoopsta1423 also mentioned above that he tried to set echo_interval to 5 and it did not work. 
I do agree checking wifi is connected or not is a better approach than an arbitrary no. of continuous failures.

hoopsta1423, 2019-06-29:
> I have echo_interval set to 5 and it still doesn't recover. @hoopsta1423 also mentioned above that he tried to set echo_interval to 5 and it did not work.
> I do agree checking wifi is connected or not is a better approach than an arbitrary no. of continuous failures.

Yup, I tried adding that echo_interval without any luck. This new PR has been working for me so far so thank you @kdvlr

kdvlr, 2019-06-29:
There seems to different behavior based on how you test the scenario. 

If I test loss of connection by stopping the CIFS share, then I see the exact same behavior as mentioned by @marcone . 

If I test loss of connection by actually moving the car away from the wifi or by turning off the WiFi router, then it just waits and does not recover. My original push using the ```timeout``` utility works about 80% of the time to solve this problem. 

However, if the archiving is going on for more than 10 minutes or so and then I move the car away from WiFi, then the timeout doesn't work and the mv never recovers and just stays there. 

I have a fix that solves the problem all the time for me. 
1. When CIFS archive-clips is running, add a timeout of 120s for each file move to complete. 
2. If a file move fails, check wifi connection and exit if the NAS server does not respond to a ping. 
3. If the mv doesn't timeout, have another process that kicks off from archive-clips that monitors if archiveloop.log hasn't been updated for 5 mins. 

1 and 2 alone works more than 80% of the time for me. In this case, when there is no wifi, the USB is attached back to the car in around 2 minutes. 
3 is for the edge case. In this case, if archive-clips is running and nothing has been written to the archiveloop for more than 5 minutes, then it forcefully kills the archive-clips process which will then allow it to mount itself back to the car. 

Overall, the USB is always re-connected to the car in around 5 minutes or less.

marcone, 2019-06-30:
When I reproduced the problem yesterday I did move the car away from wifi, though I'm currently testing with a 2nd wifi access point that I can just turn off, since that's a lot easier :)
With that, I'm still seeing the same(*) 10-second-to-fail behavior on a Pi that does not have `echo_interval` set (so is using the default of 60 seconds). Perhaps it has something to do with the server, or rather whatever options the server and client negotiate? These are the settings shown by the "mount" command for the active mount on the Pi:
```
//fileserver/Video/Tesla on /mnt/archive type cifs
   (rw,relatime,vers=3.0,sec=none,cache=strict,uid=0,noforceuid,
   gid=0,noforcegid,addr=192.168.88.248,file_mode=0777,
   dir_mode=0777,soft,nounix,serverino,mapposix,
   rsize=1048576,wsize=1048576,echo_interval=60,actimeo=1)
``` 
(*) well, almost the same: the file that is in progress when I turn the wifi off takes 2 minutes to fail, the next one after that fails in 20 seconds, then after that it's 10 seconds for each file.

kdvlr, 2019-06-30:
I don't see any major difference apart from the fact you have sec=none and I have a user I use (removed it from below)

```
//192.168.100.154/TeslaCam on /mnt/archive type cifs (rw,relatime,vers=3.0,cache=strict,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.100.154,file_mode=0777,
dir_mode=0777,soft,nounix,serverino,mapposix,rsize=1048576,wsize=1048576,echo_interval=5,actimeo=1)
```

I wish rsync timeout worked between local mounts because that's a graceful way of handling timeouts. Without that we are left with inelegant options for CIFS.

marcone, 2019-06-30:
I even tried writing a simply copy program (to replace the mv in archive-clips.sh) that kills itself if it hasn't transferred any data for the last N seconds (source: https://gist.github.com/marcone/73b0a4413a23535effc2adc4c23d1ae5) but even that blocks for 2 minutes the first time before it'll return control, after which it actually works and returns after the specified timeout.

marcone, 2019-06-30:
I think I have a [working solution](https://github.com/marcone/teslausb/commit/e2036ac663e9601208fa4c6c929fae8d29473946) for this now. You can try this in the "quickdisconnect" branch (set BRANCH to "quickdisconnect" in your /root/teslausb_setup_variables.conf, then run /root/bin/setup-teslausb).
Basically, it starts a background task in archive-clips that continuously monitors the connection by pinging the archive server, and when that fails, it kills archive-clips. With that I'm seeing it reconnect the drives to the car within 20 seconds of losing the connection:
```
Thu  3 Nov 10:17:22 PDT 2016: Moving clips to archive...
Thu  3 Nov 10:17:22 PDT 2016: Creating output directory '/mnt/archive/'
Thu  3 Nov 10:17:23 PDT 2016: Creating output directory '/mnt/archive/2019-06-28_16-43-24'
Thu  3 Nov 10:17:23 PDT 2016: Moving '2019-06-28_16-43-24/2019-06-28_16-41-44-right_repeater.mp4'
Sun 30 Jun 08:33:30 PDT 2019: Moved '2019-06-28_16-43-24/2019-06-28_16-41-44-right_repeater.mp4'
Sun 30 Jun 08:33:30 PDT 2019: Moving '2019-06-28_16-43-24/2019-06-28_16-37-41-left_repeater.mp4'
Sun 30 Jun 08:33:39 PDT 2019: Moved '2019-06-28_16-43-24/2019-06-28_16-37-41-left_repeater.mp4'
Sun 30 Jun 08:33:39 PDT 2019: Moving '2019-06-28_16-43-24/2019-06-28_16-33-38-front.mp4'
Sun 30 Jun 08:33:47 PDT 2019: Moved '2019-06-28_16-43-24/2019-06-28_16-33-38-front.mp4'
Sun 30 Jun 08:33:47 PDT 2019: Moving '2019-06-28_16-43-24/2019-06-28_16-40-43-front.mp4'
Sun 30 Jun 08:33:56 PDT 2019: Moved '2019-06-28_16-43-24/2019-06-28_16-40-43-front.mp4'
Sun 30 Jun 08:33:56 PDT 2019: Moving '2019-06-28_16-43-24/2019-06-28_16-35-39-right_repeater.mp4'
Sun 30 Jun 08:34:04 PDT 2019: Moved '2019-06-28_16-43-24/2019-06-28_16-35-39-right_repeater.mp4'
Sun 30 Jun 08:34:04 PDT 2019: Moving '2019-06-28_16-43-24/2019-06-28_16-42-45-left_repeater.mp4'
Sun 30 Jun 08:34:16 PDT 2019: connection dead, killing archive-clips
Sun 30 Jun 08:34:16 PDT 2019: Moved 0 file(s), failed to copy 0, deleted 0.
Sun 30 Jun 08:34:16 PDT 2019: Unmounting /mnt/cam...
Sun 30 Jun 08:34:16 PDT 2019: Finished moving clips to archive.
umount: /mnt/cam: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
Sun 30 Jun 08:34:16 PDT 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Sun 30 Jun 08:34:16 PDT 2019: lazily unmounted /mnt/cam
Sun 30 Jun 08:34:16 PDT 2019: Finished archiving.
Sun 30 Jun 08:34:21 PDT 2019: Music archive not configured or not reachable
Sun 30 Jun 08:34:22 PDT 2019: Connecting usb to host...
```
I could set longer timeouts, but I don't like the idea of leaving home and not having the dashcam active  for several miles, so I set fairly aggressive timeouts.

smcasagrande, 2019-07-01:
Thanks marcone, I've installed the quickdisconnect branch, and will let you know if I see anything go wrong.

However, in the past couple of weeks I've improved my wifi coverage in my garage (new hotspot).  This has vastly reduced the transfer time for the files, and therefore seems to have closed the door on my problem in the first place.

Vipercat, 2019-07-01:
@marcone, is there a way to manually start the archiving of clips? It happens sometimes that I get home and the RPI connects to my WiFi, but the footage from SavedClips is not copied to my NAS. Not sure why? All shares to my NAS are available and mounted (checked via ssh). in 99 of the 100 times everything works like a charm, even also the NGINX webserver setup that I got from another branch with the save to Gdrive. The only issue on occasion is the one I just mentioned.

Thanks,
Vipercat

hoopsta1423, 2019-07-01:
> I think I have a [working solution](https://github.com/marcone/teslausb/commit/e2036ac663e9601208fa4c6c929fae8d29473946) for this now. You can try this in the "quickdisconnect" branch (set BRANCH to [...]

So this shouldn't kill the archive process in cases where we just have a weak wifi connection and transfers take longer than usual, correct?

marcone, 2019-07-01:
Right, it should only kill the archive when the background watchdog task is unable to contact the archive server.

marcone, 2019-07-01:
@Vipercat Ssh into the Pi and then either reboot, or `touch /tmp/archive_is_unreachable` to  simulate a wifi disconnect/reconnect.

Vipercat, 2019-07-01:
@Marcone, reboot doesn't work so I'll try the touch /tmp/archive_is_unreachable method. Thanks!

marcone, 2019-07-01:
Have you looked at archiveloop.log to see why it's not copying anything?

Vipercat, 2019-07-01:
@Marcone, I did and couldn't see anything out of the ordinary. Maybe you can:

fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 187 files, 159473/1295076 clusters
Thu  3 Nov 18:17:35 CET 2016: Finished fsck on /mnt/cam.
Thu  3 Nov 18:17:35 CET 2016: Moving clips to archive...
Thu  3 Nov 18:17:35 CET 2016: Creating output directory ''
Thu  3 Nov 18:17:35 CET 2016: Creating output directory ''
Thu  3 Nov 18:17:35 CET 2016: Moved 0 file(s), failed to copy 0, deleted 0.
Thu  3 Nov 18:17:35 CET 2016: Finished moving clips to archive.
Thu  3 Nov 18:17:35 CET 2016: Unmounting /mnt/cam...
Thu  3 Nov 18:17:35 CET 2016: Unmounted /mnt/cam.
Thu  3 Nov 18:17:35 CET 2016: Finished archiving.
Thu  3 Nov 18:17:35 CET 2016: Copying music...
Thu  3 Nov 18:17:35 CET 2016: Starting music sync...
Thu  3 Nov 18:17:35 CET 2016: Ensuring music backing file is mounted...
Thu  3 Nov 18:17:35 CET 2016: Mounting /mnt/music...
Thu  3 Nov 18:17:35 CET 2016: Mounted /mnt/music.
Thu  3 Nov 18:17:36 CET 2016: Ensured music drive is mounted.
Thu  3 Nov 18:17:36 CET 2016: Running fsck on /mnt/music...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 925 files, 1723739/1725419 clusters
Wed 26 Jun 23:13:04 CEST 2019: Finished fsck on /mnt/music.
Wed 26 Jun 23:13:05 CEST 2019: Copying music from  archive...
                                                                                                                                                                                                                                                                                                                   Thu  3 Nov 18:17:20 CET 2016: Starting archiveloop...
Thu  3 Nov 18:17:21 CET 2016: Ensuring music backing file is mounted...
Thu  3 Nov 18:17:22 CET 2016: Mounting /mnt/music...
Thu  3 Nov 18:17:22 CET 2016: Mounted /mnt/music.
Thu  3 Nov 18:17:23 CET 2016: Ensured music drive is mounted.
Thu  3 Nov 18:17:23 CET 2016: Running fsck on /mnt/music...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 925 files, 1723739/1725419 clusters
Wed 26 Jun 23:14:15 CEST 2019: Finished fsck on /mnt/music.
Wed 26 Jun 23:14:16 CEST 2019: Unmounting /mnt/music...
umount: /mnt/music: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
Thu  3 Nov 18:17:24 CET 2016: Starting archiveloop...
Thu  3 Nov 18:17:25 CET 2016: Ensuring music backing file is mounted...
Thu  3 Nov 18:17:26 CET 2016: Mounting /mnt/music...
Thu  3 Nov 18:17:26 CET 2016: Mounted /mnt/music.
Thu  3 Nov 18:17:27 CET 2016: Ensured music drive is mounted.
Thu  3 Nov 18:17:27 CET 2016: Running fsck on /mnt/music...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 925 files, 1723739/1725419 clusters
Thu  3 Nov 18:17:36 CET 2016: Finished fsck on /mnt/music.
Thu  3 Nov 18:17:36 CET 2016: Unmounting /mnt/music...
umount: /mnt/music: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
Thu  3 Nov 18:17:20 CET 2016: Starting archiveloop...
Thu  3 Nov 18:17:21 CET 2016: Ensuring music backing file is mounted...
Thu  3 Nov 18:17:21 CET 2016: Mounting /mnt/music...
Thu  3 Nov 18:17:21 CET 2016: Mounted /mnt/music.
[... 166 lines cut ...]
Mon  1 Jul 21:40:43 CEST 2019: Unmounting /mnt/cam...
Mon  1 Jul 21:40:44 CEST 2019: Unmounted /mnt/cam.
Mon  1 Jul 21:40:44 CEST 2019: Finished archiving.
Mon  1 Jul 21:40:44 CEST 2019: Copying music...
Mon  1 Jul 21:40:44 CEST 2019: Starting music sync...
Mon  1 Jul 21:40:44 CEST 2019: Ensuring music backing file is mounted...
Mon  1 Jul 21:40:45 CEST 2019: Mounting /mnt/music...
Mon  1 Jul 21:40:45 CEST 2019: Mounted /mnt/music.
Mon  1 Jul 21:40:45 CEST 2019: Ensured music drive is mounted.
Mon  1 Jul 21:40:45 CEST 2019: Running fsck on /mnt/music...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 925 files, 1723739/1725419 clusters
Mon  1 Jul 21:40:48 CEST 2019: Finished fsck on /mnt/music.
Mon  1 Jul 21:40:48 CEST 2019: Copying music from  archive...
Mon  1 Jul 21:40:48 CEST 2019: Copied 0 music file(s), skipped 1 previously-copied files, encountered 0 errors.
Mon  1 Jul 21:40:48 CEST 2019: Unmounting /mnt/music...
Mon  1 Jul 21:40:48 CEST 2019: Unmounted /mnt/music.
Mon  1 Jul 21:40:48 CEST 2019: Finished copying music.
Mon  1 Jul 21:40:48 CEST 2019: Connecting usb to host...
Mon  1 Jul 21:40:48 CEST 2019: Connected usb to host.
Mon  1 Jul 21:40:48 CEST 2019: Waiting for archive to be unreachable...

marcone, 2019-07-01:
That shows it moving files. Are you saying those file didn't actually show up on your server?

Vipercat, 2019-07-01:
Yes, it's moving files in the end. After I've 'rebooted' as you instructed just a while ago. Earlier today however, after arriving home it didn't. Files were under 'SavedClips', Wifi Connection okay, NAS shares mounted, but no moving of files. Could it be that the use of the ppamidimarri/TeslaDashcamWeb fork, via which I regularly check the footage in the 'RecentClips' folder via the browser on my pc disrupts the archiveloop process?

marcone, 2019-07-01:
That seems possible. Teslausb disconnects the drive from the car and then
mounts it locally on /mnt/cam, and inspects that for files to archive. If
something is interfering with that and perhaps preventing it from mounting
the drive locally, it would just see an empty mountpoint and not archive
anything.

Vipercat, 2019-07-01:
After reviewing the 'RecentClips' footage as mentioned above (using ppamidimarri/TeslaDashcamWeb fork) I do revert back to 'recording mode' which unmounts cam. I'll check to see if I can reproduce the issue.

kdvlr, 2019-07-02:
It did not work for me somehow. 
[diagnostics_07012019.txt](https://github.com/marcone/teslausb/files/3348006/diagnostics_07012019.txt)

kdvlr, 2019-07-02:
Never mind. Restarted it and seems to work.

marcone, 2019-07-07:
The change mentioned in https://github.com/marcone/teslausb/issues/63#issuecomment-507047122 has been merged to main-dev.

closed 2019-07-07 by marcone