marcone/teslausb #66: Mounting on the PC, not in the Tesla

paulio23, 2019-06-05:
I've rebuilt this thing about 7 or 8 times already over the past couple of weeks. It flashes and sets up just fine, mounts the "CAM" filesystem on the computer (pc and mac) with the TeslaCam directory inside, and i can ssh in just fine. The problem is that when putting it in the Tesla (model s ap 2.5 or my wife's model 3), the dashcam icon doesn't show up. tried waiting around 10 minutes and nothing. i can ssh in to the pi, so i know its getting power.

any ideas?
[diagnostics.txt](https://github.com/marcone/teslausb/files/3259175/diagnostics.txt)

marcone, 2019-06-06:
Did you use the same USB cable with the car and with the PC?
Because it kinda looks like the car and the rpi aren't connecting, which is often due to a bad cable.
Normally `dmesg` should show something like:
```
[36855.275340] LUN: removable file: /backingfiles/music_disk.bin
[36855.275428] LUN: removable file: /backingfiles/cam_disk.bin
[36855.275437] Number of LUNs=2
[36855.280122] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[36855.280140] g_mass_storage gadget: g_mass_storage ready
[36855.280156] dwc2 20980000.usb: bound driver g_mass_storage
[36855.443524] dwc2 20980000.usb: new device is high-speed
[36855.598586] dwc2 20980000.usb: new device is high-speed
[36855.696850] dwc2 20980000.usb: new address 4
[36855.709348] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
```
but yours ends after that "bound driver g_mass_storage" line.

marcone, 2019-06-08:
I'm going to assume it was a bad cable

closed 2019-06-08 by marcone

paulio23, 2019-06-09:
> thread [2].
> Links:

ive tried a few different cables by now, no luck aside from mounting on 
a computer. i cant even ssh into it when plugged in to the tesla, unless 
i plug it in to a cigarette lighter adapter it seems. not sure if my car 
doesnt put out enough current for a measly pi zero, or if its a pi 
problem. regardless, i ordered a new pi zero W, should be here tomorrow.

marcone, 2019-06-09:
Dumb question, but you're using the correct USB port on the rpi when you plug it in to the car? The two ports are not interchangeable:
[image]

paulio23, 2019-06-09:
Not a dumb question, but yes, the correct port. 

An aside: I have a Qi charger for phone (was unplugged during pi testing) and my phone won’t charge of the Qi is plugged into the Tesla directly (only the 12v adapter). When the pi is plugged in to the Tesla, the little LED goes on but it isn’t reachable.

paulio23, 2019-06-19:
I think im ready to give up. I got a new zero w and microsd, rebuilt from scratch. It works like it’s supposed to on my wife’s model 3 (dash cam icon shows up, I can ssh in). On my model S (an October 2017, MCU1/AP2.5) the icon never shows and I can’t ssh in. The little green light on the zero blinks, but that’s the only sign of life. When using a flash drive in the car directly, it works fine.

The only way to ssh in is to attach another micro-usb cable to the power port and plug that in too. When I do that, I see that the virtual drive isn’t mounting. Last thing on dmesg is:

 [   32.181518] dwc2 20980000.usb: bound driver g_mass_storage

I know there’s a thread about some people needing to use two cables (one for power one for data), but I can’t even get that to work. 

My guess is that its time give up, the OS might be blocking it or something. Im on 2019.20.1, same failure on 2019.16.x before. Ill try again after my next update. I appreciate all your effort, its a really cool project and works great on my wife’s car.