marcone/teslausb #67: Tesla_api.py _log syntax error

raushel, 2019-06-06:
Tried running tesla_api.py manually after seeing the Json was blank, but ran into the following error. I’m guessing I need to escape some symbols in my password, but currently only have it in single quotes. 

root@teslausb:~/bin# python tesla_api.py
  File "tesla_api.py", line 235
    print("{}: {}".format(_get_log_timestamp(), msg), flush=flush)
                                                           ^
SyntaxError: invalid syntax

marcone, 2019-06-06:
Looks like the way the script parses out the password from the setup file isn't robust against non-ascii characters. Not sure how to handle that though. @coredumperror

marcone, 2019-06-06:
Oh I see what's going on. The script needs to be run with python 3, and the standard "python" on rpi is python 2.
Run it as `python3 tesla_api.py` instead. or just execute it directly, I suppose.

closed 2019-06-06 by marcone

raushel, 2019-06-06:
Python3 helped get rid of the execution error, thanks!

Had to get rid of the single quotes around the password, then it worked running commands (& json populuated)

marcone, 2019-06-06:
Hmm, that's probably going to cause problems when the password contains characters that are special to bash, and thus needs to be quoted/escaped.

coredumperror, 2019-06-06:
Good point. It wouldn't be hard to strip `'` off the ends of the string in the python script, though, if that would make sense. On line 381 in `tesla_api.py`1, add `.strip("'")` to the end.