marcone/teslausb #68: Sentry mode activates alarm

basdejong95, 2019-06-08:
Hi, 

First of all, thanks for your great work! Great functionality. 

We have one problem however, sometimes Sentry mode activates the alarm because the usb "disconnected". We have to reboot the PI afterwards to restore the dashcam functionality. 

Our Model 3 is on 2019.16.3.2

Is there any way we can provide you with logs or anything to troubleshoot this problem?

Kind regards,

Bas de Jong

redosk, 2019-06-08:
I had the same problem twice these days with the same version...
I think it's a protection from the car against somebody stealing the key but... the question is why does the Pi disconnects... :)

basdejong95, 2019-06-08:
Yeah i have read on the internet several times that the alarm goes of if the usb is removed during sentry mode. But indeed i have no idea why the data disconnect is happening.

thijsdejong, 2019-06-08:
Just to add to this (this is about that same car). The pi is plugged in with 2 cables into both usb slots of the car. One for data, one for power (it would otherwise keep shutting down).

marcone, 2019-06-08:
Using two cables won't help if both ports turn off :)
I doubt that's the case here though, since sentry mode was active, but you can tell by ssh'ing into it afterwards and checking `uptime`.
If the pi remained running but disconnected the drive from the car, then please run `setup-teslausb diagnose`, as described in the pinned "read this first" issue.

marcone, 2019-06-08:
Now that I've gotten firmware 2019.16.3.2 myself, I can finally reproduce the issue with the alarm going off when the drive disconnects. It takes a good amount of time on my car before the alarm actually goes off, like 15-20 seconds. Is that the case for you as well? (you can run `modprobe -r g_mass_storage` to manually disconnect the drive)

notrhj, 2019-06-12:
If  I ssh into the pi, make changes or not than reboot the pi
The Alarm goes Off and the USB port that the pi was plugged into goes dead.
The only way to get the port back is todo a two steering wheel button reset.

vfortiercldr, 2019-06-12:
> If I ssh into the pi, make changes or not than reboot the pi
> The Alarm goes Off and the USB port that the pi was plugged into goes dead.
> The only way to get the port back is todo a two steering wheel button reset.

I seem to have the same issue with mine. The alarm went off and on the next drive no icon for recording and it took like an hour and a few drives for the icon to come back.

vfortiercldr, 2019-06-13:
I just discovered a weird way on mine. I have the battery plugged in the power port and the usb (with no power pin) in the usb port. I booted the pi at 8:12 am and it was working OK in my entry as it had access to wifi. I left for lunch at 12h15 and I can see it did not have the archive access and slept before retrying. but at some point it said attempts exhausted and I think it did restart after: 
Thu 13 Jun 08:12:42 EDT 2019: Connecting usb to host...
Thu 13 Jun 08:12:42 EDT 2019: Connected usb to host.
Thu 13 Jun 08:12:43 EDT 2019: Waiting for archive to be unreachable...
Thu 13 Jun 12:15:08 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:09 EDT 2019: Retrying...
Thu 13 Jun 12:15:10 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:11 EDT 2019: Retrying...
Thu 13 Jun 12:15:12 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:13 EDT 2019: Retrying...
Thu 13 Jun 12:15:14 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:15 EDT 2019: Retrying...
Thu 13 Jun 12:15:16 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:17 EDT 2019: Retrying...
Thu 13 Jun 12:15:17 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:18 EDT 2019: Retrying...
Thu 13 Jun 12:15:18 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:20 EDT 2019: Retrying...
Thu 13 Jun 12:15:20 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:21 EDT 2019: Retrying...
Thu 13 Jun 12:15:21 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:22 EDT 2019: Retrying...
Thu 13 Jun 12:15:22 EDT 2019: Sleeping before retry...
Thu 13 Jun 12:15:23 EDT 2019: Retrying...
Thu 13 Jun 12:15:23 EDT 2019: Attempts exhausted.
Thu 13 Jun 12:15:23 EDT 2019: Archive is unreachable.
Thu 13 Jun 12:15:23 EDT 2019: Waiting for archive to be reachable...
Thu  3 Nov 13:17:26 EDT 2016: Starting archiveloop...

palma, 2019-06-16:
In my case the car alarm was activated when rebooting my Wifi router. So it seems loosing the Wifi connection makes the car think the usb (pi) is unmounted (at least in this case).

cmedved, 2019-06-18:
My pi zero w would freeze randomly. Not with this specific project though. To stabilize it, I added over_voltage=6 to /boot/config.txt

shackrock, 2019-06-25:
I am curious what the final recommendation is here to assuredly NOT get an alarm with this - should we not use the "keep car awake" feature - will that prevent the alarm?  Or are there too many things causing the sentry mode alarm at this time?

marcone, 2019-06-26:
The alarm goes off when the drive is disconnected from the car while in Sentry mode. This can happen in two ways: the car is disconnected from wifi and then reconnects, or the Pi restarts. Prevent those two things from happening, and the alarm shouldn't go off.

vfortiercldr, 2019-06-26:
So how do you prevent the pi from disconnecting of wifi??

marcone, 2019-06-26:
Make sure you have good signal where you park the car, and that your wifi access point doesn't get rebooted or otherwise boots the Pi off the network? It shouldn't disconnect on its own.

redosk, 2019-06-26:
> The alarm goes off when the drive is disconnected from the car while in Sentry mode. This can happen in two ways: the car is disconnected from wifi and then reconnects, or the Pi restarts. Prevent those two things from happening, and the alarm shouldn't go off.

You said "the car is disconnected from wifi", do you mean the Pi or really the car ?

Should the diagnose command be issued quickly (is there a delay or need the pi not to be rebooted to keep all the interesting logs) ?

shackrock, 2019-06-26:
Also why would the Pi restart, how can each of these be prevented?

marcone, 2019-06-26:
> You said "the car is disconnected from wifi", do you mean the Pi or really the car ?

Sorry, I meant the Pi.
 
> Should the diagnose command be issued quickly (is there a delay or need the pi not to be rebooted to keep all the interesting logs) ?

It's best to run it as close to the occurrence of the problem as possible.

marcone, 2019-06-26:
> Also why would the Pi restart, how can each of these be prevented?

Loss of power is the most likely cause of a restart. and there have been some issues reported when powering the Pi from a single USB port. Two cables works better apparently, but even better is to use a battery pack or a hardwired connection.

shackrock, 2019-06-26:
Is this because the USB ports power off basically?

Also, is there a reason that a typical USB stick wouldn't alarm the car
when losing power, but the Pi will?  To me it just thinks it's a USB stick,
maybe there is some sort of configuration that could be adopted?

marcone, 2019-06-26:
> Is this because the USB ports power off basically?

That's the theory.

> Also, is there a reason that a typical USB stick wouldn't alarm the car when losing power, but the Pi will? To me it just thinks it's a USB stick, maybe there is some sort of configuration that could be adopted?

The car doesn't seem to mind a brief disconnect of the USB stick, so a short power dip on the USB port  isn't an issue for a USB stick that will be available again as soon as power comes back. The Pi needs to boot up again though, which takes much longer.

alangoykhman, 2019-06-26:
Was there an update to the original problem that I missed? I can't seem to find the threads anymore, but I seem to recall people saying it happened with battery backup as well? (I know vfortiercldr mentioned it, but his issue seems to have stemmed from wifi disconnect.)

At the very least, I know I tried having the pi plugged into a backup battery and still had the alarm trigger. (This wasn't a wifi issue since I wasn't at home. I have Sentry disabled at home anyway.) This was at least 2-3 weeks ago though and I haven't really used it much since b/c it seemed to always cause the alarm to go off at some point and causes the dashcam to stop working until I pull the plug and put it back in manually. I also haven't updated the scripts in a long time...

I can re-run setup and try again with & without a battery over the next few days and see if it's still an issue. (If Sentry mode is on, the USB should never lose power, correct? So technically we shouldn't need a battery? Or does the USB power fluctuate even in Sentry mode?)

universenz, 2019-09-30:
Can anyone confirm if this is still an issue? If it has been resolved in the latest version 10 update, should we close this as resolved?

acraigl, 2019-09-30:
> Can anyone confirm if this is still an issue? If it has been resolved in the latest version 10 update, should we close this as resolved?

The added feature of disabling sentry mode when at home has made it a non-issue for me.

palma, 2019-10-01:
> Can anyone confirm if this is still an issue? If it has been resolved in the latest version 10 update, should we close this as resolved?

I haven't had this issue for months (and still running on V9), so closing it is fine with me.

redosk, 2019-10-01:
Same for me, no problem since a few updates. I think it can be closed !

prathiraj, 2019-10-01:
seems non-issue at this point. Also note, the alarm will go off when the USB drive is out of space, with the archiving solution this should not matter, but if you're out on a road trip, make sure you have enough space for Sentry

universenz, 2019-10-01:
> seems non-issue at this point. Also note, the alarm will go off when the USB drive is out of space, with the archiving solution this should not matter, but if you're out on a road trip, make sure you have enough space for Sentry

Just to be clear - the alarm does not go off when you run out of space using a normal USB stick on v9.32.2.2 With v10 it is meant to loop record. Are you stating that TeslaUSB sets the car's alarm off when the disk runs out of space? Which model and firmware are you running?

prathiraj, 2019-10-01:
not sure about v10, before I moved to teslausb, pre v10 my alarm would go off when the disk is full. Just went and learnt about loop recording, yes this would prevent running out of storage. Thanks

universenz, 2019-10-01:
> not sure about v10, before I moved to teslausb, pre v10 my alarm would go off when the disk is full. Just went and learnt about loop recording, yes this would prevent running out of storage. Thanks

Interesting. Yours is the first I've heard where running out of space set off the alarm. What model car do you have and what brand USB stick were you using?

prathiraj, 2019-10-01:
Model 3. First time I was startled and ran to the car to see what happened, nothing. Then noticed the cam was off (grey x), after plugging in the USB (Sandisk Cruzer 32 GB) noticed it was full.

universenz, 2019-10-01:
> Model 3. First time I was startled and ran to the car to see what happened, nothing. Then noticed the cam was off (grey x), after plugging in the USB (Sandisk Cruzer 32 GB) noticed it was full. Yes, got notified through the app.

Interesting. Were you notified via the Tesla app that the alarm had been set off?

marcone, 2019-11-17:
Haven't seen any reports of this recently, so I assume it's no longer an issue.

closed 2019-11-17 by marcone