marcone/teslausb #69: fsck error

pgardiner, 2019-06-08:
My drive mounts fine in the car, and files are saved find, but whenever I reconnect to wifi, and rsync runs, I get an fsck error, and the car can't see the drive any more. I assume this is because it can't dismount properly, but not sure. Here is what I get in the log:

Sat  8 Jun 17:47:09 EDT 2019: Waiting for archive to be reachable...
Sat  8 Jun 17:56:10 EDT 2019: Archive is reachable.
Sat  8 Jun 17:56:30 EDT 2019: Archiving...
Sat  8 Jun 17:56:30 EDT 2019: Disconnecting usb from host...
Sat  8 Jun 17:56:30 EDT 2019: Disconnected usb from host.
Sat  8 Jun 17:56:31 EDT 2019: Starting recording archiving...
Sat  8 Jun 17:56:31 EDT 2019: Ensuring cam file is mounted...
Sat  8 Jun 17:56:31 EDT 2019: Mounting /mnt/cam...
Sat  8 Jun 17:56:31 EDT 2019: Mounted /mnt/cam.
Sat  8 Jun 17:56:31 EDT 2019: Ensured cam file is mounted.
Sat  8 Jun 17:56:31 EDT 2019: Running fsck on /mnt/cam...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 171 files, 47198/3316141 clusters
Sat  8 Jun 17:56:33 EDT 2019: Finished fsck on /mnt/cam.
Sat  8 Jun 17:56:33 EDT 2019: Archiving through rsync...

Is there something wrong with my backing file? If so, is there a way to recreate it? Any other ideas? Note that rsync is working fine - the files are copied. However, after this runs, I can see /mnt/cam/TeslaCam when ssh'd into the pi. Before I got this error, it wasn't mounted.

marcone, 2019-06-09:
This is all working as intended. While the drive is connected to the car, the pi can't see the filesystem, and vice versa.

closed 2019-06-09 by marcone

pgardiner, 2019-06-09:
Actually, when connected to the car, the pi CAN see the filesystem, which is what I thought the problem was. However, I have since determined that there's something wrong with rsync, and it's taking a long time to time out, so /mnt/cam is mounted until it times out. Once I figure out why rsync is bombing out, it should work fine.

marcone, 2019-06-09:
Right, when I said "connected to the car" I was talking about the emulated drive, not the pi itself. The pi remains plugged in to the car, but the disk image is either connected to the car or mounted on the pi, never both at the same time. While it's mounted on the pi (i.e. during archiving), the car won't see it.

pgardiner, 2019-06-09:
Got it, thanks. I think everything is good, except rsync, which I should be able to solve.