marcone/teslausb #70: Connecting to WiFi OK, but Shares never appear

SemoTech, 2019-06-09:
Hello, 
I just installed latest build ([teslausb-20190505](https://github.com/marcone/teslausb/releases/tag/v1.2a)) onto a brand new Pi Zero W v1.1 using a 32Gb uSD card, and am testing while still connected to my Mac via either the [USB Stem](https://www.sparkfun.com/products/14526) or with a micro USB cable (no diference in behaviour). LED on the Pi is solid green either way.

Created and updated the "teslausb_setup_variables.conf" file on the /boot of the uSD card before inserting in the Pi, and also ran "./setup-teslausb" in /root/bin several times (with a fresh "teslausb_setup_variables.conf" in /boot) while troubleshooting.  At one point I think the kernel was updated.  

Anyway, while WiFi is connecting to my SSID just fine, and I can SSH into teslacam as "pi", the "Music" and TeslaCam" shares are not created/do not appear on the Mac.  

The "TESLAUSB_SETUP_FINISHED" file IS created in /boot but the "WIFI_ENABLED" is NOT.

I have linked the "diagnostics.txt" file contents [HERE](https://pastebin.com/pe2bsTeQ).   

Any help would be greatly appreciated.  Thank you in advance.

marcone, 2019-06-10:
Looks you initially tried to setup Gotify, which failed, then tried again without it?
You also have no archiveloop log, though the setup log shows /root/bin/archiveloop was installed. Can you check whether /root/bin/archiveloop actually exists, and whether the archiveloop process is running?

SemoTech, 2019-06-10:
Thank you for the prompt reply @marcone 
Actually I just uncommented the line in the config file that said "# export gotify_enabled=false" since logically it seemed like removing the "#" would actually set Gotify to false.  I did the same for "# export ifttt_enabled=false" After seeing something in the logs about it I commented both lines back and re-ran the "./setup-teslausb" file.  

As for the /root/bin/archiveloop file, it exists.  I am linking the directory contents and a list of running processes [HERE](https://pastebin.com/URTimDiH)

Should I just re-flash the uSD without latest release, and start the Pi from scratch with the "teslausb_setup_variables.conf" in /boot?

marcone, 2019-06-10:
You can try running archiveloop manually and see if it prints any helpful error messages.
I.e, ssh into the pi and
```
sudo -i
/root/bin/archiveloop
```

SemoTech, 2019-06-10:
OK, I got eager and restarted from scratch.  After internal processing a a self-reboot I did get a "CAM" shared folder, but no "Music" folder.

In order for Tesla to write to the folder it needs to be called "TeslaCam" not "CAM".  How/where can I change that?

marcone, 2019-06-10:
You shouldn't have to create a TeslaCam folder, setup already did that.
"CAM" is the label  of the emulated drive. If you doubleclick on it to open the drive you'll see the TeslaCam folder.
You're not getting the music drive because you've specified all space (100%) to be used for dashcam recordings. Set that to a lower amount in the .conf file, and you'll get the music drive.

SemoTech, 2019-06-10:
Got it!  Appreciated @marcone looks like I am almost there...

Currently the  "teslausb_setup_variables.conf" file is no longer in the /boot drive, I assume for security reasons.  So basically I have to re-create it, copy the customized contents, and just change the 100% for video to a lower percentage, halt the system, unplug/replug and it should create both shares, correct?

marcone, 2019-06-10:
You don't have to rewrite the setup config to /boot. The setup scripts move it to /root, and will read it from there too. So just edit the version in /root, then run /root/bin/setup-teslausb, and reboot afterwards.

SemoTech, 2019-06-10:
OK, edited root/teslausb_setup_variables.conf and changed "export camsize=100%" to "export camsize=60%", saved the file, then ran /root/bin/setup-teslausb, and once done, rebooted.  

The CAM share cam back on the Mac, but inside it is just the "TeslaCam" folder, there is no "Music" folder.

1) Did I miss something to get the "Music" folder?
  
2) While I understand that the filesystem will check and self-repair, is that done in real-time or based on a timer?  

3) Also, if the video partition gets full, will Tesla stop recording or will the files be replaced in order of creation?

marcone, 2019-06-10:
There is no Music folder. The rpi emulates a separate drive that contains the music.  The car sees them as two separate USB drives. If you plug the rpi into a PC, you'll see the CAM drive and a MUSIC drive. The car does not require there to be a specific folder for music.

Filesystem check is done whenever the drive(s) are disconnected from the car and mounted on the rpi to do archiving.

The car erases non-saved recordings after one hour, and teslausb erases saved recordings after they've been backed up. Therefore, if you make sure the camera drive is large enough for 1 hour of recordings, plus however much footage you think you (or Sentry mode) will save before it has access to wifi again, you'll be OK. It needs about 5.5 GB per hour.

closed 2019-06-10 by marcone

SemoTech, 2019-06-10:
Thanks for all the expanantions.  Seems I am halfway there.  

The CAM/TeslaCam volume and share appear and are writable by the car!
The MUSIC volume does NOT appear/get mounted!

1) Contents of “/root/teslausb_setup_variables.conf” are [HERE](https://pastebin.com/8T7nTnrw) - Note that Music has been given 11GB of space.
1) List of files in /boot folder is [HERE](https://pastebin.com/rEayWpft)
2) Contents of "teslausb-headless-setup.log" is [HERE](https://pastebin.com/De3ja7Zs)
3) Contents of /root/bin folder are [HERE] (https://pastebin.com/mF0V47gG)
4) Contents of /mutable folder and “archiveloop.log” are [HERE](https://pastebin.com/URX0XLbU) - Note “No music archive configured” entry towards the end of the log!

How do I get the MUSIC volume to mount?

SemoTech, 2019-06-10:
OK, looks like problem is finally Solved!

Seems there is an issue with expecting things to change by modifying the "/root/teslausb_setup_variables.conf" file AFTER the initial setup, then running /root/bin/setup-teslausb and then rebooting.  At least for the MUSIC volume, it does not work.  

Only way to get everything working was to flash the uSD card again, copy the updated "teslausb_setup_variables.conf" config file to /boot so it can be processed the 1st time with all changes.  There is definitely a bug with processing changes afterwards.

@marcone my sincere thanks for all your help and maybe the logs and info I posted above will help you figure out the re-configuration bug.

marcone, 2019-06-10:
Oh right, setup won't recreate the music file if there's already a camera file. If you delete that it will reconfigure both.
Note that "No music archive configured" means it wasn't configured to automatically sync music to the music drive, not that no music drive was configured.

BTW an easier way to get all that info and more is to run `/root/bin/setup-teslausb diagnose`