marcone/teslausb #71: 2019.16.2 and dwc2 / g_mass_storage crashing on boot

jcuster, 2019-06-10:
I've been trying to get teslausb setup and I've run into a really weird issue that I can't seem to wrap my head around.

Here is how I can minimally reproduce the problem:
Download and flash the `teslausb-20190505` image.
Copy `teslausb_setup_variables.conf` to `/boot/` containing this:
````
export ARCHIVE_SYSTEM=cifs
export archiveserver=nas
export sharename=teslausb
export shareuser=guest
export sharepassword=
export camsize=100%
export SSID=myssid
export WIFIPASS='mypassword'
export HEADLESS_SETUP=true
````

- Setup completes successfully, and the pi can mount my NAS and can connect to my WiFi.
- I can connect to the pi over WiFi.
- I can mount CAM to both my Windows and Mac machines.
- My car is AP 2.5 Model S, running 2019.16.2 with MCU1.
- If I use the power-only port on the pi on any USB port in my Tesla, then it boots fine.
- If I use the data port connected to either of the front USB ports, the pi starts to boot but when it hits the dwc2 / g_mass_storage kernel module then the pi reboots. This repeats in a loop. I confirmed this by manually setting things up and setting up a writeable `/` so that I get the kernel logs.
- If I use the data port connected to the rear ports in the Tesla, everything is fine.
- I ensured I am using the latest kernel available (the teslausb script did this for me).
- I reset the MCU using the double scroll-wheel, and also did a `Power off` in the MCU settings.

Am I missing something here? Is anyone running into this problem with 2019.16.2 (or later?) with a similar car configuration as mine (or even a different configuration?)?

marcone, 2019-06-10:
Try connecting both the USB and the power port on the pi to the car, or use a portable phone charger or other power source on the power port.

jcuster, 2019-06-10:
That worked using either a battery bank or the car's second port. I have to wait until the Pi boots and then connect the data cable before it will work. If I connect the cable before boot is complete it doesn't work. So, what's the problem that I'm having that is different than other people where theirs "just works?"

marcone, 2019-06-10:
Unknown. The car seems to toggle power on the USB port, causing the pi to briefly lose power and thus reboot. I think the people for whom it "just works" are using two cables, batteries or other power sources.

closed 2019-06-10 by marcone