marcone/teslausb #72: Downloading files from raspbian fails when running ./setup-teslausb

dpkonofa, 2019-06-10:
I've attempted to run ./setup-teslausb several times on the Pi0W but it seems like the process is failing while trying to download some things. I know it's not an internet connection issue but don't know why else this would keep failing. Any thoughts?

`root@raspberrypi:~# ./setup-teslausb
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Err:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
  Hash Sum mismatch
  Hashes of expected file:
   - Filesize:11661868 [weak]
   - SHA256:0ad01a1714425ffe1960d168eae667d1a340b2d3e70b33d1b744f2510606f104
   - SHA1:2258dd6aca6f229bd9ebf5b7b376a8a5fe079acc [weak]
   - MD5Sum:5a84963fd9996c1d361ae2b713ab046c [weak]
  Hashes of received file:
   - SHA256:f28fea5c3c236b2da3c1ab109a06a641a07af35d0ceb69bee6ec7a2d978ab625
   - SHA1:3731a7bb1243818c3463cc5a1aed185627c5b4ac [weak]
   - MD5Sum:bcd812c9a826e5f199fc9e3b5f2734ad [weak]
   - Filesize:2112046 [weak]
  Last modification reported: Sat, 08 Jun 2019 16:29:03 +0000
  Release file created at: Sun, 09 Jun 2019 22:27:51 +0000
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 1631 kB in 2min 27s (11.1 kB/s)
Reading package lists... Done
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages.xz  Hash Sum mismatch
   Hashes of expected file:
    - Filesize:11661868 [weak]
    - SHA256:0ad01a1714425ffe1960d168eae667d1a340b2d3e70b33d1b744f2510606f104
    - SHA1:2258dd6aca6f229bd9ebf5b7b376a8a5fe079acc [weak]
    - MD5Sum:5a84963fd9996c1d361ae2b713ab046c [weak]
   Hashes of received file:
    - SHA256:f28fea5c3c236b2da3c1ab109a06a641a07af35d0ceb69bee6ec7a2d978ab625
    - SHA1:3731a7bb1243818c3463cc5a1aed185627c5b4ac [weak]
    - MD5Sum:bcd812c9a826e5f199fc9e3b5f2734ad [weak]
    - Filesize:2112046 [weak]
   Last modification reported: Sat, 08 Jun 2019 16:29:03 +0000
   Release file created at: Sun, 09 Jun 2019 22:27:51 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
root@raspberrypi:~# ./setup-teslausb
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Err:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
  Hash Sum mismatch
  Hashes of expected file:
   - Filesize:11661868 [weak]
   - SHA256:0ad01a1714425ffe1960d168eae667d1a340b2d3e70b33d1b744f2510606f104
   - SHA1:2258dd6aca6f229bd9ebf5b7b376a8a5fe079acc [weak]
   - MD5Sum:5a84963fd9996c1d361ae2b713ab046c [weak]
  Hashes of received file:
   - SHA256:2b13f286dfcb279a9fe2cea775618fad21d01d23aefa6ce47f4c80224f9deaf8
   - SHA1:c2edcff6364c761c0a43b210a23a71ecf95dd798 [weak]
   - MD5Sum:ac1001893c50dfd8a38837d2e78e4985 [weak]
   - Filesize:395041 [weak]
  Last modification reported: Sat, 08 Jun 2019 16:29:03 +0000
  Release file created at: Sun, 09 Jun 2019 22:27:51 +0000
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 410 kB in 4min 37s (1475 B/s)
Reading package lists... Done
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages.xz  Hash Sum mismatch
   Hashes of expected file:
    - Filesize:11661868 [weak]
    - SHA256:0ad01a1714425ffe1960d168eae667d1a340b2d3e70b33d1b744f2510606f104
    - SHA1:2258dd6aca6f229bd9ebf5b7b376a8a5fe079acc [weak]
    - MD5Sum:5a84963fd9996c1d361ae2b713ab046c [weak]
   Hashes of received file:
    - SHA256:2b13f286dfcb279a9fe2cea775618fad21d01d23aefa6ce47f4c80224f9deaf8
    - SHA1:c2edcff6364c761c0a43b210a23a71ecf95dd798 [weak]
    - MD5Sum:ac1001893c50dfd8a38837d2e78e4985 [weak]
    - Filesize:395041 [weak]
   Last modification reported: Sat, 08 Jun 2019 16:29:03 +0000
   Release file created at: Sun, 09 Jun 2019 22:27:51 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.`

marcone, 2019-06-10:
Never seen that before, but that looks like an apt-get issue, not a teslausb issue.
Did you flash the [latest release image](https://github.com/marcone/teslausb/releases/tag/v1.2a)?

marcone, 2019-06-11:
The internet suggests that this is caused by a bad package download due to network problems, and that once it's in that state you need to run `apt-get clean` to recover.

closed 2019-06-11 by marcone