marcone/teslausb #75: Write Speed Concerns?

acraigl, 2019-06-12:
This isn't an issue exactly, but I've been getting some video corruption from time to time and began to wonder if the USB 2.0 bus in the Pi Zero is to blame? I understand there's not a lot we can do about it given the hardware limitation, but thought it was worth bringing up all the same.

In all other ways, this solution is working perfectly, so thanks again @marcone!

marcone, 2019-06-12:
If the corruption you're seeing is of the "part/all of the picture is green for a while" variety, then that's a car-problem, not a pi/usb problem. It happens with regular USB drives as well, often at the start of a recording.

The car itself also has USB 2.0 ports, so there's no benefit to using a USB 3 drive. USB 2.0 is more than fast enough for the 1.5 megabytes per second that the car currently uses for the three camera streams. Even with the Pi Zero's mass storage emulation overhead, I'm measuring write speeds well above that, with enough headroom for Tesla to add more cameras in the future.

closed 2019-06-12 by marcone

acraigl, 2019-06-12:
Cool, thanks for the info.