marcone/teslausb #76: How to force connect to home network when connected to car-hotspot?

pimvdnoll, 2019-06-12:
This repo is really amazing and works really well! 
I would like to know if it's possible to let the teslausb search for my home network while its connected to my car hotspot (the hotspot and teslausb are both powered by an always-on-powerbank) so the archiving loop is never triggered 😭 

 I tried to add scripts to the crontab to reset the network at midnight but these crontab entries are not preserved after reboot? (this will fix my issue also I think 👍 )

thanks for helping out and keep up the good work! ⭐️

marcone, 2019-06-13:
While it's not mentioned in the manpage for wpa_supplicant.conf, https://raspberrypi.stackexchange.com/questions/58304/how-to-set-wifi-network-priority and other pages mention a "priority" setting that can be specified for networks, so you might be able to set things up such that it always prefers your home network, when in range.

Otherwise, if you want to go the crontab route you'll need to make sure that /var/spool/cron/crontabs points to somewhere on the /mutable partition. Right now /var/spool links to /tmp, and so is not preserved across reboots.

closed 2019-06-13 by marcone

pimvdnoll, 2019-06-13:
Awesome @marcone for the quick response. I added Priority already in the wpa_supplicant.conf but that is not checked when connected to the car hotspot and the home wireless is also in range. 
Will fix the cronjob that wil reset the wireless adapter at midnight with a script running on the /mutable partition 👍

pimvdnoll, 2019-07-07:
Ok, I have to go to the crontab route but when editting crontab -e I can't get it to work to execute a script located on /mutable
How can I add crons which stays working after reboot? thnx @marcone !

marcone, 2019-07-07:
Permission problem maybe? The 'crontabs' folder needs to have its sticky bit set.
This seems to work for me:
```
mkdir /mutable/crontabs
chmod 1730 /mutable/crontabs
echo '/mutable/crontabs /var/spool/cron/crontabs none defaults,bind 0 0' >> /etc/fstab'
```

pimvdnoll, 2019-07-09:
You are the man. Works like a charm @marcone 🥇