marcone/teslausb #78: Feature request: AP and Web interface

gkwok1, 2019-06-14:
1) Integration with the following repo to provide access to the drive while you are not at home and do not have access to a hotspot.

https://github.com/ppamidimarri/TeslaDashcamWeb
https://github.com/lukicdarkoo/rpi-wifi

It would be awesome if the TeslaDashCamWeb can play all 3 videos at once similar to https://sentrycam.appspot.com/

Use case: I see a sentry alert on the dash when I return and I want to immediately watch the video instead of waiting until I get home or disconnect the drive from the car to connect to my phone via USB-OTG adapter.

2) Mirror the TeslaCam folders (SavedClips) and (RecentClips) to my Share drive which has a folder called TeslaCam. Simulating rsync-ing the USB drive to my share.

Use case: I want all the videos recorded to be on my share.

kdvlr, 2019-06-14:
I had a case where I parked my car at the airport and I got alerts about sentry mode, but could not do anything till I got home. I can see this being useful if someone notices something at a location, but does not have to wait till they go home to view the footage. I did setup a simple [filebrowser.xyz](filebrowser.xyz) instance, but don't want to corrupt existing footage. 

Also, the simultaneous mode wifi is pretty cool. I did not know that. 

The browser setup from TeslaDashcamWeb seems complicated. It's a fairly straightforward process to setup a simple file browser. 
```
curl -fsSL https://filebrowser.xyz/get.sh | bash
filebrowser -a 0.0.0.0 /mnt/cam
```
I know mounting /mnt/cam in rw simultaneously with usb gadget will result in corruption. However, I don't know if it is equally risky to mount /mnt/cam as readonly when archive is not reachable and allow filebrowser to serve files from a read only mount. 

Basically, 
when archive is unreachable
mount cam as ro
start filebrowser 
when archive is reachable
stop filebrowser
mount cam as rw
copy files

marcone, 2019-06-14:
The simultaneous client/AP wifi stuff is indeed pretty cool, and I did not know such a thing was possible either. I definitely want to add support for that. The filebrowser thing also seems pretty neat, and more universal than the current CIFS server support, though I'm a little apprehensive of it because it's fairly large but not under active development, per their github page.

gkwok1, 2019-06-14:
I got the simultaneous client/AP working without crontab or bringing ap0 or wlan0 down and up. I added the following lines to /root/rc.local to enable dns and internet forwarding.

sudo sysctl -w net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -s 192.168.10.0/24 ! -d 192.168.10.0/24 -j MASQUERADE
sudo systemctl restart dnsmasq

I also added /var/lib/misc to tmpfs in /etc/fstab since dnsmasq needs to write to that folder.

ppamidimarri, 2019-06-14:
@gkwok1 I just came across this fork. I will work on getting TeslaDashcamWeb to load all three videos side-by-side. 

The simultaneous AP and managed mode stuff is interesting, I have to learn how it works.

wwwb0n3zcom, 2019-06-21:
Something like this would be amazing:
https://github.com/ppamidimarri/TeslaDashcamWeb

ppamidimarri, 2019-06-25:
@gkwok1 I poked around a bit and I have bad news. The only way TeslaDashcamWeb can show all three videos combined is if I run ffmpeg to merge video files when they are saved. Merging three one-minute files is taking more than three full minutes in my Pi0W in my tests so far. Doing the merge in real-time when the user wants to see the videos is therefore a no-go. One option is to have a script running that merges files whenever they are put in SavedClips, so they are ready for consumption later when connected at home. I tried this, but with very mixed results. For one thing, it seems to make the Pi not very responsive to the car (I guess because ffmpeg eats up CPU) and I see the dashcam icon disappearing in the car. I'd rather have a reliably working teslausb than have a better TeslaDashcamWeb, so I reverted to my old build. 

Another alternative I am thinking of is to setup a network share at home on a more powerful computer, and then have that computer do the video merges and serve up a website to view the merged footage. I just got my second Jetson Nano and this may be a good project for that device. I will post that if it turns out well.

Any other ideas to solve this problem are more than welcome! 

@marcone I know that right now this is not integrated into your project, but I do want to figure out a way to integrate. So thanks for tolerating an off-topic discussion here.

ppamidimarri, 2019-06-25:
> The simultaneous client/AP wifi stuff is indeed pretty cool, and I did not know such a thing was possible either. I definitely want to add support for that. The filebrowser thing also seems pretty [...]

@marcone I tried filebrowser on a Jetson Nano (as an easier alternative to ResponsiveFileManager that I was using for my TeslaDashcamWeb project). I really like the filebrowser interface when it works, but I found it to be quite flaky. It's possible that it is just stupidity on my part. But your apprehensions on this may be well-founded.

raushel, 2019-06-25:
There is a python repo (tesla_dashcam) that will merge the 3 videos together using ffmpeg that I’ve been using for awhile now to do just this. 

I wrote a PowerShell wrapper to merge / rename / timestamp the files for loading into Plex, which is then viewable anywhere to solve for this. Runs as a scheduled task on my media VM. 

https://github.com/raushel/tesla_wrappers

gkwok1, 2019-06-25:
> @gkwok1 I poked around a bit and I have bad news. The only way TeslaDashcamWeb can show all three videos combined is if I run ffmpeg to merge video files when they are saved. Merging three one-minute files is taking more than three full minutes in my Pi0W in my tests so far. Doing the merge in real-time when the user wants to see the videos is therefore a no-go. One option is to have a script running that merges files whenever they are put in SavedClips, so they are ready for consumption later when connected at home. I tried this, but with very mixed results. For one thing, it seems to make the Pi not very responsive to the car (I guess because ffmpeg eats up CPU and does a lot of SD card read/writes) and I see the dashcam icon disappearing in the car. I'd rather have a reliably working teslausb than have a better TeslaDashcamWeb, so I reverted to my old build.
> 
> Another alternative I am thinking of is to setup a network share at home on a more powerful computer, and then have that computer do the video merges and serve up a website to view the merged footage. I just got my second Jetson Nano and this may be a good project for that device. I will post that if it turns out well.
> 
> Any other ideas to solve this problem are more than welcome!
> 
> @marcone I know that right now this is not integrated into your project, but I do want to figure out a way to integrate. So thanks for tolerating an off-topic discussion here.

@ppamidimarri The only thing I can think of is to reach out to the author of https://sentrycam.appspot.com/ to see if they can shed some light into this project since website supports mobile browsers as well. Maybe instead of "loading" the video, the page will scan the drive and load the list.

kdvlr, 2019-06-25:
Another option is https://github.com/BobStrogg/teslacam-browser . I did raise an issue a couple of weeks back if we can implement this directly on a Pi and the last update seems like it could happen. https://github.com/BobStrogg/teslacam-browser/issues/29

marcone, 2019-09-03:
The AP part of this feature request is now implemented in the main branch.

SemoTech, 2019-09-19:
This is great news @marcone!  I wonder if now we can edit the settings for the teslacam directly, using just a web browser connected to the AP on the Pi, and then initiate an update and reboot for the changes to take hold?

marcone, 2019-09-19:
There's no technical reason that couldn't be done (in fact this could have been done before too, it's just that now you can also do it while away from wifi), but I have no plans to do so.

SemoTech, 2019-09-19:
> There's no technical reason that couldn't be done (in fact this could have been done before too, it's just that now you can also do it while away from wifi), but I have no plans to do so.

It's good to hear it's possible.  May I ask why you do not want to implement?  Is it the complexity?  Seems it would greatly simplify setup and updates while away from a computer.

marcone, 2019-09-19:
The setup process downloads the current versions of all scripts from github, so you'd still need an internet connection for the Pi, or change setup so it has a mode where it only updates current settings but does not download new scripts.
Personally, I've never needed to change the setup while on the go, so I just see it as a lot of complexity for very little practical benefit.

SemoTech, 2019-09-19:
I see, assuming there is a hotspot in the car, would it make sense to setup a cron job to, for example, run the update script once a week, and thus bring in the latest script / feature changes?  It could also pull the latest version of the config sample file, and merge it with the customized live one, so you'd always have the latest version of both each week, and also be able to modify the config to take advantage of new features through SSH if a web interface is too complex?

marcone, 2019-09-19:
I'm not sure I like the idea of automatic updates (something could go wrong during the update, leaving things in an inconsistent state, which you wouldn't realize until later when things suddenly don't work).
I like the idea of merging in the latest sample config file though. Might also be neat to have it at least check if an update is available, and somehow notify of that.

stepir, 2019-10-08:
Hi @marcone thanks a lot for your work. I'm using it in conjunction with https://github.com/ppamidimarri/TeslaCamMerge 

Do you have plan to embed also a web server so that clips can be browser on car while being connected to the local AP?

marcone, 2019-10-08:
The car will not access local IP addresses, so you'd have to have an internet-connected hotspot that both the car and pi connect to, and that hotspot needs to provide address translation and routing such that the pi appears to the car as being on the public internet.

stepir, 2019-10-08:
but what I'd like to do is using my iPhone and connect to the local AP of the PI then browse the clips directly from the phone

marcone, 2019-10-09:
OK, I guess I misunderstood what you meant by "on car" then :)
You should already be able to browse the clips from your iphone, by configuring Samba on the Pi. You can then browse to the Samba share from the phone and play the videos.

stepir, 2019-10-09:
yeah sorry!!!! 

I know I can use Samba but browsing video from a web interface is more handy when you are on the road, +filebrowser is giving a previews of the files and you can really play them on the fly.

do you see any challenges in using it? the r/o filesystem?

skipfire, 2019-10-13:
Is there any documentation on accessing the web server?  Just installed into my car and I can hit port 80, but there is no content, not even the basic HTML tag.  I expect I am doing something wrong, but not sure where to find what that is.  Basic details like that would be great for the WiKi.

marcone, 2019-10-13:
@skipfire there is no web server

skipfire, 2019-10-13:
@marcone thanks, I missed that it clearly says samba in the hotspot notes.

drgreenberg, 2019-10-27:
> OK, I guess I misunderstood what you meant by "on car" then :)
> You should already be able to browse the clips from your iphone, by configuring Samba on the Pi. You can then browse to the Samba share from the phone and play the videos

@marcone Which folder needs to be shared in smb.conf to enable a phone to see and access the videos?

nockstarr, 2019-10-27:
@drgreenberg 
https://github.com/marcone/teslausb/blob/main-dev/doc/teslausb_setup_variables.conf.sample#L46

drgreenberg, 2019-10-27:
> @drgreenberg
> https://github.com/marcone/teslausb/blob/main-dev/doc/teslausb_setup_variables.conf.sample#L46

So, I had already uncommented this section at setup, but when I attempt to connect, I can’t seem to authenticate using pi or root credentials. I was thinking the share was set up with its own access permissions or still needed to have these permissions manually set up by me. What is the next step after enabling Samba to connect successfully (I am able to connect to the AP, just not to an smb share)?

marcone, 2019-10-27:
If you did not enable Samba guest mode in the config, then the user/password are pi/raspberry.
This is true even if you later changed the pi user's login password to something else, since Samba has its own user list (you can change the password for Samba's pi user with `smbpasswd`)
Also note that it can take several seconds to connect, and I've found that some phone apps will time out and report an error before the connection is established.

drgreenberg, 2019-10-27:
> If you did not enable Samba guest mode in the config, then the user/password are pi/raspberry.
> This is true even if you later changed the pi user's login password to something else, since Samba has [...]

Ah, thanks!

Raiders95, 2019-11-08:
Sorry for what is probably a noob/dumb question, but I didn't uncomment the Samba share enable line in the teslausb_setup_variables.conf filewhen I originally setup the Pi, but now I want to enable so I can connect to the AP when I'm out and view clips.

Can I simply go back into the file and uncomment and reboot?  Does the Pi re-read the setup variables on each boot, or can I force it to re-setup?  I'm comfortable with ssh'ing into the pi, but that's where my knowledge and ability ends.

Thanks!

SemoTech, 2019-11-08:
@Raiders95, its a good question.  Do this:

0 - SSH to the Pi. 
1 - become root:  sudo -i 
2 - enable writing:   /root/bin/remountfs_rw
3 - edit config as you see fit:    nano /root/teslausb_setup_variables.conf
4 - run   /root/bin/setup-teslausb selfupdate
5 - run  /root/bin/setup-teslausb
6 - reboot

senf42, 2019-11-08:
Another dumb question... Should the AP be visible all the time? I uncommented those lines in the setup config and set a custom password, but I haven't seen a network being broadcast from my RPi.

SemoTech, 2019-11-08:
@senf42  Yes it should. Make sure you have no spaces before and after the comment symbol “#” then see the commands above.

marcone, 2019-11-12:
I don't think I'll be working on a web interface, and since AP is implemented, I'm closing this issue.

closed 2019-11-12 by marcone