marcone/teslausb #79: Can't get one step setup going

cribbstechnologies, 2019-06-14:
I downloaded the stretch image and flashed it using etcher. I copied the teslausb_setup_variables.conf.sample file to teslausb_setup_variables.conf and made the appropriate edits, saved, ejected the disk (windows) and booted the pi. The pi never got on my wifi network and isn't accessible via USB. The activity light on the pi is flashing like it's booting Raspbian so it looks like it's doing the right thing. 

After letting it sit for 10 minutes or so it's still not accessible so I power it down and remove the SD card, stick it in the reader and put it back into my computer. There's no log file like the readme suggests and the file I created teslausb_setup_variables.conf is gone. What gives?

marcone, 2019-06-14:
One step setup doesn't work with plain Stretch.
Please use the [teslausb release image](https://github.com/marcone/teslausb/releases) instead.

cribbstechnologies, 2019-06-14:
I used that image, not vanilla Stretch.

The pi isn't reporting itself over bonjour at all. I can see my pi-hole, and my OpenSprinklerPi reporting via bonjour but this pi zero isn't.

marcone, 2019-06-14:
Looks like you didn't configure wifi correctly in the one-step config file then. Maybe a typo in the AP name or password, or forgot to quote/escape special characters in your password. It should be on wifi very quickly, since configuring wifi is the first thing the setup script does.

cribbstechnologies, 2019-06-14:
Is that file supposed to be deleted/moved after boot?

marcone, 2019-06-14:
The setup script moves it from /boot to /root.
Setup also writes some marker files in the /boot folder, like WIFI_ENABLED and TESLAUSB_SETUP_STARTED. Do you see any of those?

cribbstechnologies, 2019-06-14:
I didn't escape the spaces in my SSID, it booted now. I'll make a PR to make this plainly obvious to other such fools :)

closed 2019-06-14 by cribbstechnologies